openbmc/openbmc #3767: ASPEED ast2500 BMC fails to boot

ushasrinivasan-cornelisnetworks, 2021-03-03:
I had a working image built 2/24/2021 and had been happily using it until this morning when the BMC stopped booting.  It would hang shortly after the DRAM message from U-Boot.  Here is a brief snap-shot:
U-Boot 2016.07 (Feb 05 2021 - 19:15:39 +0000)
       Watchdog enabled
DRAM:  496 MiB

DRAM Init-V12-DDR4
0abc1-4Gb-Done
Read margin-DL:0.3705/DH:0.3784 CK (min:0.30)

U-Boot 2016.07 (Feb 05 2021 - 19:15:39 +0000)
       Watchdog enabled
DRAM:  496 MiB

DRAM Init-V12-DDR4
0abc1-4Gb-Done
Read margin-DL:0.3901/DH:0.3960 CK (min:0.30)

U-Boot 2016.07 (Feb 05 2021 - 19:15:39 +0000)

Can someone tell me what "Read margin-DL" message means?   And, is it the cause of my issue? Because, with repeated attempts, the margin error disappeared but the boot hung in different places, e.g., while loading kernel or ramdisk. Once it booted all the way up to the login prompt and then hung.  

The last patch I pulled was: 779a182c6adac1c33ae63e0a20f73e19e97047d3 Feb 3.

Thanks.

arjuncr, 2021-04-03:
```
DRAM Init-V12-DDR4
0abc1-4Gb-Done
Read margin-DL:0.3901/DH:0.3960 CK (min:0.30)
```
The above message will come when u do fresh boot of an evb (ac power cycle the evb) so do not worry  about that.  
This print coming frm aspeed primary bootloader (READ ONLY BOOTLOADER)  

did u modified  any code or configuration in u-boot  ?  
or improper firmware flash also will cause this kind of issue

ushasrinivasan-cornelisnetworks, 2021-04-05:
Thanks, Arjun, for your comment.  I went back to my notes from a month ago to see if I documented how this issue got resolved and I haven't found the resolution, yet.  What was concerning was the various places in which I saw the 'hang.'  I have not seen it since then but I will keep an eye out to see if it reappears.

ushasrinivasan-cornelisnetworks, 2022-04-18:
This problem was possibly due to SPI flash image issues.

closed 2022-04-18 by ushasrinivasan-cornelisnetworks