openbmc/openbmc #3768: Encounter WARNINGs while building the meta-intel/meta-s2600wf

Tommy-git2033, 2021-03-05:
**Unexpected behavior you saw**
Hi，guys，when i build the meta-intel/meta-s2600wf, I encounter Warnings, look the picture:
[image]

Can anyone answer me my doubts, why these warnings are produced and how they will affect it? thanks so much！
In addition, My compilation operation is as follows ：
export TEMPLATECONF=meta-intel/meta-s2600sf/conf
. openbmc-env
bitbake obmc-phosphor-iamge

mdmillerii, 2021-03-05:
The OpenBMC project mantains a kernel tree based off of mainline and works to upstream its patches.   The warnings you see are generated from the yocto config system, using additional metadata in their kernel tree.

Some of the above features do require specific config items to be enabled and you will need to check the layers when combined will result in a usable config.  Specifically gpio based input keyboard, polled and or interrupt driven, and vlans in the network layer.

The following is a link to some content in our documentation repository
https://github.com/openbmc/docs/blob/master/kernel-development.md.  There are discussions in our mailing list from when yocto made this change last year.

In general the community prefers to use our email list and irc and discord forums for general questions but some of us still scan at least selected repository github issues.  You will generally find a broader group there.

Tommy-git2033, 2021-03-05:
thanks for your reply!
I will get a further look at the kernel-development.md file. 
By the way ,as i am a newcomer to openbmc ,I would think that I caused you troubles the way i ask question. Could you please tell me more about the mailing list？

mdmillerii, 2021-03-05:
On the front page of the docs repository :-)

https://github.com/openbmc/docs.

Please consider closing this issue.  Thanks.

Tommy-git2033, 2021-03-05:
Ok, thanks so much ,i will close this issue soon.

closed 2021-03-05 by Tommy-git2033