openbmc/openbmc #3769: Use QEMU running romulus, encounter Sensor issue.

Tommy-git2033, 2021-03-05:
Hi,guys, since i do these step and i feel confused!
 1、export TEMPLATECONF=meta-ibm/meta-romulus/conf
 2、. openbmc-env
 3、bitbake obmc-phosphor-image
 4、./qemu-system-arm -m 256 -M romulus-bmc -nographic -drive file=./tmp/deploy/images/romulus/obmc-phosphor-image-romulus.static.mtd,format=raw,if=mtd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu

everything is ok, and then i login the webui: https://localhost:2443  i go to check the sensor info.
I was teached to modify dts file if i want to add sensor, and i check the dts file: aspeed-bmc-opp-romulus.dts
[image]
furthermore, i check the define:
[image]
 look, there define three sensors.but i cannot find them on webui:
[image]
Could someone please help me? and could someone talk about how to add sensor in OpenBMC project？
Sincere regards！

closed 2021-03-05 by Tommy-git2033

bjwyman, 2021-03-05:
There is a wiki page that lists models on the TODO list, things in development, etc. https://github.com/openbmc/qemu/wiki

Bigcaibird, 2021-06-09:
hi  @Tommy-git2033,I also met this problem, have you solved it? Can you tell me how you do it？Thank you!