openbmc/openbmc #3770: Can't locate mrw/Targets.pm

geissonator, 2021-03-06:
If you build a witherspoon with latest master without an sstate cache you'll get this error:
```
| ERROR: Execution of '/usr/local/google/home/gmouse/RPM_work/security/openbmc/witherspoon-size/build/tmp/work/x86_64-linux/phosphor-logging-callouts-mrw-native/1.0-r1/temp/run.do_install.1161842' failed with exit code 2:
| Can't locate mrw/Targets.pm
```

https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/40974 introduced this issue. But really that commit was fixing a valid warning with the latest yocto and appears to have caused a different bug that was relying on the out of order includes.

https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/41051 was an attempt at a quick fix of this issue but it's not working.

If you're in a pinch, revert 40974, delete all sstate, downloads, and your tmp/ build dir and you should be ok.

geissonator, 2021-03-08:
Here's the the type of diff we see between the non-working and working cases:

```
< export PERL_ARCHLIB="/home/andrewg/Code/openbmc/build/witherspoon/tmp/work/x86_64-linux/mrw-perl-tools-native/1.0+gitAUTOINC+721dcbd23d-r1/recipe-sysroot-native/usr/lib/perl5/None/None"
---
> export PERL_ARCHLIB="/home/andrewg/Code/openbmc/build/witherspoon/tmp/work/x86_64-linux/mrw-perl-tools-native/1.0+gitAUTOINC+721dcbd23d-r1/recipe-sysroot-native/usr/lib/perl5/5.32.0/x86_64-linux"
```

Note the "None" in the non-working case for the directory indicating an issue with determining the perl version.

geissonator, 2021-06-04:
The workaround done in https://github.com/openbmc/openbmc/commit/aae820c542c0bbf907fff93e3b0fc865465fad60 fixed this issue but now causes warnings like these:

```
WARNING: /media/home/official/openbmc/meta-ibm/meta-romulus/recipes-phosphor/logging/romulus-phosphor-logging-callouts-native.bb: QA Issue: romulus-phosphor-logging-callouts-native: native/nativesdk class is not inherited last, this can result in unexpected behaviour. Classes inherited after native/nativesdk: phosphor-logging.bbclass [native-last]
WARNING: /media/home/official/openbmc/meta-ibm/meta-romulus/recipes-phosphor/occ/romulus-occ-control-config-native.bb: QA Issue: romulus-occ-control-config-native: native/nativesdk class is not inherited last, this can result in unexpected behaviour. Classes inherited after native/nativesdk: openpower-occ-control.bbclass [native-last]
WARNING: /media/home/official/openbmc/meta-phosphor/recipes-phosphor/mrw/mrw-api-native.bb: QA Issue: mrw-api-native: native/nativesdk class is not inherited last, this can result in unexpected behaviour. Classes inherited after native/nativesdk: mrw-rev.bbclass perl-version.bbclass cpan-base.bbclass perlnative.bbclass [native-last]
```

pefoley2, 2022-04-22:
This is still an issue at head, unfortunatly.

williamspatrick, 2023-04-05:
I sent a fix upstream for Yocto.  That needs to merge first and then we can fix the inherit order.

https://lore.kernel.org/openembedded-core/20230405003850.384813-1-patrick@stwcx.xyz/T/#u