openbmc/openbmc #3771: obmc-phosphor-image failed for ast2600-evb

sandeepajesh, 2021-03-15:
I am trying to build from latest openbmc for ast2600 eval board, but getting error's. Anyone has similar issue ?

Command used:- 
TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2600/conf . openbmc-env

NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.49.2"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "evb-ast2600"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm armv7a vfp vfpv4d16 callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-oe              
meta-networking      
meta-python          
meta-phosphor        
meta-aspeed          
meta-evb-ast2600     = "master:5521d249699db8f0fac09afe0eb9bd6d5996b4f8"

Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 1 Local 0 Network 0 Missed 1 Current 1377 (0% match, 99% complete)
NOTE: Executing Tasks
**ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Image '/home/admin/openbmc/build/tmp/deploy/images/evb-ast2600/u-boot.bin' is too large!**
ERROR: Logfile of failure stored in: /home/admin/openbmc/build/tmp/work/evb_ast2600-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.27180
ERROR: Task (/home/admin/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1

jk-ozlabs, 2021-03-16:
It looks like you're hitting the issue described in this thread:

  https://lists.ozlabs.org/pipermail/openbmc/2021-February/024932.html

Of those two patches, the second has been merged:

 https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/41007

But the first (meta-phosphor) change is still under review:

 https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/39343

If you can pull the latter into your tree, looks like that should fix the issue.

sandeepajesh, 2021-03-16:
Thanks Jeremy. I applied patch locally and it worked.

LeeTroy, 2021-03-26:
The patch has been submitted into gerrit, please help to review.
https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/41640

williamspatrick, 2023-09-25:
I am going to assume the evb builds correctly now.  Please reopen if you still see this.

closed 2023-09-25 by williamspatrick