openbmc/openbmc #3772: [Question] How to control power and reset of custom system

makise-homura, 2021-03-16:
I have AST2500 BMC, and platform that has power and reset control by four GPIO lines:
* `ATX_PWROK`: BMC input, platform output, active-high, push-pull, indicates system power state: 1 = on, 0 = off.
* `CTRL_RDY`: BMC output, platform input, active-high, pull-down, activates a multiplexer that connects other GPIO lines to CPU if both BMC is ready, and system power is on.
* `POWER_BUTTON#`: BMC output, platform input, active-low, pull-up:
should be set to 0 for 1 sec by BMC to power-on the system if `ATX_PWROK` is low;
should be set to 0 for 1 sec by BMC to request a soft power-off of the system if `ATX_PWROK` is high;
should be set to 0 for 5 sec by BMC to hard-power-on the system if `ATX_PWROK` is high.
* `RESET_BUTTON#`: BMC output, platform input, active-low, pull-up, should be set to 0 for 1 sec by BMC to hard-reset the system.

Last 2 signals are paralleled with chassis power and reset buttons, so user may press them independently of BMC, so power on and off the system and reset it without BMC knowing of the fact that these buttons have been pressed. (It is the requirement based on suggestion that system should work without BMC installed, because some of our systems come without BMC, but with pin-to-pin compatible VGA chip on its place.)

I've read [this](https://github.com/openbmc/docs/blob/master/development/add-new-system.md) document, but have no idea what to implement to let OpenBMC power and reset control be completely configured for this system, except for `gpio_defs.json`.

So, my `gpio_defs.json`:
```
{
    "gpio_configs": {
        "power_config": {
            "power_good_in": "ATX_PWROK",
            "power_up_outs": [
                {"name": "CTRL_RDY", "polarity": true}
            ],
            "reset_outs": [
                {"name": "RESET_BUTTON", "polarity": false}
            ],
        }
    },

    "gpio_definitions": [
        {
            "name": "RESET_BUTTON",
            "pin": "P3",
            "direction": "out"
        },
        {
            "name": "ATX_PWROK",
            "pin": "O2",
            "direction": "in"
        },
        {
            "name": "POWER_BUTTON",
            "pin": "N5",
            "direction": "out"
        },
        {
            "name": "CTRL_RDY",
            "pin": "N7",
            "direction": "out"
        }
    ]
}
```

Still I have no success on OpenBMC being able to monitor power state and allow me to power on and off the system using its web interface.

I've also created systemd services with the following contents:
* `host-poweroff-hard.service`:
```
[Unit]
Description=Unconditionally Stop Host
Conflicts=obmc-chassis-poweron@0.target

[Service]
RemainAfterExit=yes
Type=oneshot
ExecStart=/usr/bin/env server_pwrbut_h
SyslogIdentifier=buttonscripts

[Install]
WantedBy=obmc-chassis-hard-poweroff@0.target
```
* `host-poweroff.service`:
```
[Unit]
Description=Stop Host
Conflicts=obmc-chassis-poweron@0.target

[Service]
RemainAfterExit=yes
Type=oneshot
ExecStart=/usr/bin/env server_pwrbut_s
SyslogIdentifier=buttonscripts

[Install]
WantedBy=obmc-chassis-poweroff@0.target
```
* `host-poweron.service`:
```
[Unit]
Description=Start Host
Conflicts=obmc-chassis-poweroff@0.target

[Service]
RemainAfterExit=yes
Type=oneshot
ExecStart=/usr/bin/env server_pwrbut_s
ExecStart=/usr/bin/env server_watchdog_reset
SyslogIdentifier=buttonscripts

[Install]
WantedBy=obmc-chassis-poweron@0.target
```
* `host-reset.service`:
```
[Unit]
Description=Reset Host
Conflicts=obmc-chassis-poweroff@0.target

[Service]
RemainAfterExit=yes
Type=oneshot
ExecStart=/usr/bin/env server_reset
SyslogIdentifier=buttonscripts

[Install]
WantedBy=obmc-host-reset@0.target
```

There, `server_pwrbut_s` activates `POWER_BUTTON` for 1 sec, `server_pwrbut_h` activates `POWER_BUTTON` for 5 sec, and `server_reset` activates `RESET_BUTTON` for 1 sec. `server_watchdog_reset` waits for `ATX_PWROK` for 10 seconds, and once it went up, executes `ipmitool mc watchdog off`. These all are shell scripts.

The question is: what am I doing wrong, and how should I control the power and reset state for my system? I'm sure I'm missing something obvious, but there's no documentation on how to do this correctly I guess.

jk-ozlabs, 2021-03-18:
As you've indicated, this is more of a question, so is probably better for the openbmc mailing list (https://lists.ozlabs.org/listinfo/openbmc), rather than an issue in openbmc. But in the meantime:

You mention "Still I have no success on OpenBMC being able to monitor power state and allow me to power on and off the system using its web interface", but that doesn't indicate what you're seeing. Aside from the power states not changing, what happens? Be as detailed as you can, we like detail :)

A few things to check out, and possibly include as data in your posts to the list:

 * Are you able to control the host power state by running your utility tools directly?
 * and/or can you control the host power state by directly interacting with your BMC output GPIOs, using `gpioset`?
 * Do you see the systemd targets being activated when you trigger the power state transition requests (say, through the web interface)
 * Do you see your systemd services being activated?
 * Do you see any journal output from your utility programs?

wangmin239, 2021-03-18:
Hi makise-homura，I solved a similar issuse via the following steps.
1，add a recipe named ‘obmc-op-control-power' .  the codes are shown on the webpage 'https://github.com/openbmc/skeleton'.
2，check if a process named 'power_control.exe' is running when openBmc runs on a BMC machine.
3，type the command 'journalctl -u power_control ' to print the 'power_control.exe'  process's log.

makise-homura, 2021-03-18:
@jk-ozlabs Thank you for help! I'm not so used to mailing lists, but if so, I'll try to get help there if the discussion here won't help me.
Well, still I'll try to clarify what I have:

> Are you able to control the host power state by running your utility tools directly?

Yes. More than that, I can control power state by directly starting (and then immediately stopping) corresponding systemd service.  E.g. I can use `systemctl start host-poweron && systemctl stop host-poweron` to power host on, `systemctl start host-poweroff-hard && systemctl stop host-poweroff-hard` to power it off unconditionally, and `systemctl start host-reset && systemctl stop host-reset` to perform an unconditional reset.

> can you control the host power state by directly interacting with your BMC output GPIOs, using gpioset?

Yes, too. Hardware is ok, there is something inside the systemd internals.

> Do you see the systemd targets being activated when you trigger the power state transition requests (say, through the web interface)

Yes, but it is a bit obscure (see below).

> Do you see your systemd services being activated?

No. That's the core problem I guess.

> Do you see any journal output from your utility programs?

Originally I've had no debug `echo`es in my scripts. Just added it there (using `systemd-cat`), and now each script being called writes 'Called' into the journal. So, I went to web interface and pressed "Power on" button, and got the following output:

```
мар 18 16:00:21 daiyousei phosphor-host-state-manager[387]: Host state transition request of xyz.openbmc_project.State.Host.Transition.On
мар 18 16:00:21 daiyousei systemd[1]: unit_file_find_fragment: obmc-host-start@.target+0 → obmc-host-start@0.target
мар 18 16:00:22 daiyousei systemd[1]: Created slice system-obmc\x2denable\x2dhost\x2dwatchdog.slice.
мар 18 16:00:22 daiyousei systemd[1]: Created slice system-phosphor\x2dreset\x2dhost\x2dreboot\x2dattempts.slice.
мар 18 16:00:22 daiyousei systemd[1]: Created slice system-phosphor\x2dset\x2dhost\x2dtransition\x2dto\x2drunning.slice.
мар 18 16:00:22 daiyousei systemd[1]: Created slice system-phosphor\x2dwatchdog.slice.
мар 18 16:00:22 daiyousei systemd[1]: Starting Wait for /xyz/openbmc_project/watchdog/host0...
мар 18 16:00:22 daiyousei systemd[1]: Starting Reset host reboot counter...
мар 18 16:00:22 daiyousei systemd[1]: Starting Set host state to transition to running...
мар 18 16:00:22 daiyousei systemd[1]: Started Phosphor poweron watchdog.
мар 18 16:00:23 daiyousei phosphor-host-state-manager[387]: Change to Host State: xyz.openbmc_project.State.Host.HostState.TransitioningToRunning
мар 18 16:00:23 daiyousei systemd[1]: phosphor-reset-host-reboot-attempts@0.service: Succeeded.
мар 18 16:00:23 daiyousei systemd[1]: Finished Reset host reboot counter.
мар 18 16:00:23 daiyousei systemd[1]: phosphor-set-host-transition-to-running@0.service: Succeeded.
мар 18 16:00:23 daiyousei systemd[1]: Finished Set host state to transition to running.
мар 18 16:00:23 daiyousei systemd[1]: Reached target Start Host0 (Pre).
мар 18 16:00:23 daiyousei systemd[1]: Reached target Host0 (Starting).
мар 18 16:00:23 daiyousei systemd[1]: Reached target Host0 (Started).
мар 18 16:00:23 daiyousei phosphor-watchdog[12547]: Action Targets:
мар 18 16:00:23 daiyousei phosphor-watchdog[12547]:   xyz.openbmc_project.State.Watchdog.Action.PowerCycle -> obmc-host-timeout@0.target
мар 18 16:00:23 daiyousei phosphor-watchdog[12547]:   xyz.openbmc_project.State.Watchdog.Action.PowerOff -> obmc-host-timeout@0.target
мар 18 16:00:23 daiyousei phosphor-watchdog[12547]:   xyz.openbmc_project.State.Watchdog.Action.HardReset -> obmc-host-timeout@0.target
мар 18 16:00:23 daiyousei systemd[1]: Finished Wait for /xyz/openbmc_project/watchdog/host0.
мар 18 16:00:23 daiyousei systemd[1]: Starting Start Watchdog0...
мар 18 16:00:24 daiyousei phosphor-watchdog[12547]: watchdog: enabled and started
мар 18 16:00:24 daiyousei systemd[1]: Finished Start Watchdog0.
мар 18 16:00:24 daiyousei systemd[1]: Reached target Start Host0 Minimum.
мар 18 16:00:24 daiyousei systemd[1]: Reached target Start Host0.
мар 18 16:00:24 daiyousei phosphor-host-state-manager[387]: Received signal that host is running
мар 18 16:00:24 daiyousei phosphor-host-state-manager[387]: Change to Host State: xyz.openbmc_project.State.Host.HostState.Running
```
Then, no power on occured, and watchdog is counting until it expires, and then I see:
```
мар 18 16:00:54 daiyousei phosphor-watchdog[12547]: watchdog: Timed out
мар 18 16:00:54 daiyousei systemd[1]: unit_file_find_fragment: obmc-host-timeout@.target+0 → obmc-host-timeout@0.target
мар 18 16:00:54 daiyousei phosphor-watchdog[12547]: watchdog: disabled
мар 18 16:00:54 daiyousei systemd[1]: Starting Reset host sensors...
мар 18 16:00:54 daiyousei systemd[1]: Stopping Phosphor poweron watchdog...
мар 18 16:00:54 daiyousei systemd[1]: phosphor-watchdog@poweron.service: Succeeded.
мар 18 16:00:54 daiyousei systemd[1]: Stopped Phosphor poweron watchdog.
мар 18 16:00:55 daiyousei systemd[1]: phosphor-reset-sensor-states@0.service: Succeeded.
мар 18 16:00:55 daiyousei systemd[1]: Finished Reset host sensors.
мар 18 16:00:55 daiyousei systemd[1]: Reached target Quiesce Target.
мар 18 16:00:55 daiyousei systemd[1]: Reached target Host instance 0 timed out refreshing Watchdog.
мар 18 16:00:55 daiyousei phosphor-host-state-manager[387]: Auto reboot enabled, rebooting
[... cut ...]
мар 18 16:00:59 daiyousei systemd[1]: Finished Reset host sensors.
```
As you see, once I try to power on the system, no scripts are called, and neither `host-poweron.service` is started. But, when watchdog expires, and host is going to power off, something is calling `server_pwrbut_s`. It is very strange, because the only services that may call it, are `host-poweroff.service` and `host-poweron.service`, but I see neither of them starting in systemd journal. Instead, I see `host-reset.service` started, but no `server_reset` is called.

makise-homura, 2021-03-18:
@wangmin239 Thank you for suggestion! It looks like `obmc-op-control-power` is the actual thing that uses `gpio_defs.json` and activates power-related GPIOs. I built an image with it, and now it looks like I've had the actual power states (but still had to take out buttons from JSON, because otherwise `obmc-op-control-power` blocks these GPIOs by setting them to output instead of active-low, impulse, pull-up). So, I'm gonna see how I should deal with operating a power button in `on_set_power_state` (it looks like there's a few code changes required). At least it seems I'm on the right way. Thanks!

mdmillerii, 2021-03-18:
The default is to start the watchdog at power on and expect bios to stop it via ipmi.  If you don't have a host ipmi stack then you will need to disable it.  It has come up before but I don't have a link handy.

mdmillerii, 2021-03-18:
I believe some of the x86 systems use ata power supplies.  OpenPOWER systems use a direct gpio state is requested power state via a sequencer (microcontroller or fpga).

wangmin239, 2021-03-20:
@makise-homura, you may need to change some code snippets, I also applied a patch to make the power control work. Good luck.

makise-homura, 2021-03-25:
Thank you everyone for help! I ended up in investigating under @wangmin239 suggestion, and found out that I have to do the following:
* implement my own daemon in place of `obmc-op-control-power` that only polls PGOOD and don't mess with any buttons at all (original phosphor's power control tends to tie buttons to 0 continuously instead of short 1- of 5-sec impulse);
* made this daemon (`reimu-power-control`) setting `xyz.openbmc_project.State.Host` property `СurrentHostState` to `Off` or `Running` according to current PGOOD state, if it was changed from the last polling iteration;
* edit `phosphor-dbus-interfaces` to allow the following host transitions: `HardOff`, `Off`, `On`, `Reboot`;
* edit `phosphor-webui` to make only one reboot type possible, and send only the specified transitions while pressing `Power On`, `Reboot`, and `Power Off` buttons; and in the last case, send `HardOff` or `Off` transitions for immediate or orderly power off request;
* edit my service files not allowing them to `RemainAfterExit` (so the same service may be started again if we wish; like we can reboot a several times);
* edit `phosphor-state-manager` to run my systemd services instead of trying to isolate to targets (e.g., change `obmc-host-reboot@0.target` to `host-reset.service`, `obmc-host-start@0.target` to `host-poweron.service`, and so on, and remove complex logic that distinguised between different targets) -- so now `phosphor-state-manager` just asks systemd to start a specified service on host transition to some state, and performing nothing else;
And now nearly everything works!

The only thing that does not work now, is the notifications for `phosphor-webui`: I expect, that once power state is changed, something should perform a certain DBus call or something else, and this will end up in `phosphor-webui` being notified that power state is changed, and it will auto-refresh its power control page (for now, I should perform a manual refresh after clicking any button on power control page). Would please anyone suggest me where to look for it?

makise-homura, 2021-03-31:
Ok, seems like there is some issue with network, because I can't spot this failure with direct connection. Just added a warning to be sure and left it as it is.
And as I know now, `phosphor-webui` is deprecated in favor of `webui-vue`, so I'd plan to switch to the latter.
I guess nothing left here, so closing the issue.

closed 2021-03-31 by makise-homura

wijdim, 2022-11-07:
@williamspatrick so better to use phosphor-power ? my goal is just to Handle the GPIO-pin of PGOOD and provide it in an interface so it can be readable by the PMS... By changing the power_control.cpp I still cannot have any service running,
what do we have to use unstead of Skeleton? any idea?

spinler, 2022-11-07:
On our 'p10bmc' system in meta-ibm we use phosphor-power.  Hopefully you can follow that.

To start though, how about you just do a bitbake phosphor-power, copy on the executable, and make sure it does what you need?

wijdim, 2022-11-08:
@spinler  after modifying the phosphor-power with devtool, and a bitbake phosphor-power I get the executable pseq-monitor in my local workspace not in the image that I run after with qemu. But anw, I copied the binary in my image then after running the exe : 
./pseq-monitor --action runtime-monitor 5 
./pseq-monitor --action pgood-monitor 1000
 I get always (Invalid interval value). 
In my understanding, after compiling i expect to have a new dbus interface org.openbmc.control.Power including the pgood value given via GPIO pin. 
In your 'p10bmc' system in meta-ibm you added a psu.json file any documentation about it?