openbmc/openbmc #3773: Root user is rejected to login an aarch64 openBmc system

wangmin239, 2021-03-23:
Hi all,
At present,  an aarch64 root file system was build and  ran on an aarch64 soc. However, root is rejected to login. It is skeptical that there are some wrong configurations in the file “/etc/fstab”.  Anyone has ideas on the issue, please give your comments.  Thanks in advance. 
BTY, the root file system was compressed by mksquash.

The following text snippets are the contents of the file “/etc/fstab”.

**# stock fstab - you probably want to override this with a machine specific one

 

/dev/root            /                    auto       defaults              1  1

proc                 /proc                proc       defaults              0  0

devpts               /dev/pts             devpts     mode=0620,ptmxmode=0666,gid=5      0  0

tmpfs                /run                 tmpfs      mode=0755,nodev,nosuid,strictatime 0  0

tmpfs                /var/volatile        tmpfs      defaults              0  0

 

# uncomment this if your device has a SD/MMC/Transflash slot

#/dev/mmcblk0p1       /media/card          auto       defaults,sync,noauto  0  0**

The startup log is shown the below text lines.

- 
- [    2.044027][ 3] VFS: Mounted root (squashfs filesystem) readonly on device 1:0.
- [    2.051540][ 3] devtmpfs: mounted
- [    2.055297][ 3] Freeing unused kernel memory: 1344K
- [    2.060279][ 3] Run /sbin/init as init process
- [    2.126531][ 3] systemd[1]: systemd 244.3+ running in system mode. (+PAM -AUDIT -SELINUX -IMA -APPAR)
- [    2.148585][ 3] systemd[1]: Detected architecture arm64.
- 
- Welcome to Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0!
- 
- [    2.170085][ 3] systemd[1]: Set hostname to <ft2500>.
- [    2.177862][ 3] random: systemd: uninitialized urandom read (16 bytes read)
- [    2.184835][ 3] systemd[1]: Initializing machine ID from random generator.
- [    2.191748][ 3] systemd[1]: Installed transient /etc/machine-id file.
- [    2.250899][ 3] random: systemd: uninitialized urandom read (16 bytes read)
- [    2.257923][ 3] systemd[1]: system-getty.slice: unit configures an IP firewall, but the local system.
- [    2.270618][ 3] systemd[1]: (This warning is only shown for the first unit using IP firewalling.)
- [    2.280218][ 3] systemd[1]: Created slice system-getty.slice.
- [  OK  ] Created slice system-getty.slice.
- [    2.302077][ 3] random: systemd: uninitialized urandom read (16 bytes read)
- [    2.309286][ 3] systemd[1]: Created slice system-serial\x2dgetty.slice.
- [  OK  ] Created slice system-serial\x2dgetty.slice.
- [    2.330128][ 3] systemd[1]: Started Dispatch Password Requests to Console Directory Watch.
- [  OK  ] Started Dispatch Password âÀ¦ts to Console Directory Watch.
- [    2.354103][ 3] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
- [  OK  ] Started Forward Password Râ…uests to Wall Directory Watch.
- [    2.378081][ 3] systemd[1]: Reached target Paths.
- [  OK  ] Reached target Paths.
- [    2.394068][ 3] systemd[1]: Reached target Remote File Systems.
- [  OK  ] Reached target Remote File Systems.
- [    2.414068][ 3] systemd[1]: Reached target Slices.
- [  OK  ] Reached target Slices.
- [    2.430071][ 3] systemd[1]: Reached target Swap.
- [  OK  ] Reached target Swap.
- [    2.446293][ 3] systemd[1]: Listening on Syslog Socket.
- [  OK  ] Listening on Syslog Socket.
- [    2.467220][ 3] systemd[1]: Listening on Process Core Dump Socket.
- [  OK  ] Listening on Process Core Dump Socket.
- [    2.486120][ 3] systemd[1]: Listening on initctl Compatibility Named Pipe.
- [  OK  ] Listening on initctl[    2.494822][ 0] usb 2-2: new SuperSpeed Gen 1 USB device number 2 usingd
-  Compatibility Named Pipe.
- [    2.514215][ 3] systemd[1]: Listening on Journal Audit Socket.
[... 496 lines cut ...]
- [   17.398133][24] audit: type=1701 audit(1616511064.752:11): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=3560 comm="phosphor-ldap-c" exe="/usr/bin/phosphor-ldap-conf" sig=6 res=1
- 
- Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 ft2500 ttyS0
- 
- ft2500 login: 
- ft2500 login: root
- 
- Login incorrect
- ft2500 login: phytium
- Password: 
- 
- Login incorrect
- ft2500 login: 
- Log
- Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 ft2500 ttyS0
- 
- ft2500 login: root
- 
- Login incorrect
- ft2500 login: phytium
- Password: 
- 
- Login incorrect
- ft2500 login: phytium
- Password:

wangmin239, 2021-03-26:
The cause of the issue was found out.

amboar, 2021-03-26:
@wangmin239 can you please explain what the cause was so if anyone else finds this issue they know how to fix it?

wangmin239, 2021-04-19:
@amboar the cause of the openBmc runs in an emergency mode
was found. The openBmc rootfs was built by a not root user, so all the files in the rootfs are not assigned root privilege.
The below text lines are printed while the system is starting to run as emergency mode.

[FAILED] Failed to mount Huge Pages File System.
See 'systemctl status dev-hugepages.mount' for details.
[FAILED] Failed to mount POSIX Message Queue File System.
See 'systemctl status dev-mqueue.mount' for details.
[FAILED] Failed to mount Kernel Debug File System.
See 'systemctl status sys-kernel-debug.mount' for details.
[FAILED] Failed to mount Temporary Directory (/tmp).
See 'systemctl status tmp.mount' for details.

After setting the rootfs's owner to root, the openBMC is able to run on an aarch64 soc.

williamspatrick, 2021-04-19:
Do you know which files specifically were not owned by root?  The resulting permissions that come out of a bitbake image are mostly handled by bitbake itself, but if we knew which files you are seeing wrong we could take a look at those recipes and see if there is something wrong with them.

Have you successfully built an arm32 image without manual modification?  It is just aarch64 images which have this issue to you?

mdmillerii, 2021-04-19:
Actually that looks like a generic base file system without the phosphor customization of `/etc/fstab` and similar, or the initrd to mount the empty but writable layer was skipped.   For instance.

All 3 choices (`static`, `ubi` and `mmc`) end up with a writable file system overlay on `/etc`, with static layout having it over the whole file system for legacy reasons.

wangmin239, 2021-04-19:
@williamspatrick  Hi, The issue was only triggered in an aarch64 images. However,  there were some recipes modificated to successfuly compile an aarch64 images.  So the issue may be introduced due to my changing code snippets.  It may take some times to dig out the root cause of the issue.

mdmillerii, 2021-04-19:
I can point out the scripts that mount the overlay.  Are you trying to boot from mtd or block device like mmc? (Or UBI on MTD)

wangmin239, 2021-04-19:
@mdmillerii  I guess you view is close to correctly  answer the issue. The aarch64 image wasn't boot from mtd, it was directly copy into main memory. and the feature of the overlayfs  isn't applied. So I wander why aarch32 image using overlayfs is able to successfully boot, thanks a lot.

mdmillerii, 2021-04-19:
We mount a writable fs (`jffs2` for MTD, `ubifs` for UBI on MTD,  `ext4` for mmc) and then mount a subdirectory of the writable over `/` or `/etc` .

However the initfs scripts can also take the `squashfs` and mount a `tmpfs` (or `ramfs`).

https://github.com/openbmc/openbmc/blob/master/meta-phosphor/recipes-phosphor/initrdscripts/obmc-phosphor-initfs.bb

The `init` script In `phosphor-initfs` can optionally download the squashfs or have the squashfs embedded for bringup.  The default BusyBox doesn't include up configuration but you can use `ip=` on the kernel command line to support that.

Another option is to use a writable fs format such as `ext4` for the base root file system or even format it as a `cpio` for initramfs.

wangmin239, 2021-04-19:
@mdmillerii Thanks your clarification.

closed 2021-04-21 by wangmin239