openbmc/openbmc #3776: Unable to create FRU details in AMD daytonax platform.

jdhanasekar, 2021-04-05:
Hi All,

I have created new system and name it as ```meta-daytonax```. I am working on the enable FRU details. 
I have adding the below code in ```phosphor-ipmi-fru_%.bbappend```, 

```inherit obmc-phosphor-systemd
DEPENDS_append_daytonax = " daytonax-yaml-config"

FILESEXTRAPATHS_prepend_daytonax := "${THISDIR}/${PN}:"

EEPROM_NAMES = "motherboard"

EEPROMFMT = "system/chassis/{0}"
EEPROM_ESCAPEDFMT = "system-chassis-{0}"
EEPROMS = "${@compose_list(d, 'EEPROMFMT', 'EEPROM_NAMES')}"
EEPROMS_ESCAPED = "${@compose_list(d, 'EEPROM_ESCAPEDFMT', 'EEPROM_NAMES')}"

ENVFMT = "obmc/eeproms/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN}_append_gbs := " ${@compose_list(d, 'ENVFMT', 'EEPROMS')}"

TMPL = "obmc-read-eeprom@.service"
TGT = "multi-user.target"
INSTFMT = "obmc-read-eeprom@{0}.service"
FMT = "../${TMPL}:${TGT}.wants/${INSTFMT}"

SYSTEMD_LINK_${PN}_append_gbs := " ${@compose_list(d, 'FMT', 'EEPROMS_ESCAPED')}"

EXTRA_OECONF_append_daytonax = " \
    YAML_GEN=${STAGING_DIR_HOST}${datadir}/daytonax-yaml-config/ipmi-fru-read.yaml \
    PROP_YAML=${STAGING_DIR_HOST}${datadir}/daytonax-yaml-config/ipmi-extra-properties.yaml \
    "
```
After adding and built the source code, I am getting below error,

```
ERROR: phosphor-ipmi-fru-1.0+gitAUTOINC+0968237b47-r1 do_compile: oe_runmake failed
ERROR: phosphor-ipmi-fru-1.0+gitAUTOINC+0968237b47-r1 do_compile: Execution of '/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/temp/run.do_compile.6422' failed with exit code 1:
/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/bin/python3-native/python3 ../git/scripts/fru_gen.py -i /home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot/usr/share/daytonax-yaml-config/ipmi-fru-read.yaml -o . generate-cpp
/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/bin/python3-native/python3 ../git/scripts/extra-properties.py -e /home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot/usr/share/daytonax-yaml-config/ipmi-extra-properties.yaml
Traceback (most recent call last):
  File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/build/../git/scripts/fru_gen.py", line 59, in <module>
    main()
  File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/build/../git/scripts/fru_gen.py", line 55, in main
    function(args.inventory_yaml, args.outputdir)
  File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/build/../git/scripts/fru_gen.py", line 24, in generate_cpp
    fd.write(t.render(fruDict=ifile))
  File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/lib/python3.9/site-packages/mako/template.py", line 476, in render
    return runtime._render(self, self.callable_, args, data)
  File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/lib/python3.9/site-packages/mako/runtime.py", line 878, in _render
    _render_context(
  File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/lib/python3.9/site-packages/mako/runtime.py", line 920, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/lib/python3.9/site-packages/mako/runtime.py", line 947, in _exec_template
    callable_(context, *args, **kwargs)
  File "_home_user_dhanasekar_BMC_AMD_daytonax_build_daytonax_tmp_work_arm1176jzs_openbmc_linux_gnueabi_phosphor_ipmi_fru_1_0_gitAUTOINC_0968237b47_r1_git_scripts_writefru_mako_cpp", line 36, in render_body
TypeError: 'NoneType' object is not subscriptable
Makefile:1132: recipe for target 'fru-gen.cpp' failed
make: *** [fru-gen.cpp] Error 1
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/temp/log.do_compile.6422
Log data follows:
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 32
| /home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/bin/python3-native/python3 ../git/scripts/fru_gen.py -i /home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot/usr/share/daytonax-yaml-config/ipmi-fru-read.yaml -o . generate-cpp
| /home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/bin/python3-native/python3 ../git/scripts/extra-properties.py -e /home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot/usr/share/daytonax-yaml-config/ipmi-extra-properties.yaml
| Traceback (most recent call last):
[... 25 lines cut ...]
|     main()
|   File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/build/../git/scripts/fru_gen.py", line 55, in main
|     function(args.inventory_yaml, args.outputdir)
|   File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/build/../git/scripts/fru_gen.py", line 24, in generate_cpp
|     fd.write(t.render(fruDict=ifile))
|   File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/lib/python3.9/site-packages/mako/template.py", line 476, in render
|     return runtime._render(self, self.callable_, args, data)
|   File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/lib/python3.9/site-packages/mako/runtime.py", line 878, in _render
|     _render_context(
|   File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/lib/python3.9/site-packages/mako/runtime.py", line 920, in _render_context
|     _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
|   File "/home/user/dhanasekar/BMC/AMD/daytonax/build/daytonax/tmp/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0+gitAUTOINC+0968237b47-r1/recipe-sysroot-native/usr/lib/python3.9/site-packages/mako/runtime.py", line 947, in _exec_template
|     callable_(context, *args, **kwargs)
|   File "_home_user_dhanasekar_BMC_AMD_daytonax_build_daytonax_tmp_work_arm1176jzs_openbmc_linux_gnueabi_phosphor_ipmi_fru_1_0_gitAUTOINC_0968237b47_r1_git_scripts_writefru_mako_cpp", line 36, in render_body
| TypeError: 'NoneType' object is not subscriptable
| Makefile:1132: recipe for target 'fru-gen.cpp' failed
| make: *** [fru-gen.cpp] Error 1
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/user/dhanasekar/BMC/AMD/daytonax/meta-phosphor/recipes-phosphor/ipmi/phosphor-ipmi-fru_git.bb:do_compile) failed with exit code '1'
```

FYI,In this line ```EXTRA_OECONF_append_daytonax = " \```. If I change ```daytonax``` to ```ethanolx```. I am not getting error. 

Please help me to solve this issue.

jdhanasekar, 2021-04-09:
Solved this issue. Closing it.

jk-ozlabs, 2021-04-12:
@jdhanasekar can you include some details on how you solved this, so the next person seeing this error can fix it?

jdhanasekar, 2021-04-14:
Actually, I have removed fru contents from ```daytonax-ipmi-fru.yml``` which are not needed for my board. 
My board has two fru , I added those contents only in .yml file,,

```0:
  /system/chassis/bmc:
    entityID: 7
    entityInstance: 1
    interfaces:
      xyz.openbmc_project.Inventory.Decorator.Asset:
        BuildDate:
          IPMIFruProperty: Mfg Date
          IPMIFruSection: Board
        SerialNumber:
          IPMIFruProperty: Serial Number
          IPMIFruSection: Board
        PartNumber:
          IPMIFruProperty: Part Number
          IPMIFruSection: Board
        Manufacturer:
          IPMIFruProperty: Manufacturer
          IPMIFruSection: Board
      xyz.openbmc_project.Inventory.Item:
        PrettyName:
          IPMIFruProperty: Name
          IPMIFruSection: Board
      xyz.openbmc_project.Inventory.Decorator.Revision:
        Version:
          IPMIFruProperty: FRU File ID
          IPMIFruSection: Board
1:
  /system/chassis/bmc:
    entityID: 0x0c
    entityInstance: 1
    interfaces:
      xyz.openbmc_project.Inventory.Decorator.Asset:
        BuildDate:
          IPMIFruProperty: Mfg Date
          IPMIFruSection: Board
        SerialNumber:
          IPMIFruProperty: Serial Number
          IPMIFruSection: Board
        PartNumber:
          IPMIFruProperty: Part Number
          IPMIFruSection: Board
        Manufacturer:
          IPMIFruProperty: Manufacturer
          IPMIFruSection: Board
      xyz.openbmc_project.Inventory.Item:
        PrettyName:
          IPMIFruProperty: Name
          IPMIFruSection: Board
      xyz.openbmc_project.Inventory.Decorator.Revision:
        Version:
          IPMIFruProperty: FRU File ID
          IPMIFruSection: Board
```

closed 2021-04-14 by jdhanasekar