openbmc/openbmc #3777: [Question] How to measure the boot time

MichaelJeiShen, 2021-04-07:
I am trying to measure the bootup timing as detailed as possible.
For example how long it takes on bootloader/uboot/kernel/systemd etc.
Does openBMC has a tool for doing that?

mdmillerii, 2021-04-07:
Systemd has bootchart support showing dependencies and recording the time.

The aspeed hardware has a free running us scale counter but I am not aware of any u-boot timing.  I have not kept up with u-boot development and standard time collection.

If you are looking to optimize the boot time there are tradeoffs for boot flow.   We'd need a description of your storage (mmc, flash, which bmc doc).

MichaelJeiShen, 2021-04-08:
> Systemd has bootchart support showing dependencies and recording the time.
Nice, then how do we obtain these times?
I just found a tool called `systemd-analyze` but not sure it is the correct tool.

> The aspeed hardware has a free running us scale counter but I am not aware of any u-boot timing. I have not kept up with u-boot development and standard time collection.
U-boot boot time is a good-to-have for me. It’s not necessary.

> If you are looking to optimize the boot time there are tradeoffs for boot flow. We'd need a description of your storage (mmc, flash, which bmc doc).
I just want to measure the boot time. Optimization is not included.

geissonator, 2021-04-09:
> Nice, then how do we obtain these times?
> I just found a tool called systemd-analyze but not sure it is the correct tool.

https://github.com/openbmc/openbmc/wiki/Performance-Profiling-in-OpenBMC has some info on this. I did this back a few years ago and got some good data about BMC boot time and where we were spending the most time.