openbmc/openbmc #3780: Can openbmc support ikvm for ast2400 soc?

gooaler, 2021-04-20:
The IP-KVM is not working on master, that it shows a blank page on OpenBMC WebUI. 
The video0 device node is not generated under the system.

closed 2022-05-24 by edtanous

leonli2002, 2025-03-03:
> The IP-KVM is not working on master, that it shows a blank page on OpenBMC WebUI. The video0 device node is not generated under the system.

It still does not work for ast2400 , no video0 node in /dev

mdmillerii, 2025-03-12:
Make sure your device tree [enables](https://github.com/openbmc/linux/blob/0f9b4c3ca5fdf3e177266ef994071b1a03f07318/arch/arm/boot/dts/aspeed/aspeed-g4.dtsi#L249) the hardware, includes a [buffer](https://github.com/openbmc/linux/blob/0f9b4c3ca5fdf3e177266ef994071b1a03f07318/Documentation/devicetree/bindings/media/aspeed-video.txt#L19) , and the driver is [enabled](https://github.com/openbmc/linux/blob/0f9b4c3ca5fdf3e177266ef994071b1a03f07318/drivers/media/platform/aspeed/Kconfig#L5)