openbmc/openbmc #3781: New bug

recha0, 2021-04-21:
When
export TEMPLATECONF=meta-evb/meta-evb-enclustra/meta-evb-zx3-pm3/conf

time bitbake obmc-phosphor-image   . openbmc-env

there is a bug:
ERROR: Layer xilinx is not compatible with the core layer which only supports these series: hardknott (layer is compatible with gatesgarth dunfell)

229024527, 2021-04-22:
I've met the question as you too.

williamspatrick, 2021-04-22:
I don't know anyone active on the project that regularly uses `meta-evb-enclustra`, so it is maintained at a best effort and doesn't have any CI behind it.

This layer is picking up another meta layer from elsewhere ("Layer xilinx is not compatible") and that layer doesn't work with the latest Yocto revision, it appears.  This should be reported to https://github.com/Xilinx/meta-xilinx .  Once they have a fix that supports hardknott, we can sync their subtree here.

verilogzhou, 2021-07-31:
Xilinx已有支持Hardnott的补丁，麻烦更新一下。

williamspatrick, 2021-08-03:
I've emailed Xilinx to get some clarification on a few items.  github and yocto are now diverged for meta-xilinx and I'm not sure which is official.

williamspatrick, 2021-08-04:
>> 1. I see that https://github.com/Xilinx/meta-xilinx is now different from                                                    
>>     http://git.yoctoproject.org/cgit/cgit.cgi/meta-xilinx/.  Is that intentional?                                            
>>     Which one should we be pointing to?                                                                                      

> It is temporary.  We're working on the Xilinx version first, once it's working then we merge it to git.yoctoproject.org.
> The github version will also contain petalinux branches that won't appear on the git.yoctoproject.org.

I got this from one of the Xilinx maintainers.  While they have hardknott supported in their github repository, that's not what we are pointing at as we tend to only pick up Yocto meta-layers.  Once they've completed this work we can pick up the updated tree from git.yoctoproject.org.

recha0, 2021-08-05:
Thinks to much for your replay! I understand. Now I use the history version of openbmc/meta-xilinx.

edtanous, 2022-05-23:
The xilinx layer was removed due to lack of maintainers keeping it up to date, so this bug no longer exists.  If the machines are re-added, they will need to compile against current yocto.

Closing.

closed 2022-05-23 by edtanous