openbmc/openbmc #3784: How to realize virtual media

kingzmm, 2021-04-27:
I want to implement the virtual media function of openbmc, who can help me to understand the process of virtual media implementation

mdmillerii, 2021-04-27:
The usb gadget is setup here https://github.com/openbmc/phosphor-misc/tree/master/usb-ctrl

Please test the in-progress updates

kingzmm, 2021-04-29:
@mdmillerii：Thank you very much for your answer. I have encountered a new problem now. I tried to use virtual media to transfer ordinary files and failed. I can only transfer ISO files. Do you have any good attention to help me solve this problem?

mdmillerii, 2021-04-29:
The current code just supplies an image to the host.   However there is code in the script to create a vfat volume with a defined volume ID and label, and mount that on the bmc and also unmount and delete it.  After unmount you can also share the volume as a HDD, optionally read-only or removable drive like a flash stick.

**Do not try to mount when exposed to the host** you will likely crash both the host os and the bmc kernel.

I don't know the history of which vendor(s) use this feature.

kingzmm, 2021-04-29:
@mdmillerii: I don’t understand what you are talking about. Can you give me an example? thank you very much!

mdmillerii, 2021-04-29:
Actually, no, because I don't see a command to unmount the image from the bmc without erasing  the file, which tells me the user is not following my advice.  It might
work for usb-ro exposed to the host but its risky at best.

The file is in phosphor-misc/usb_ctrl and there are patches pending in gerrit
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-misc/+/42670
please take a read though the script (usage help is near the bottom).

jk-ozlabs, 2021-04-30:
@kingzmm the virtual media function exposes a local file as a USB mass-storage device to your host.

So, whatever file type you share using the virtual media function, it's up to your host operating system to interpret that data as a standard USB-connected filesystem.

For example, if you try to share a pdf, this would be equivalent to writing the raw PDF data to a USB drive, rather than storing it in a filesystem on that USB drive.

jk-ozlabs, 2021-04-30:
Ah! but it looks like support for generating the filesystem has been added since I last looked at this code. @leiyu-bytedance , can you provide some help for how to use this?

leiyu-bytedance, 2021-04-30:
> Ah! but it looks like support for generating the filesystem has been added since I last looked at this code. @leiyu-bytedance , can you provide some help for how to use this?

Nope, I did not know any support for generating the filesystem. For now, the scripts only support mounting a file (or a device) to the host.

wangmin239, 2022-01-06:
@kingzmm，  I encountered a problem when  ISO files were transfered to a host OS by virtual media. There was no ISO file to be showed on the host OS.  using 'lsusb' is able to find the mass storage device which is created by the host OS.  Do I miss somethings.

edtanous, 2022-05-23:
This was discussed on the mailing list, and I believe the question originally asked has been answered.  Please reopen if this is still a concern.

closed 2022-05-23 by edtanous