openbmc/openbmc #3786: Nuvoton WPCM450 BMC

slavonnet, 2021-04-30:
in kerknel add support for 
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/log/?h=next-20210430&qt=grep&q=WPCM450

Many Supermicrro MB target X8, X9, X10  use this BMC
Also i have IIntel S2600GL4 platworms that also have WPCM450  BMC

If any one create simple meta for this BMC i can test IPMI firmware

From additional we use mlx4 and sas3008

amboar, 2021-04-30:
Hi @slavonnet, given this isn't an explicit list of things that need to be done to support the WPCM450 nor a bug in OpenBMC with respect to its WPCM450 support (there can't be bugs as we don't yet support it), your offer to test any work in this area would probably be best as a post to the mailing list. I expect Jonathan Neuschäfer would appreciate others helping him out so make sure to add him on Cc to any mail you send: j.neuschaefer@gmx.net

slavonnet, 2021-04-30:
> Hi @slavonnet, given this isn't an explicit list of things that need to be done to support the WPCM450 nor a bug in OpenBMC with respect to its WPCM450 support (there can't be bugs as we don't yet support any platforms with it), your offer to test any work in this area would probably be best as a post to the mailing list. I expect Jonathan Neuschäfer would appreciate others helping him out so make sure to add him on Cc to any mail you send.

I hate BBS, Lists and other forum like disscussions.  I need inital fiorst success build and run.. After first tree i will add other modules and parts;  Some help neeed for first run only.

This package have WinBond sdk. Uvoot part have patches what lokk to have all definea tro i2c and othher subsystems:
[WinBoard SDK example ](https://www.supermicro.com/wdl/GPL/AMI/SMC_winbond_opensrc_10090602.tar.gz)

Full Web package from Sup[[erMicro (last release is 3.54). May be look where for detalis BIOS and Firmware update (and block MCE) logic:
[Real Image for most X9 supermicro MBs](https://www.supermicro.com/wdl/GPL/SMT/SDK_SMT_X9_317.tar.gz)

slavonnet, 2021-04-30:
@neuschaefer Is there a minute here too?

neuschaefer, 2021-04-30:
Hello @slavonnet,

Joel Stanley applied my initial WPCM450 kernel patchset to the openbmc/dev-5.10, so it's now available through the OpenBMC build system (see https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/42706).

Note, however, that the kernel situation is still limited:

- Mainline and dev-5.10 support only very basic features (UART, timer, interrupts)
- ATEN's, AMI's or Dell's vendor kernels are based on rather old kernel versions (all in the 2.6.15-25 range), which probably won't work as a basis for OpenBMC: I think systemd needs a newer kernel, and some other components might too

On the OpenBMC userspace side, the situation is even more limited, noone has started porting a WPCM450 platform, as far as I'm aware. Volunteers welcome!

neuschaefer, 2021-04-30:
I don't understand your question "@neuschaefer Is there a minute here too?"

g0r3, 2022-01-19:
Just wondering what the state of this is. I got myself an old Supermicro X9 board and I would be very interested to replace the old BMC firmware

neuschaefer, 2022-01-19:
@g0r3 upstream kernel support is slowly coming along, but hasn't yet reached the point where all the essential drivers are there (notably still missing: Ethernet, SPI flash, USB, video capture. The GPIO driver will be merged into 5.18). AFAIK, noone has started contributing an OpenBMC port yet.

EmperorArthur, 2022-02-05:
Thank you @neuschaefer for the update.  Leaving this issue open to track progress versus using the listserv is appreciated.  It's a high ranking search result for "nuvoton wpcm450", and allows people who are casually interested in progress to subscribe only to this particular topic.

stikves, 2023-01-13:
> @g0r3 upstream kernel support is slowly coming along, but hasn't yet reached the point where all the essential drivers are there (notably still missing: Ethernet, SPI flash, USB, video capture. The GPIO driver will be merged into 5.18). AFAIK, noone has started contributing an OpenBMC port yet.

Is there a tracker for these sub items?

neuschaefer, 2023-01-15:
> Is there a tracker for these sub items?

Yes, in the [neuschaefer/wpcm450](https://github.com/neuschaefer/wpcm450/issues) bug tracker, I have an issue for every peripheral/driver, and my aim is to close them once the specific driver is upstreamed.

stikves, 2023-01-16:
What is the best way to try this "at home"? 

It looks like there is no readily available target definition in:
https://github.com/openbmc/openbmc

And, if I can build in image, somehow, shall I just upload it to the BMC, have it reboot, and try to connect it via a serial(?) interface (locally on the server hardware)?

neuschaefer, 2023-01-16:
@stikves using OpenBMC on WPCM450 would still require some porting work, so I don't think it's a good _first_ thing to do. I've used a [Buildroot](https://buildroot.org/) userspace so far (but I don't have it in a reproducible and published shape at the moment).

Installing code on the BMC will doubtless still require some manual work, the exact details of which depend on the preinstalled firmware of your particular board.

Serial port access is something I would highly recommend at this point in time. The debug serial port tends to be fairly easy to find (might even be marked in the silkscreen) on DELL boards, and pretty difficult to find on Supermicro boards of the WPCM450 generation. I've documented debug UART access for some Supermicro boards in [my linux wiki](https://github.com/neuschaefer/linux/wiki).

stikves, 2023-02-05:
Do you mean this document:
https://github.com/neuschaefer/linux/wiki/Supermicro-X8DTT-family#uart--h-and--h-variants

I assume that requires soldering a TTL device to that open part.

I think that would be beyond my current skill set. :)

(Setting a reminder for 3 months myself to see if there is any progress).

Thanks anyway.

neuschaefer, 2023-02-06:
Yes, this is one of the pages I meant. You need a 3.3V (TTL) UART adapter, yes.

EasyNetDev, 2023-04-15:
Hi,

I have x2 MB X9SRi-3F and I'm interested to change the BMC firmware.

I'm going to buy a clamp adapter for my programmer CH341a for reflash MX25L128 IC.
In case we need to test the new firmware.

neuschaefer, 2023-04-15:
Hello @EasyNetDev, I'll be happy to help you with questions around WPCM450 firmware, but please be aware that there isn't currently a ready-made modern firmware for WPCM450 yet, so some firmware development (at the very least, configuring something like buildroot) will be needed. My work on this topic has been progressing somewhat slowly.

EasyNetDev, 2023-04-15:
I'm glad if I can help with anything.
I was able fee days ago to extract all files from the original firmware, but of course is not helping me too much. I was thinking to try to implement a noVNC instead Java client. But I don't think is easy.

williamspatrick, 2023-09-25:
This is a feature request that would require reverse engineering to support these boards in question.  There isn't anyone that I know of on the project regularly engaging in RE efforts.

If the kernel now supports this chip, you could contribute an EVB-style system to meta-nuvoton in our repository to at least get something building.  I suspect there is u-boot support, etc. that still needs to be done?

closed 2023-09-25 by williamspatrick

neuschaefer, 2023-09-25:
I do RE, but not at a particularly fast speed, because I only work on it on and off. (Modern) U-Boot support for WPCM450 doesn't exist yet.

In any case, anyone who is interested in this is welcome to continue the discussion at [neuschaefer/wpcm450](https://github.com/neuschaefer/wpcm450).