openbmc/openbmc #3789: Bitbake fails to build mklibs-native for p10bmc

mikecgithub, 2021-05-04:
**Note on Bugs**
Please note that questions or feature requests should be discussed on the
mailing list or Discord. Please see https://github.com/openbmc/openbmc#contact
for info on how to contact us. Github Issues in openbmc/openbmc are for bugs
only. Please remove this section when opening a bug.

**Unexpected behavior you saw**
build fails for p10bmc on bitbake package mklibs-native-0.1.44-r0

**Expected behavior**
package should compile, allowing the phosphor-image build to complete

**To Reproduce**
Steps to reproduce the behavior:
1. checkout master (confirmed as of * master 1408ee549 openpower-hw-diags: srcrev bump 2d5c63d90c..eea45427d3)
2.  . setup  p10bmc ~/builds/p10bmc
3.  bitbake mklibs-native
4.  It will fail with a compiler error like this:

```
| make[3]: Entering directory '/home/mcapps99/builds/p10bmc/tmp/work/x86_64-linux/mklibs-native/0.1.44-r0/build/src/mklibs-readelf'
| g++  -DHAVE_CONFIG_H -I. -I../../../mklibs-0.1.44/src/mklibs-readelf -I../..   -isystem/home/mcapps99/builds/p10bmc/tmp/work/x86_64-linux/mklibs-native/0.1.44-r0/recipe-sysroot-native/usr/include  -isystem/home/mcapps99/builds/p10bmc/tmp/work/x86_64-linux/mklibs-native/0.1.44-r0/recipe-sysroot-native/usr/include -O2 -pipe -c -o elf.o ../../../mklibs-0.1.44/src/mklibs-readelf/elf.cpp
| g++  -DHAVE_CONFIG_H -I. -I../../../mklibs-0.1.44/src/mklibs-readelf -I../..   -isystem/home/mcapps99/builds/p10bmc/tmp/work/x86_64-linux/mklibs-native/0.1.44-r0/recipe-sysroot-native/usr/include  -isystem/home/mcapps99/builds/p10bmc/tmp/work/x86_64-linux/mklibs-native/0.1.44-r0/recipe-sysroot-native/usr/include -O2 -pipe -c -o main.o ../../../mklibs-0.1.44/src/mklibs-readelf/main.cpp
| In file included from ../../../mklibs-0.1.44/src/mklibs-readelf/elf_data.hpp:24,
|                  from ../../../mklibs-0.1.44/src/mklibs-readelf/elf.cpp:21:
| ../../../mklibs-0.1.44/src/mklibs-readelf/elf.hpp:52:56: error: ISO C++17 does not allow dynamic exception specifications
|    52 |       const section &get_section(unsigned int i) const throw (std::out_of_range) { return *sections.at(i); };
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:

```
NAME=Fedora
VERSION="34 (Workstation Edition)"
ID=fedora
VERSION_ID=34
VERSION_CODENAME=""
PLATFORM_ID="platform:f34"
PRETTY_NAME="Fedora 34 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:34"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/34/system-administrators-guide/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=34
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=34
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```

 - uname -a: 

Linux fedora 5.11.15-300.fc34.x86_64 #1 SMP Fri Apr 16 13:41:48 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

 - Any additional patches or changes:

**Additional context**

narrowed it down that BB package after trying to build phosphor-image

williamspatrick, 2021-05-04:
This should be reported upstream to Yocto.  But, I suspect upstream doesn't support Fedora 34 officially yet (or else we might need to get @geissonator to sync our branches from upstream).

What I see:
1. The mklibs-native recipe is using the host compiler, rather than the bitbake-generated native compiler.  I don't know if this is expected but might be a problem with the recipe (would need to be asked to Yocto and fixed there).
2. The version of GCC in Fedora 34 is defaulting to C++17, which is a new default for GCC in recent version of GCC.
3. The upstream `mklibs` package has some code which does not compile under C++17.

There are a few avenues of fixes:
- Check if there is an update to `mklibs` in Yocto that we don't have in our repo and ask Andrew to sync if so.
- Check if the real `mklibs` repository has a fix for this C++17 compile issue.  If not, report to them.
- Contribute a fix to Yocto that either:
    * bumps the mklibs version to include fix.
    * backports a patch for mklibs to the version currently used in Yocto
    * Forces off C++17 when compiling mklibs.

rafaelvc, 2021-05-26:
Hi, I am studying Yocto on a Fedora 34 and also hit same problem. I've [patched mklibs](https://gist.github.com/rafaelvc/449f89e56ed52cb8be768551fc3d1f32) to workaround that compiling issue. 

That is for sure not the right solution but can help if you need.  The important thing here is that the only issue I had to use Poky/Yocto to build a core-image-minimal on Fedora 34 was this issue with mklib. I am able to run the generated image on qemu.

So this add the patch to Poky repo:

```
diff --git a/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb b/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb
index 1784af1f4c..86476266fd 100644
--- a/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb
+++ b/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb
@@ -12,6 +12,7 @@ SRC_URI = "http://snapshot.debian.org/archive/debian/20180828T214102Z/pool/main/
        file://avoid-failure-on-symbol-provided-by-application.patch \
        file://show-GNU-unique-symbols-as-provided-symbols.patch \
        file://fix_cross_compile.patch \
+       file://c17_throw_param_remove.patch \
 "
```

depate, 2021-05-27:
Is there any update on this?

I encounter the same error as mike, however on an Arch Linux system. 

Do I see it correctly, that rafael's patch only deletes the error-exception? :thinking:

rafaelvc, 2021-05-27:
@depate  It does not deletes the exception itself. It makes the exception default to std::terminate when you use throw() if I remember correctly.  C++17 is not accepting dynamic exceptions. I think that patch won't case any wrong functioning but the exception needs to be better handled instead. I just did not have time to wait upstream.

depate, 2021-06-07:
Ok, I see. Has a process started to fix this upstream either in Yocto or mklibs?

BrunoVernay, 2021-06-09:
I would be interested to follow the upstream issue, if there is a link??

williamspatrick, 2021-07-08:
Fixed via 76f91bca10fbb88a5e376f4a338baa30778f9ec8.

closed 2021-08-03 by mikecgithub