openbmc/openbmc #3791: BMC watchdog status

NKV-obmc, 2021-05-11:
**Note on Bugs**
Please note that questions or feature requests should be discussed on the
mailing list or Discord. Please see https://github.com/openbmc/openbmc#contact
for info on how to contact us. Github Issues in openbmc/openbmc are for bugs
only. Please remove this section when opening a bug.

**Unexpected behavior you saw**
A clear and concise description of what unexpectedly happened.

**Expected behavior**
A clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:
 - uname -a:
 - Any additional patches or changes:

**Additional context**
Add any other context about the problem here.

solonfan, 2021-06-28:
Check uboot code,firmware identify watchdog if happened by WDT **timeout status register**.

mdmillerii, 2021-07-17:
On Aspeed we export the status but I don't remember the file

williamspatrick, 2023-09-25:
Support request.  Appears to be answered?

closed 2023-09-25 by williamspatrick