openbmc/openbmc #3796: KVM  Super Keys (XK_Super_L & XK_Super_R) are not working

tejas-patil20, 2021-05-21:
## Description:

- If I try to send the super keys then there is no response coming for it.
- As I can see, super keys (XK_Super_L & XK_Super_R) are not configured with obmc-ikvm server.
- Only for below modifiers the key codes has been added:

[image]

- And additionally if I use "XK_Meta_L" key instead of "XK_Super_L", then once i used Super key, then other keys will not work.

- Can anyone help me out on this?
- I need to add Super Keys  with obmc-ikvm.
- What are the scan codes for the Super keys? 

- @yoojae
- @williamspatrick 
- @geissonator 
- @eddiejames 
- @gtmills 
- @mdmillerii 

- Anybody please reply on this!

williamspatrick, 2021-05-21:
Why am I @‘d on this?

williamspatrick, 2023-09-25:
Please feel free to contribute changes to obmc-ikvm.  This is not an openbmc/openbmc issue but a feature request on a sub-component.

closed 2023-09-25 by williamspatrick