openbmc/openbmc #3797: intermittent build failure: image-full: No such file or directory

geissonator, 2021-05-24:
**Unexpected behavior you saw**
I see in CI that we hit this error once or twice a day. I haven't seen a rhyme or reason. It hits different machine configs and it hits on different build servers. A retrigger of the job that fails always works.

**Expected behavior**
We should not hit this build failure.

**To Reproduce**
Intermittent failure that openbmc-ci finds.  Here's an example:

https://jenkins.openbmc.org/job/ci-openbmc/4233/distro=ubuntu,label=docker-builder,target=tiogapass/console
```
16:00:23 | DEBUG: sed -e 's:^[^/]*/:/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/work/tiogapass-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/recipe-sysroot-native/:g' /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/sysroots-components/x86_64/gcc-cross-arm/fixmepath /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/sysroots-components/x86_64/bison-native/fixmepath /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/sysroots-components/x86_64/gmp-native/fixmepath | xargs sed -i -e 's:FIXMESTAGINGDIRTARGET:/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/work/tiogapass-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/recipe-sysroot:g; s:FIXMESTAGINGDIRHOST:/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/work/tiogapass-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/recipe-sysroot-native:g' -e 's:FIXME_PSEUDO_SYSROOT:/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/sysroots-components/x86_64/pseudo-native:g' -e 's:FIXME_HOSTTOOLS_DIR:/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/hosttools:g' -e 's:FIXME_PKGDATA_DIR:/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/pkgdata/tiogapass:g' -e 's:FIXME_PSEUDO_LOCALSTATEDIR:/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/work/tiogapass-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/pseudo/:g' -e 's:FIXME_LOGFIFO:/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/work/tiogapass-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/fifo.151842:g'
16:00:23 | DEBUG: Python function extend_recipe_sysroot finished
16:00:23 | DEBUG: Executing shell function do_generate_static_tar
16:00:23 | 229+1 records in
16:00:23 | 229+1 records out
16:00:23 | 234612 bytes (235 kB, 229 KiB) copied, 0.403096 s, 582 kB/s
16:00:23 | image-full: No such file or directory
16:00:23 | WARNING: exit code 1 from a shell command.
16:00:23 | ERROR: Execution of '/data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/tiogapass/build/work/tiogapass-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/run.do_generate_static_tar.151842' failed with exit code 1:
16:00:23 | 229+1 records in
16:00:23 | 229+1 records out
16:00:23 | 234612 bytes (235 kB, 229 KiB) copied, 0.403096 s, 582 kB/s
16:00:23 | image-full: No such file or directory
16:00:23 | WARNING: exit code 1 from a shell command.
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 Building master

mine260309, 2021-05-25:
This looks related to the signature for the full image: f0afcf836
@lxwinspur could you kindly check the issue?

lxwinspur, 2021-05-28:
@geissonator 
TBH, I have never hit this issue
But I pushed a new patch temporary
https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/43615
Could you check it?

Also, I will trace it once I hit this issue.

closed 2021-06-03 by geissonator