openbmc/openbmc #3798: Build fails: openssl do_configure

depate, 2021-05-27:
**Unexpected behavior you saw**

The build fails with:

```
Summary: 1 task failed:
virtual:native:/home/patrik/devel/openbmc/meta/recipes-connectivity/openssl/openssl_1.1.1k.bb:do_configure
Summary: There were 10 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

**To Reproduce**
Steps to reproduce the behavior:

- Follow the get started guide

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:
 ```
NAME=Fedora
VERSION="34 (Server Edition)"
ID=fedora
VERSION_ID=34
VERSION_CODENAME=""
PLATFORM_ID="platform:f34"
PRETTY_NAME="Fedora 34 (Server Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:34"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/34/system-administrators-guide/"
SUPPORT_URL="https://fedoraproject.org/wiki/Communicating_and_getting_help"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=34
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=34
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Server Edition"
VARIANT_ID=server
```
 
 - uname -a: 
 ```
Linux fatalis 5.12.6-300.fc34.x86_64 #1 SMP Sat May 22 20:42:55 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```
 - Any additional patches or changes:

depate, 2021-06-07:
I digged a bit deeper in to the error and I've found the log:

```Log data follows:                                                                                                                                                                                                    | DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/patrik/devel/openbmc/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot']                                                                                  | NOTE: Installed into sysroot: []                                                                                                                                                                                   | NOTE: Skipping as already exists in sysroot: ['quilt-native']
| DEBUG: Python function extend_recipe_sysroot finished                                                                                                                                                              | DEBUG: Executing shell function do_configure                                                                                                                                                                       | Can't locate FindBin.pm in @INC (you may need to install the FindBin module) (@INC contains: /home/patrik/devel/openbmc/build/tiogapass/tmp/work/x86_64-linux/openssl-native/1.1.1k-r0/openssl-1.1.1k/external/perl
/Text-Template-1.46/lib/ /usr/local/lib64/perl5/5.32 /usr/local/share/perl5/5.32 /usr/lib64/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib64/perl5 /usr/share/perl5) at /home/patrik/devel/openbmc/build/tiogapass/tmp/work/x86_64-linux/openssl-native/1.1.1k-r0/openssl-1.1.1k/Configure line 15.```

Apparently `perl-FindBin` was missing. 

If you're on a fedora system install it via:

```dnf install perl-FindBin```

closed 2021-06-07 by depate

anoo1, 2021-06-07:
README updated with https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/43840