openbmc/openbmc #3799: DT-bindings directory

kthoru01, 2021-05-27:
Hello,

I apologize for asking such a novice question, but I saw that one of the .dts files references a dt-bindings directory. Where could I find that? I would like to take a look at the definitions to understand the requirements for building a DT for a BMC. Thank you very much!

mdmillerii, 2021-05-27:
We ask that questions be directed to email or discord.
The device tree bindings are in the Linux kernel repository.

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings
https://www.kernel.org/doc/html/latest/devicetree/index.html
https://www.devicetree.org/

edtanous, 2022-05-23:
Per the first response, please ask your questions to the relevant communities forums, per mdmilleriis response.

Closing.

closed 2022-05-23 by edtanous