openbmc/openbmc #3800: Clone the code from git, but can not compile pass

damon3000, 2021-05-28:
**Note on Bugs**
Use the steps to try use openbmc to compile, but failed.
#git clone git@github.com:openbmc/openbmc.git
#cd openbmc-master
#source setup romulus
#bitbake obmc-phosphor-image

**Unexpected behavior you saw**
ERROR: pdbg-3.2-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1001/bus"; export SSH_AGENT_PID="76584"; export SSH_AUTH_SOCK="/tmp/ssh-cFuPrCcGYeCq/agent.76357"; export PATH="/home3/openbmc/openbmc-master/scripts:/home3/openbmc/openbmc-master/build/rainier/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/pdbg/3.2-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/home3/openbmc/openbmc-master/build/rainier/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/pdbg/3.2-r0/recipe-sysroot/usr/bin/crossscripts:/home3/openbmc/openbmc-master/build/rainier/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/pdbg/3.2-r0/recipe-sysroot-native/usr/sbin:/home3/openbmc/openbmc-master/build/rainier/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/pdbg/3.2-r0/recipe-sysroot-native/usr/bin:/home3/openbmc/openbmc-master/build/rainier/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/pdbg/3.2-r0/recipe-sysroot-native/sbin:/home3/openbmc/openbmc-master/build/rainier/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/pdbg/3.2-r0/recipe-sysroot-native/bin:/home3/openbmc/openbmc-master/poky/bitbake/bin:/home3/openbmc/openbmc-master/build/rainier/tmp/hosttools"; export HOME="/home/damon"; git -c core.fsyncobjectfiles=0 ls-remote git://github.com/open-power/pdbg.git  failed with exit code 128, output:
fatal: read error: Connection reset by peer

ERROR: Logfile of failure stored in: /home3/openbmc/openbmc-master/build/rainier/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/pdbg/3.2-r0/temp/log.do_fetch.1232872
ERROR: Task (/home3/openbmc/openbmc-master/meta-openpower/recipes-bsp/pdbg/pdbg_3.2.bb:do_fetch) failed with exit code '1'
WARNING: openpower-libhei-0.1+gitAUTOINC+0a905012d0-r0 do_fetch: Failed to fetch URL git://github.com/openbmc/openpower-libhei, attempting MIRRORS if available
Waiting for 3 running tasks to finish:

**Expected behavior**
Compile pass

**To Reproduce**
Steps to reproduce the behavior:
#git clone git@github.com:openbmc/openbmc.git
#cd openbmc-master
#source setup romulus
#bitbake obmc-phosphor-image

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
 - /etc/os-release:
 - uname -a:
 - Any additional patches or changes:

**Additional context**
Compile failed, checked the pdbg_3.2.bb and found that
➜  pdbg cat pdbg_3.2.bb
SUMMARY     = "PowerPC FSI Debugger"
DESCRIPTION = "pdbg allows JTAG-like debugging of the host POWER processors"
LICENSE     = "Apache-2.0"
LIC_FILES_CHKSUM = "file://${S}/COPYING;md5=3b83ef96387f14655fc854ddc3c6bd57"

SRC_URI += "git://github.com/open-power/pdbg.git"
SRCREV = "v3.2"

DEPENDS += "dtc-native"

S = "${WORKDIR}/git"

inherit autotools

BBCLASSEXTEND = "native"
➜  pdbg

The SRC_URI is "git://github.com/open-power/pdbg.git", I try to clone it but failed.
I checked the pdbg git and found its SSH link is git@github.com:open-power/pdbg.git and its HTTPS link is https://github.com/open-power/pdbg.git

I used https link to clone, it can be cloned. So did the pdbg_3.2.bb file configuration error?

geissonator, 2021-06-01:
Doing a "git clone git://github.com/open-power/pdbg.git" worked ok for me. I wonder, have you ever setup your public/private keys within github?  https://docs.github.com/en/github/authenticating-to-github/connecting-to-github-with-ssh. Or if it's just that one repo. maybe a timing issue? Have you retried a few times?

damon3000, 2021-06-07:
I can not use your method to clone the repo, but can use https link to clone it. The SSH link failed maybe due to the SSH auth configuration. Below is the log.
I have tried so many times but failed, also I tried other source except romuls, still failed.

```
➜  /home3 sudo git clone git://github.com/open-power/pdbg.git
Cloning into 'pdbg'...
fatal: read error: Connection reset by peer
➜  /home3 sudo git clone https://github.com/open-power/pdbg
Cloning into 'pdbg'...
remote: Enumerating objects: 4509, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 4509 (delta 3), reused 7 (delta 2), pack-reused 4496
Receiving objects: 100% (4509/4509), 1.29 MiB | 1.17 MiB/s, done.
Resolving deltas: 100% (3287/3287), done.
➜  /home3 sudo rm -rf pdbg
➜  /home3 sudo git clone git@github.com:open-power/pdbg.git
Cloning into 'pdbg'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
➜  /home3
```

williamspatrick, 2021-06-07:
You should not need to run ‘sudo’ for git. That will cause it to use root’s ssh config and not your own, which could be why you are failing to clone in the last attempt.  Did you set up your ssh keys as suggested?

The second attempt passed because it is using the normal https clone path.

The first one looks to me like a firewall issue.  Do you need any kind of proxy to get out of your network when using non-http/ssh protocols?