openbmc/openbmc #3804: Unable to flash openbmc to spi-flash

Bigcaibird, 2021-06-18:
Some errors have been encountered when i try to flash romulus bmc image to my board ,the `socflash `show me that:
`ASPEED SOC Flash Utility v.1.18.00 
Warning:
SoCflash utility is only for engineers to update the firmware in lab,
it is not a commercialized software product,
ASPEED has not done compatibility/reliability stress test for SoCflash.
Please do not use this utility for any mass production purpose.
Press y to continue if you are agree ....
y
Find ASPEED Device 1a03:2000 on 4:0.0 
MMIO Virtual Address: c3c03000 
Relocate IO Base: 3000 
Found ASPEED Device 1a03:2500 rev. 41 
Static Memory Controller Information: 
CS0 Flash Type is SPI 
CS1 Flash Type is SYS SPI 
CS2 Flash Type is SYS SPI 
CS3 Flash Type is SYS SPI 
CS4 Flash Type is SYS SPI 
Boot CS is 1 
Option Information: 
CS: 1 
Flash Type: SPI 
The BMC is proteced, skip to update firmware image!! 
`
In the right case, it will show ：
`ASPEED SOC Flash Utility v.1.18.00 
Warning:
SoCflash utility is only for engineers to update the firmware in lab,
it is not a commercialized software product,
ASPEED has not done compatibility/reliability stress test for SoCflash.
Please do not use this utility for any mass production purpose.
Press y to continue if you are agree ....
y
Find ASPEED Device 1a03:2000 on 4:0.0 
MMIO Virtual Address: be94b000 
Relocate IO Base: 3000 
Found ASPEED Device 1a03:2500 rev. 41 
Static Memory Controller Information: 
CS0 Flash Type is SPI 
CS1 Flash Type is SPI 
CS2 Flash Type is SPI 
CS3 Flash Type is NOR 
CS4 Flash Type is NOR 
Boot CS is 0 
Option Information: 
CS: 0 
Flash Type: SPI 
[Warning] Don't AC OFF or Reboot System During BMC Firmware Update!! 
[SOCFLASH] Flash ID : 1940ef 
Find Flash Chip #1: WinbondW25Q256/257 
Update Flash Chip #1 O.K.            
Update Flash Chip O.K.  `

I've tried for a long time and still can't solve this problem,can someone give me some suggestions?
Best regards!

mdmillerii, 2021-06-21:
```
The BMC is proteced, skip to update firmware image!!
```

As the text says it's an unsupported utility to write BMC storage from the host.  The BMC apparently has blocked something perhaps it locked a portion of the flash chip.

Do you actually have a Romulus?   There are several items including gpio usage and network connections (phy vs NCSI etc) that are board unique.

Bigcaibird, 2021-06-22:
> ```
> The BMC is proteced, skip to update firmware image!!
> ```
>
> As the text says it's an unsupported utility to write BMC storage from the host. The BMC apparently has blocked something perhaps it [...]

@mdmillerii    actually,I'm trying to port Romulus to my own machine and I don't have a Romulus,I guess this situation is caused by GPIO, because when there is a Romulus image in flash, socflash can't recognize spiflash correctly。

mdmillerii, 2021-06-22:
Actually OpenBMC U-Boot configures the aspeed soc to block the access that SoCflash uses.

amboar, 2021-06-22:
On @mdmillerii 's point, this patch:

https://github.com/openbmc/openbmc/blob/master/meta-phosphor/aspeed-layer/recipes-bsp/u-boot/files/0001-aspeed-Disable-unnecessary-features.patch

Addresses this issue:

https://github.com/openbmc/openbmc/issues/3475

Socflash uses these bridges to operate.

Bigcaibird, 2021-06-22:
> Actually OpenBMC U-Boot configures the aspeed soc to block the access that SoCflash uses.

I tried to modify the code in uboot about flash protect, but it didn't work，and meta-evb / meta-aspeed / meta-aspeed-ast-2500 does not have this problem.

And I want to ask, what may be the cause of the difference below

`Static Memory Controller Information:
 CS0 Flash Type is SPI 
CS1 Flash Type is SYS SPI 
CS2 Flash Type is SYS SPI 
CS3 Flash Type is SYS SPI 
CS4 Flash Type is SYS SPI 
Boot CS is 1 
Option Information: CS: 1 
Flash Type: SPI`

` Static Memory Controller Information:
 CS0 Flash Type is SPI CS1 Flash Type is SPI 
CS2 Flash Type is SPI 
CS3 Flash Type is NOR 
CS4 Flash Type is NOR 
Boot CS is 0 Option Information: CS: 0 
Flash Type: SPI 
[Warning] Don't AC OFF or Reboot System During BMC Firmware Update!! 
[SOCFLASH] Flash ID : 1940ef 
Find Flash Chip #1: WinbondW25Q256/257
 Update Flash Chip #1 O.K. Update Flash Chip O.K. `

mdmillerii, 2021-06-22:
Modify the bitbake recipe to remove the source patch that @amboar pointed out.

Bigcaibird, 2021-06-22:
> 在 @mdmillerii 的观点上，这个补丁：
> 
> [https://github.com/openbmc/openbmc/blob/master/meta-phosphor/aspeed-layer/recipes-bsp/u-boot/files/0001-aspeed-Disable-unnecessary-features.patch ](https://github.com/openbmc/openbmc/blob/master/meta-phosphor/aspeed-layer/recipes-bsp/u-boot/files/0001-aspeed-Disable-unnecessary-features.patch)
> 
> 解决了这个问题：
> 
> [#3475 ](https://github.com/openbmc/openbmc/issues/3475)
> 
> Socflash 使用这些网桥进行操作。

Thanks a lot!! @mdmillerii @amboar

closed 2021-06-22 by Bigcaibird