openbmc/openbmc #3805: able to confirm /sbin/init is an executable non-empty file in merged file system mounted at /root

dufan89, 2021-06-21:
I use qemu to simulate at local PC,then it occurs a problem described as follows:
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.14-56621ca (oe-user@oe-host) (arm-openbmc-linux-gnueabi-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.0.20200730) #1 Thu Jun 17 02:52:14 UTC 2021
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00093177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: Palmetto BMC
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000004effffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000004effffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000004effffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 60960
[    0.000000] Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 231024K/245760K available (7168K kernel code, 885K rwdata, 1416K rodata, 1024K init, 185K bss, 14736K reserved, 0K cma-reserved)
[    0.000000] random: get_random_u32 called from cache_alloc_refill+0x420/0x96c with crng_init=0
[    0.000000] ftrace: allocating 23223 entries in 46 pages
[    0.000000] ftrace: allocated 46 pages with 4 groups
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] i2c controller registered, irq 16
[    0.000000] clocksource: FTTMR010-TIMER2: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 39817925974 ns
[    0.000139] sched_clock: 32 bits at 48MHz, resolution 20ns, wraps every 44739242997ns
[    0.001639] Switching to timer-based delay loop, resolution 20ns
[    0.008216] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.008544] pid_max: default: 32768 minimum: 301
[    0.011448] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.011551] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.056777] CPU: Testing write buffer coherency: ok
[    0.069382] Setting up static identity map for 0x40100000 - 0x40100058
[    0.071398] ASPEED AST2400 rev A1 (02010303)
[    0.135432] devtmpfs: initialized
[    0.152526] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.152979] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.154278] pinctrl core: initialized pinctrl subsystem
[    0.161615] NET: Registered protocol family 16
[    0.165238] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.385273] clocksource: Switched to clocksource FTTMR010-TIMER2
[    1.304266] NET: Registered protocol family 2
[    1.311588] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    1.311786] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    1.311940] TCP bind hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    1.312113] TCP: Hash tables configured (established 2048 bind 2048)
[    1.313602] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    1.313886] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    1.316677] NET: Registered protocol family 1
[    1.321630] RPC: Registered named UNIX socket transport module.
[    1.321755] RPC: Registered udp transport module.
[    1.321796] RPC: Registered tcp transport module.
[    1.321829] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    1.327955] Unpacking initramfs...
[    2.379213] Freeing initrd memory: 1124K
[    2.382912] workingset: timestamp_bits=30 max_order=16 bucket_order=0
[    2.384138] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    2.387598] NFS: Registering the id_resolver key type
[    2.398514] Key type id_resolver registered
[    2.398592] Key type id_legacy registered
[    2.398757] nfs4filelayout_init: NFSv4 File Layout Driver Registering...
[    2.398857] nfs4flexfilelayout_init: NFSv4 Flexfile Layout Driver Registering...
[    2.399093] jffs2: version 2.2. (SUMMARY)  © 2001-2006 Red Hat, Inc.
[    2.400598] NET: Registered protocol family 38
[    2.404464] gpio-824 (func_mode0): hogged as output/low
[    2.404601] gpio-825 (func_mode1): hogged as output/low
[    2.404646] gpio-826 (func_mode2): hogged as output/low
[    2.404953] gpio-804 (BMC_FAN_RESERVED_N): hogged as input
[... 85 lines cut ...]
Change Root test failed!
Fatal error, triggering kernel panic!
[    8.482215] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
[    8.483113] CPU: 0 PID: 1 Comm: init Not tainted 5.10.14-56621ca #1
[    8.483675] Hardware name: Generic DT based system
[    8.484107] Backtrace: 
[    8.484768] [<807aaa90>] (dump_backtrace) from [<807aacec>] (show_stack+0x20/0x24)
[    8.485887]  r7:808e07a0 r6:80b09dd0 r5:00000000 r4:808efda0
[    8.486433] [<807aaccc>] (show_stack) from [<807b27ec>] (dump_stack+0x28/0x30)
[    8.487083] [<807b27c4>] (dump_stack) from [<807ab0c0>] (panic+0xf8/0x330)
[    8.487728]  r5:00000000 r4:80bde268
[    8.488076] [<807aafc8>] (panic) from [<80116440>] (do_exit+0x93c/0x99c)
[    8.488575]  r3:00000001 r2:60000013 r1:00000100 r0:808e07a0
[    8.488989]  r7:ffffe000
[    8.489260] [<80115b04>] (do_exit) from [<8011653c>] (do_group_exit+0x4c/0xc4)
[    8.489843]  r7:000000f8
[    8.490121] [<801164f0>] (do_group_exit) from [<801165d4>] (sys_exit_group+0x20/0x24)
[    8.490948]  r5:00000004 r4:0248f190
[    8.491332] [<801165b4>] (sys_exit_group) from [<80100060>] (ret_fast_syscall+0x0/0x50)
[    8.492005] Exception stack(0x810adfa8 to 0x810adff0)
[    8.492866] dfa0:                   0248f190 00000004 00000001 00000000 00000001 0248f190
[    8.493883] dfc0: 0248f190 00000004 00000004 000000f8 00000000 00000000 00598a7c 00000000
[    8.494490] dfe0: 00598b34 7e88fa10 005215f0 76e70b04

then it begins to reboot.....

mdmillerii, 2021-06-21:
The immediate cause is the squashfs failed to mount as evidenced by the line
```
miunt: mounting /dev/mtdblock4 on run/initramfs/ro failed: Invalid argument
```

I also noticed unexpected but probably harmless `root=/dev/ram rw` in your kernel command line.

Are you following the guide linked under development here (in our README.md)?
https://github.com/openbmc/openbmc#openbmc-development

williamspatrick, 2023-09-25:
I have no idea what is going on here.  Lots of unformatted copy/paste of a boot log with no details.

> [ 0.000000] Linux version 5.10.14-56621ca (oe-user@oe-host) (arm-openbmc-linux-gnueabi-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.0.20200730) https://github.com/openbmc/openbmc/pull/1 Thu Jun 17 02:52:14 UTC 2021

We haven't used 5.10 or GCC-10 in a long time.  Can you give more details on what you're trying to accomplish?

closed 2023-09-25 by williamspatrick

mdmillerii, 2023-09-25:
> `[ 3.457754] Run /init as  process`
> `roofs = mtd4 squashfs rwfs = mtd5 jffs2`

> `mount: mounting /dev/mtdblock4 on run/initramfs/ro failed: Invalid argument`

You don't have a squashfs image at the location defined by your partition layout for the 4th partition with label rofs