openbmc/openbmc #3807: sensor not display at webui!

Bigcaibird, 2021-07-02:
hi, all:
     i'm try to porting romulus to our hardware, so i add 4 temp sensor at i2c2 (addr:0x48;0x49;0x4a;0x4b)and add iio-hwmon to Monitor voltage，but now i have a problem that only one temp sensor(ox48) show at webui and no voltage info show.
Can you give me some advice? thanks a lot!
here is some info below:

1.my dts
 **i2c**
```&i2c2 {
	status = "okay";

	nst175@48 {
		compatible = "national,lm75";
		reg = <0x48>;
	};

	nst175@49 {
		compatible = "national,lm75";
		reg = <0x49>;
	};

	nst175@4a {
		compatible = "national,lm75";
		reg = <0x4a>;
	};

	nst175@4b {
		compatible = "national,lm75";
		reg = <0x4b>;
	};
};```

**iio-hwmon**

```iio-hwmon {
		compatible = "iio-hwmon";
		io-channels = <&adc 0> , <&adc 1> , 
		<&adc 2> , <&adc 3> , <&adc 4> , 
		<&adc 5> , <&adc 6> , <&adc 7>,
		<&adc 8> , <&adc 9> , <&adc 10> ,
		<&adc 11> , <&adc 12> , 
		<&adc 13> , <&adc 14>;
	};

&adc{
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_adc0_default
			&pinctrl_adc1_default
			&pinctrl_adc2_default
			&pinctrl_adc3_default
			&pinctrl_adc4_default
			&pinctrl_adc5_default
			&pinctrl_adc6_default
			&pinctrl_adc7_default
			&pinctrl_adc8_default
			&pinctrl_adc9_default
			&pinctrl_adc10_default
			&pinctrl_adc11_default
			&pinctrl_adc12_default
			&pinctrl_adc13_default
			&pinctrl_adc14_default
			&pinctrl_adc15_default>;
};```

2.my phosphor-hwmon_%.bbappend

```FILESEXTRAPATHS_prepend_net1833 := "${THISDIR}/${PN}:"

EXTRA_OEMESON_append_net1833 = " -Dnegative-errno-on-fail=true"

CHIPS = " \
        bus@1e78a000/i2c-bus@c0/nst175@48 \
        bus@1e78a000/i2c-bus@c0/nst175@49 \
        bus@1e78a000/i2c-bus@c0/nst175@4a \
        bus@1e78a000/i2c-bus@c0/nst175@4b \
        pwm-tacho-controller@1e786000 \
        "
ITEMSFMT = "ahb/apb/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"

ITEMS += "iio-hwmon.conf"
ITEMS += "iio-hwmon-battery.conf"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN}_append_net1833 = " ${@compose_list(d, 'ENVS', 'ITEMS')}"
```

3. recipes-kernel/linux/linux-aspeed/net1833.cfg

```CONFIG_SENSORS_ASPEED=y
CONFIG_ASPEED_ADC=y
CONFIG_SENSORS_IIO_HWMON=y
CONFIG_SENSORS_LM75=y
CONFIG_PMBUS=y
CONFIG_EEPROM_AT24=y
CONFIG_SENSORS_ASPEED=y
CONFIG_I2C_MUX=y
CONFIG_I2C_SLAVE=y
CONFIG_NET_NCSI=y
CONFIG_VLAN_8021Q=y
CONFIG_EEPROM_AT24=y
CONFIG_FTGMAC100=y
CONFIG_SPI=y```

4. iio-hwmon.conf

```LABEL_in1=12v_sensor
GAIN_in1=8.85
WARNLO_in1=11500
WARNHI_in1=12500
CRITLO_in1=11400
CRITHI_in1=12600

LABEL_in2=5v_sensor
GAIN_in2=3.73
WARNLO_in2=4800
WARNHI_in2=5200
CRITLO_in2=4750
CRITHI_in2=5250

LABEL_in3=3.3v_sensor
GAIN_in3=2.43
WARNLO_in3=3200
WARNHI_in3=3400
CRITLO_in3=3135
CRITHI_in3=5250

LABEL_in4=cpu1_io_sensor
GAIN_in4=1
WARNLO_in4=1750
WARNHI_in4=1850
CRITLO_in4=1710
CRITHI_in4=1890

LABEL_in5=cpu1_ddr_l_sensor
GAIN_in5=1
WARNLO_in5=1150
WARNHI_in5=1250
CRITLO_in5=1140
CRITHI_in5=1260

LABEL_in6=cpu1_ddr_r_sensor
GAIN_in6=1
WARNLO_in6=1150
WARNHI_in6=1250
CRITLO_in6=1140
CRITHI_in6=1260

LABEL_in7=cpu1_vp_sensor
GAIN_in7=1
WARNLO_in7=870
WARNHI_in7=980
CRITLO_in7=860
CRITHI_in7=990

LABEL_in8=cpu1_pcie_sensor
GAIN_in8=1
WARNLO_in8=870
WARNHI_in8=840
CRITLO_in8=830
CRITHI_in8=880

LABEL_in9=cpu1_core_sensor
GAIN_in9=1
WARNLO_in9=870
WARNHI_in9=840
CRITLO_in9=830
CRITHI_in9=880

LABEL_in10=cpu2_io_sensor
GAIN_in10=1
WARNLO_in10=1750
WARNHI_in10=1850
CRITLO_in10=1710
CRITHI_in10=1890

LABEL_in11=cpu2_ddr_l_sensor
GAIN_in11=1
WARNLO_in11=1150
WARNHI_in11=1250
CRITLO_in11=1140
CRITHI_in11=1260

LABEL_in12=cpu2_ddr_r_sensor
GAIN_in12=1
WARNLO_in12=1150
WARNHI_in12=1250
CRITLO_in12=1140
CRITHI_in12=1260

LABEL_in13=cpu2_vp_sensor
GAIN_in13=1
WARNLO_in13=870
WARNHI_in13=980
CRITLO_in13=860
CRITHI_in13=990

LABEL_in14=cpu2_pcie_sensor
GAIN_in14=1
WARNLO_in14=870
WARNHI_in14=840
CRITLO_in14=830
CRITHI_in14=880

LABEL_in15=cpu2_cor_sensor
GAIN_in15=1
WARNLO_in15=870
WARNHI_in15=840
CRITLO_in15=830
CRITHI_in15=880

```

5.

```
root@net1833:~# ls /sys/class/hwmon/hwmon1/
device      in12_input  in15_input  in3_input   in6_input   in9_input   subsystem
in10_input  in13_input  in1_input   in4_input   in7_input   name        uevent
in11_input  in14_input  in2_input   in5_input   in8_input   of_node

```

6.

```
root@net1833:~# ls -l /sys/devices/platform/ahb/ahb\:apb/ahb\:apb\:bus\@1e78a000/1e78a0c0.i2c-bus/i2c-2/
2-0048/        2-0049/        2-004a/        2-004b/        delete_device  i2c-dev/       name           new_device     of_node/       subsystem/     uevent

```

8.jouralctl

```root@net1833:~# journalctl |grep hwmon
Jan 01 00:00:13 net1833 kernel: lm75 2-0048: hwmon1: sensor 'lm75'
Jan 01 00:00:13 net1833 kernel: lm75 2-0049: hwmon2: sensor 'lm75'
Jan 01 00:00:13 net1833 kernel: lm75 2-004a: hwmon3: sensor 'lm75'
Jan 01 00:00:13 net1833 kernel: lm75 2-004b: hwmon4: sensor 'lm75'
Jan 01 00:00:47 net1833 phosphor-hwmon-readd[264]: terminate called after throwing an instance of 'sdbusplus::exception::SdBusError'
Jan 01 00:00:47 net1833 phosphor-hwmon-readd[264]:   what():  sd_bus_add_object_vtable: org.freedesktop.DBus.Error.InvalidArgs: Invalid argument
Jan 01 00:01:10 net1833 systemd-coredump[291]: Process 264 (phosphor-hwmon-) of user 0 dumped core.
Jan 01 00:01:10 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Main process exited, code=dumped, status=6/ABRT
Jan 01 00:01:10 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Failed with result 'core-dump'.
Jan 01 00:01:11 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Scheduled restart job, restart counter is at 1.
Jan 01 00:01:11 net1833 phosphor-hwmon-readd[351]: terminate called after throwing an instance of 'sdbusplus::exception::SdBusError'
Jan 01 00:01:11 net1833 phosphor-hwmon-readd[351]:   what():  sd_bus_add_object_vtable: org.freedesktop.DBus.Error.InvalidArgs: Invalid argument
Jan 01 00:01:20 net1833 systemd-coredump[360]: Process 351 (phosphor-hwmon-) of user 0 dumped core.
Jan 01 00:01:20 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Main process exited, code=dumped, status=6/ABRT
Jan 01 00:01:20 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Failed with result 'core-dump'.
Jan 01 00:01:21 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Scheduled restart job, restart counter is at 2.
Jan 01 00:01:22 net1833 phosphor-hwmon-readd[440]: terminate called after throwing an instance of 'sdbusplus::exception::SdBusError'
Jan 01 00:01:22 net1833 phosphor-hwmon-readd[440]:   what():  sd_bus_add_object_vtable: org.freedesktop.DBus.Error.InvalidArgs: Invalid argument
Jan 01 00:01:30 net1833 systemd-coredump[451]: Process 440 (phosphor-hwmon-) of user 0 dumped core.
Jan 01 00:01:30 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Main process exited, code=dumped, status=6/ABRT
Jan 01 00:01:30 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Failed with result 'core-dump'.
Jan 01 00:01:31 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Scheduled restart job, restart counter is at 3.
Jan 01 00:01:32 net1833 phosphor-hwmon-readd[527]: terminate called after throwing an instance of 'sdbusplus::exception::SdBusError'
Jan 01 00:01:32 net1833 phosphor-hwmon-readd[527]:   what():  sd_bus_add_object_vtable: org.freedesktop.DBus.Error.InvalidArgs: Invalid argument
Jan 01 00:01:40 net1833 systemd-coredump[540]: Process 527 (phosphor-hwmon-) of user 0 dumped core.
Jan 01 00:01:40 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Main process exited, code=dumped, status=6/ABRT
Jan 01 00:01:40 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Failed with result 'core-dump'.
Jan 01 00:01:41 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Scheduled restart job, restart counter is at 4.
Jan 01 00:01:41 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Start request repeated too quickly.
Jan 01 00:01:41 net1833 systemd[1]: xyz.openbmc_project.Hwmon@-iio\x2dhwmon.service: Failed with result 'core-dump'.```

Bigcaibird, 2021-07-02:
phosphor-hwmon-readd is working
```
327     1 root     S     9200   4%   0% phosphor-hwmon-readd -i ce7d9e13a44d4639a24ae52c28396b7b0d60b331a595b4a7972d7462f65bf78e -o /iio-hwmon
  261     1 root     S     9096   4%   0% phosphor-hwmon-readd -i 90dc5dd3857daeb224c11f832395c5c454995ef20ee9cda4c1747f544f1f8541 -o /iio-hwmon-battery
  257     1 root     S     9096   2%   0% phosphor-hwmon-readd -i c29b1870b1b136904a427110ab84ccf9af799a14bb660d5f1429ae187cc8636a -o /ahb/apb/bus@1e78a000/i2c-bus@c0/nst175@48

```

spinler, 2021-07-06:
I wonder if you're hitting https://github.com/openbmc/phosphor-hwmon/blob/master/README.iio.md

Bigcaibird, 2021-07-07:
> I wonder if you're hitting https://github.com/openbmc/phosphor-hwmon/blob/master/README.iio.md
Does this mean that I can only use one between hwmon-iio and hwmon-iio-battery? I tried to remove the hwmon-iio-battery node from the device tree, but the ADC information defined in hwmon-iio is still not displayed on WebUI

Bigcaibird, 2021-07-26:
solved

closed 2021-07-26 by Bigcaibird

wu790607, 2021-08-06:
Hello Bigcaibird 

Can you explain how to fixed this issue? I have same question.

Bigcaibird, 2021-08-06:
> Hello Bigcaibird
> 
> Can you explain how to fixed this issue? I have same question.

You need to provide more information so that I can help you analyze the problem.Such as your dts and .conf

wu790607, 2021-08-06:
I use the sensor driver I write. 
My dts as below
``` 
&i2c6 {
	status = "okay";

	it8728f@2d {
		compatible = "ite,it8728f";
		reg = <0x2d>;
	};
};
```

and .conf
```

Bigcaibird, 2021-08-06:
Try ``busctl | grep Hwmon -> busctl tree "insert the name"``,Can you see these sensors on dubs?

wu790607, 2021-08-06:
I run `busctl | grep Hwmon` and get nothing.
Is I missing something?

Bigcaibird, 2021-08-06:
> I run `busctl | grep Hwmon` and get nothing.
> Is I missing something?
Did you mentioned the .conf file at  phosphor-hwmon_%.bbappend?

Bigcaibird, 2021-08-06:
like that 
```FILESEXTRAPATHS_prepend_s2500 := "${THISDIR}/${PN}:"

EXTRA_OEMESON_append_s2500 = " -Dnegative-errno-on-fail=true"

CHIPS = " \
        bus@1e78a000/i2c-bus@c0/nst175@48 \
        bus@1e78a000/i2c-bus@c0/nst175@49 \
        bus@1e78a000/i2c-bus@c0/nst175@4a \
        bus@1e78a000/i2c-bus@c0/nst175@4b \
        bus@1e78a000/i2c-bus@100/psu@58 \
        bus@1e78a000/i2c-bus@100/psu@59 \
        pwm-tacho-controller@1e786000 \
        "
ITEMSFMT = "ahb/apb/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"

ITEMS += "iio-hwmon.conf"
ITEMS += "iio-hwmon-battery.conf"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN}_append_s2500 = " ${@compose_list(d, 'ENVS', 'ITEMS')}"

```

wu790607, 2021-08-06:
My file as below
```
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"
EXTRA_OECONF_append_bmc7830 = " --enable-negative-errno-on-fail"

CHIPS = " \
        bus@1e78a000/i2c-bus@1c0/it8728f@2d \
        "
ITEMSFMT = "ahb/apb/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"

ITEMS += "iio-hwmon.conf"
ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN}_append_bmc7830 = " ${@compose_list(d, 'ENVS', 'ITEMS')}"
```

And I can fine it8728f@2d.conf (/etc/default/obmc/hwmon/ahb/apb/bus\@1e78a000/i2c-bus\@1c0/it8728f\@2d.conf )

Bigcaibird, 2021-08-06:
> I use the sensor driver I write.
> My dts as below
> 
> ```
> &i2c6 {
> 	status = "okay";
> 
> 	it8728f@2d {
> 		compatible = "ite,it8728f";
> 		reg = <0x2d>;
> 	};
> };
> ```
> 
> and .conf
> 
> ```
**Should start with LABEL_in1=xxxx**
> LABEL_in0=VCORE
> GAIN_in0=0.012
> WARNHI_in0=1650
> WARNLO_in0=0
> CRITHI_in0=1800
> CRITLO_in0=0
> 
> LABEL_in1=VMEM
> GAIN_in1=0.012
> WARNHI_in1=1210
> WARNLO_in1=1100
> CRITHI_in1=1320
> CRITLO_in1=1045
> 
**Try use LABEL_in2=P12V(Use a-z A-Z,and``_``)**
> LABEL_in2=+12V
> GAIN_in2=0.0516
> WARNHI_in2=12540
> WARNLO_in2=11385
> CRITHI_in2=13200
> CRITLO_in2=10780
> 
> LABEL_in3=+5V
> GAIN_in3=0.0218
> WARNHI_in3=5232
> WARNLO_in3=4752
> CRITHI_in3=5471
> CRITLO_in3=4490
> 
> LABEL_in4=+5V_SB
> GAIN_in4=0.048
> WARNHI_in4=5232
> WARNLO_in4=4752
> CRITHI_in4=5471
> CRITLO_in4=4490
> 
> LABEL_in5=VBAT
> GAIN_in5=0.024
> WARNHI_in5=3144
> WARNLO_in5=2832
> CRITHI_in5=3288
> CRITLO_in5=2688
> ```
> 
> I see the error as below
> 
> ```
> Jan 01 00:01:17 bmc7830 systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x401c0-it8728f\x402d.service: Main process exited, code=dumped, status=6/ABRT                          
> Jan 01 00:01:17 bmc7830 systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x401c0-it8728f\x402d.service: Failed with result 'core-dump'.                                          
> Jan 01 00:01:18 bmc7830 systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x401c0-it8728f\x402d.service: Scheduled restart job, restart counter is at 1.                          
> Jan 01 00:01:20 bmc7830 systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x401c0-it8728f\x402d.service: Main process exited, code=dumped, status=6/ABRT                          
> Jan 01 00:01:20 bmc7830 systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x401c0-it8728f\x402d.service: Failed with result 'core-dump'.                                          
> Jan 01 00:01:21 bmc7830 systemd[1]: xyz.openbmc_project.Hwmon@-ahb-apb-bus\x401e78a000-i2c\x2dbus\x401c0-it8728f\x402d.service: Scheduled restart job, restart counter is at 2.    
> ```
> 
> I can cat the value on /sys/bus/i2c/devices/6-002d/hwmon/hwmon0/in1_input
> I think driver is work.
> 
> Many thanks for you help.

wu790607, 2021-08-06:
Thanks your advice, I can see the sensors now!