openbmc/openbmc #3810: Files/directories were installed but not shipped in any package

yufangliwyc, 2021-08-14:
Hi,
I got a error in openbmc v2.11.0, when I add a new receipe in openbmc. This receipe works well in openbmc v2.9.0.

This is my bb file, I add this receipe in phosphor layer.

`SUMMARY = "DMC Manager to get nvme log by nvmemi over mctp over smbus protocol"
DESCRIPTION = "Daemon to monitor and report the log of NVMe drives by dmc protocol"
PR = "r1"
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://${COMMON_LICENSE_DIR}/MIT;md5=0835ade698e0bcf8506ecda2f7b4f302"

inherit meson pkgconfig
inherit systemd

DEPENDS += "sdbusplus"
DEPENDS += "phosphor-dbus-interfaces"
DEPENDS += "sdeventplus"
DEPENDS += "phosphor-logging"
DEPENDS += "nlohmann-json"
DEPENDS += "systemd"
DEPENDS += "mctp"

SRC_URI = "file://phosphor-dmc"

S = "${WORKDIR}/phosphor-dmc"
INSANE_SKIP_${PN} += "ldflags"
IMAGE_INSTALL += "phosphor-dmc"

SYSTEMD_SERVICE_${PN} = "xyz.openbmc_project.dmc.manager.service"
`
When I execute "bitbake obmc-phosphor-image" command, I got an error.

_ERROR: phosphor-dmc-1.0-r1 do_package: QA Issue: phosphor-dmc: Files/directories were installed but not shipped in any package:
  /lib
  /lib/systemd
  /lib/systemd/system
  /lib/systemd/system/xyz.openbmc_project.dmc.manager.service
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
phosphor-dmc: 4 installed and not shipped files. [installed-vs-shipped]
ERROR: phosphor-dmc-1.0-r1 do_package: Fatal QA errors found, failing task.
ERROR: Logfile of failure stored in: /home/liyufang/yufang/openBMC/openbmc_git/openbmc/build/tmp/work/cortexa7t2hf-neon-vfpv4-openbmc-linux-gnueabi/phosphor-dmc/1.0-r1/temp/log.do_package.22294
ERROR: Task (/home/liyufang/yufang/openBMC/openbmc_git/openbmc/meta-phosphor/recipes-phosphor/phosphor-dmc/phosphor-dmc.bb:do_package) failed with exit code '1'_

I tried many methods, such as 
1). Add this INSANE_SKIP_{$PN} += "installed-vs-shipped" in bb file
2). Add FILES_{$PN} in bb file
   FILES_${PN} += "/lib"
   FILES_${PN} += " /lib/systemd"
  FILES_${PN} += " /lib/systemd/system"
  FILES_${PN} += " /lib/systemd/system/xyz.openbmc_project.dmc.manager.service"

It does not work......

Can you help me to solve my problem? It is very helpful for me, if you can give me some suggestions....

williamspatrick, 2021-08-16:
https://lore.kernel.org/openbmc/YQ1FD5q8KbhbXVBK@heinlein/

edtanous, 2022-05-23:
Question was answered, and is concerning code that isn't on master for what looks like a custom daemon.  Closing.  Please reopen if this is an issue on openbmc master, and please consider asking questions on discord or the mailing list in the future.

closed 2022-05-23 by edtanous

alnoman612, 2022-08-14:
If you are using the new version of yocto this will help: FILES:${PN} ="name of the dirs is not shipping"; In old versions, it is FILES_${PN}.

So your answer should be: 
FILES:${PN}= "/lib /lib/systemd/"
or, FILES:${PN}= "/lib"   should fix it

fehimiltek, 2023-05-13:
> If you are using the new version of yocto this will help: FILES:${PN} ="name of the dirs is not shipping"; In old versions, it is FILES_${PN}.
> 
> So your answer should be: FILES:${PN}= "/lib /lib/systemd/" or, FILES:${PN}= "/lib" should fix it

you saved my life )) i couldnt see that for hours ))