openbmc/openbmc #3812: raspberrypi4-64: u-boot.bin is too large

jebr224, 2021-09-02:
DEBUG: obmc-phosphor-image-1.0-r0 do_generate_static: Compare needed=585520 available=524288 margin=-61232

Raspberrypi use exclusively sd cards (SDXC), so there I would think it is easy to give the u-boot partition the 61 KiB it needs, by moving the kernel offset up. I don't know how many other partitions would need to be adjusted.

Alternatively we could decrease the size of uboot. I don't know how managle an effor that would be.

Context:
https://stackoverflow.com/questions/69008701/image-is-too-large-keeps-on-happening-to-openbmc-image-for-raspberrypi-platfor/69034398#69034398

jebr224, 2021-09-02:
The change below is designed to resolve this issue.
https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/46507

Student2C0, 2021-09-07:
Hi. Jebr224,
Thank you for your analysis and resolution to this issue.

After I had this issue, I tried older releases and saw what happened to understand a change in which release could cause this. I picked up 2.8.0 release first and made it. I also made sure the 2.8.0 image actually was booted on my Raspberrypi4 platform.
My second choice was 2.7.0 to prove if I happened to make it on 2.8.0. However, I had different issues for 2.7.0 and failed.
Although I wanted to show 2.7.0 and 2.8.0 had no issues related to this  "image is too large", I made sure at least 2.8.0 worked without this.

I pasted some of logs below just for Raspberrypi users reference.

**Step#1: Git clone 2.8.0 to my local machine.**
```
yocto@yocto-VM:~/Rpi4$ git clone https://github.com/openbmc/openbmc.git -b 2.8.0
```

**Step#2: Set TEMPLATECONF for Raspberrypi.**
```
yocto@yocto-VM:~/Rpi4/openbmc$ export TEMPLATECONF=meta-evb/meta-evb-raspberrypi/conf
yocto@yocto-VM:~/Rpi4/openbmc$ echo $TEMPLATECONF
meta-evb/meta-evb-raspberrypi/conf
```

**Step#3: Set my environment by an openbmc script.**
```
yocto@yocto-VM:~/Rpi4/openbmc$ . openbmc-env 
```

**Step#4: Configure local.conf for my environment.**
```
yocto@yocto-VM:~/Rpi4/openbmc/build/conf$ diff local.conf local.conf.bk 
  1c1
  < MACHINE ??= "raspberrypi4"
  ---
  > MACHINE ??= "raspberrypi"
  18,22d17
  < 
  < DL_DIR ?= "/home/yocto/Yocto/downloads-2"
  < SSTATE_DIR ?= "/home/yocto/Yocto/sstate-cache-2"
  < IMAGE_FSTYPES += " rpi-sdimg"
```

**Step#5: Configure FLASH_SIZE for my environment.**
```
yocto@yocto-VM:~/Rpi4/openbmc/meta-phosphor/classes$ diff image_types_phosphor.bbclass image_types_phosphor.bbclass.bk 
  38c38
  < FLASH_SIZE ?= "131072"   <<<  FLASH_SIZE is changed from 32768 to 131072.
  ---
  > FLASH_SIZE ?= "32768"
```

**Step#6: Start bitbake**
```
yocto@yocto-VM:~/Rpi4/openbmc/build$ bitbake obmc-phosphor-image

  Snip the log and left the important logs here.

  Initialising tasks: 100% |########################################################| Time: 0:00:06
  Sstate summary: Wanted 1415 Found 0 Missed 1415 Current 0 (0% match, 0% complete)
  NOTE: Executing Tasks
  WARNING: expat-native-2.2.9-r0 do_fetch: Failed to fetch URL https://downloads.sourceforge.net/expat/expat-2.2.9.tar.bz2, attempting MIRRORS if available
  WARNING: bjam-native-1.72.0-r0 do_fetch: Failed to fetch URL https://dl.bintray.com/boostorg/release/1.72.0/source/boost_1_72_0.tar.bz2, attempting MIRRORS if available
  WARNING: netbase-1_6.1-r0 do_fetch: Failed to fetch URL http://ftp.debian.org/debian/pool/main/n/netbase/netbase_6.1.tar.xz, attempting MIRRORS if available
  WARNING: obmc-phosphor-sysd-1.0-r1 do_package_qa: QA Issue: obmc-phosphor-sysd: SRC_URI uses PN not BPN [src-uri-bad]
  Using an insecure image signing key!
>>  NOTE: Tasks Summary: Attempted 4108 tasks of which 2 didn't need to be rerun and all succeeded.
  Summary: There were 4 WARNING messages shown.
```

Now, I will try your suggestion on my machine and update the result here.

caohj1996, 2022-11-08:
Hi Student2C0, i want to cpmpile openbmc for my raspberry 4B but i get lots of error , could i get your version about your environment? like python version ,etc. thank you

Student2C0, 2022-11-12:
Hi caohj1996,
I built my openbmc Rpi-image a year ago. So, I am not sure if this could help you but I share my information with you below.

**#1: HW environment**
   Rpi-4B

**#2: Open BMC Release**
   2.9.0 and 2.8.0
   I could build both releases.

**#3: Python pkg**
  Please see the following output.
  I remembered I got a few errors related to some pkg and they all needed other pkg to build the image.
  You could find what you needed in build logs when you hit the error.

`tm@tm-VB1:~/Rpi4-2.9.0/openbmc$` apt list --installed | grep python

    WARNING:` apt does not have a stable CLI interface. Use with caution in scripts.

    libpython2-stdlib/focal,now 2.7.17-2ubuntu4 amd64
    libpython2.7-minimal/focal-updates,focal-security,now 2.7.18-1~20.04.1 amd64
    libpython2.7-stdlib/focal-updates,focal-security,now 2.7.18-1~20.04.1 amd64
    libpython3-dev/focal,now 3.8.2-0ubuntu2 amd64
    libpython3-stdlib/focal,now 3.8.2-0ubuntu2 amd64
    libpython3.8-dev/focal-updates,focal-security,now 3.8.10-0ubuntu1~20.04.4 amd64
    libpython3.8-minimal/focal-updates,focal-security,now 3.8.10-0ubuntu1~20.04.4 amd64
    libpython3.8-stdlib/focal-updates,focal-security,now 3.8.10-0ubuntu1~20.04.4 amd64
    libpython3.8/focal-updates,focal-security,now 3.8.10-0ubuntu1~20.04.4 amd64
    python-apt-common/focal-updates,focal-updates,now 2.0.0ubuntu0.20.04.7 all
    python-dbus/focal,now 1.2.16-1build1 amd64
    python-gi/focal,now 3.36.0-1 amd64
    python-pip-whl/focal-updates,focal-updates,now` 20.0.2-5ubuntu1.6 all
    python2-minimal/focal,now 2.7.17-2ubuntu4 amd64
    python2.7-minimal/focal-updates,focal-security,now 2.7.18-1~20.04.1 amd64
    python2.7/focal-updates,focal-security,now 2.7.18-1~20.04.1 amd64
    python2/focal,now 2.7.17-2ubuntu4 amd64
    python3-apport/focal-updates,focal-updates,focal-security,focal-security,now 2.20.11-0ubuntu27.24 all
    python3-apt/focal-updates,now 2.0.0ubuntu0.20.04.7 amd64
    python3-aptdaemon.gtk3widgets/focal-updates,focal-updates,focal-security,focal-security,now 1.1.1+bzr982-0ubuntu32.3 all
    python3-aptdaemon/focal-updates,focal-updates,focal-security,focal-security,now 1.1.1+bzr982-0ubuntu32.3 all
    python3-bcrypt/focal,now 3.1.7-2ubuntu1 amd64
    python3-blinker/focal,focal,now 1.4+dfsg1-0.3ubuntu1 all
    python3-brlapi/focal,now 6.0+dfsg-4ubuntu6 amd64
    python3-cairo/focal,now 1.16.2-2ubuntu2 amd64
    python3-certifi/focal,focal,now 2019.11.28-1 all
    python3-cffi-backend/focal,now 1.14.0-1build1 amd64
    python3-chardet/focal,focal,now 3.0.4-4build1 all
    python3-click/focal,focal,now 7.0-3 all
    python3-colorama/focal,focal,now 0.4.3-1build1 all
    python3-commandnotfound/focal-updates,focal-updates,now 20.04.6 all
    python3-cryptography/focal-updates,focal-security,now 2.8-3ubuntu0.1 amd64
    python3-cups/focal,now 1.9.73-3build1 amd64
    python3-cupshelpers/focal-updates,focal-updates,now 1.5.12-0ubuntu1.1 all
    python3-dateutil/focal,focal,now 2.7.3-3ubuntu1 all
    python3-dbus/focal,now 1.2.16-1build1 amd64
    python3-debconf/focal,focal,now 1.5.73 all
    python3-debian/focal,focal,now 0.1.36ubuntu1 all
    python3-defer/focal,focal,now 1.0.6-2.1 all
    python3-dev/focal,now 3.8.2-0ubuntu2 amd64
    python3-distro-info/focal,focal,now 0.23ubuntu1 all
    python3-distro/focal,focal,now 1.4.0-1 all
    python3-distupgrade/focal-updates,focal-updates,now 1:20.04.38 all
    python3-distutils/focal-updates,focal-updates,focal-security,focal-security,now 3.8.10-0ubuntu1~20.04 all
    python3-entrypoints/focal,focal,now 0.3-2ubuntu1 all
    python3-fasteners/focal,focal,now 0.14.1-2 all
    python3-future/focal,focal,now 0.18.2-2 all
    python3-gdbm/focal-updates,focal-security,now 3.8.10-0ubuntu1~20.04 amd64
    python3-gi-cairo/focal,now 3.36.0-1 amd64
    python3-gi/focal,now 3.36.0-1 amd64
    python3-httplib2/focal,focal,now 0.14.0-1ubuntu1 all
[... 106 lines cut ...]
    MACHINE              = "raspberrypi4"
    DISTRO               = "openbmc-phosphor"
    DISTRO_VERSION       = "nodistro.0"
    TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
    TARGET_FPU           = "hard"
    meta                 
    meta-poky            
    meta-oe              
    meta-networking      
    meta-perl            
    meta-python          
    meta-webserver       
    meta-security        
    meta-phosphor        
    meta-raspberrypi     = "master:feb21b20bf5e5820f28eb336f6190713bde6c00f"
    
    Initialising tasks: 100%     |#####################################################################################################| Time: 0:00:07
    Sstate summary: Wanted 5 Local 0 Network 0 Missed 5 Current 1385 (0% match, 99% complete)
    Removing 4 stale sstate objects for arch raspberrypi4: 100%     |##################################################################| Time: 0:00:00
    NOTE: Executing Tasks
    NOTE: Tasks Summary: Attempted 3835 tasks of which 3810 didn't need to be rerun and all succeeded.
    yocto@yocto-VM:~/Rpi4-2.9.0/openbmc/build$ 

Partition sample :
| SPL | u-boot | Boot FS | Root FS | User Data |

caohj1996, 2022-11-12:
> Hi caohj1996, I built my openbmc Rpi-image a year ago. So, I am not sure if this could help you but I share my information with you below.
>
> **#1: HW environment** Rpi-4B
>
> **#2: Open BMC Release** [...]

Hi, thanks for your help ,i have solved it,i will show you my solution next line as a exchange idea

caohj1996, 2022-11-12:
> Hi caohj1996, I built my openbmc Rpi-image a year ago. So, I am not sure if this could help you but I share my information with you below.
>
> **#1: HW environment** Rpi-4B
>
> **#2: Open BMC Release** [...]

Hi, thanks for your help ,i have solved it,i will show you my solution link next line as a exchanging my idea to you 
https://github.com/openbmc/openbmc/issues/3884

closed 2023-09-25 by williamspatrick