openbmc/openbmc #3813: The command 'ipmitool power off' doesn't work

wangmin239, 2021-09-03:
Hi folks,
'ipmitool power on', 'ipmitool power off', 'ipmitool power status' worked properly in the openbmc V2.6.0. The version of the openbmc was upgraded from V2.6.0 to V2.9.0. 'ipmitool power on' worked properly, but the command of the 'ipmitool power off' was unable to executed.   The below text snippet was printed  after executing  'ipmitool power on' .

> Sep 03 15:32:38 s2500 phosphor-host-state-manager[317]: Host state transition request of xyz.openbmc_project.State.Host.Transition.On
Sep 03 15:32:38 s2500 systemd[1]: unit_file_find_fragment: obmc-host-start@.target+0 → obmc-host-start@0.target
Sep 03 15:32:38 s2500 ipmid[293]: Transition request initiated successfully
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-mapper\x2dsubtree\x2dremove.slice.
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-obmc\x2denable\x2dhost\x2dwatchdog.slice.
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-op\x2dpower\x2dstart.slice.
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-op\x2dwait\x2dpower\x2don.slice.
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-phosphor\x2dfan\x2dcontrol\x2dinit.slice.
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-phosphor\x2dfan\x2dmonitor\x2dinit.slice.
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-phosphor\x2dfan\x2dpresence\x2dtach.slice.
Sep 03 15:32:38 s2500 systemd-journald[102]: Forwarding to syslog missed 1 messages.
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-phosphor\x2dreset\x2dhost\x2dreboot\x2dattempts.slice.
Sep 03 15:32:38 s2500 systemd[1]: Created slice system-phosphor\x2dwatchdog.slice.
Sep 03 15:32:38 s2500 systemd[1]: Reached target Host0 (Started).
Sep 03 15:32:38 s2500 systemd[1]: Starting mapper subtree-remove /xyz/openbmc_project/logging:xyz.openbmc_project.Logging.ErrorBlocksTransition...
Sep 03 15:32:38 s2500 systemd[1]: Starting mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition...
Sep 03 15:32:38 s2500 systemd[1]: Starting Wait for /xyz/openbmc_project/led/groups/power_on...
Sep 03 15:32:38 s2500 systemd[1]: Starting Wait for /xyz/openbmc_project/watchdog/host0...
Sep 03 15:32:38 s2500 systemd[1]: Starting Reset host reboot counter...
Sep 03 15:32:39 s2500 systemd[1]: Started Phosphor poweron watchdog.
Sep 03 15:32:39 s2500 systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/logging:xyz.openbmc_project.Logging.ErrorBlocksTransition.
Sep 03 15:32:39 s2500 systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Sep 03 15:32:39 s2500 systemd[1]: Started Wait for /xyz/openbmc_project/led/groups/power_on.
Sep 03 15:32:39 s2500 phosphor-watchdog[467]: Action Targets:
Sep 03 15:32:39 s2500 phosphor-watchdog[467]:   xyz.openbmc_project.State.Watchdog.Action.PowerCycle -> obmc-host-timeout@0.target
Sep 03 15:32:39 s2500 phosphor-watchdog[467]:   xyz.openbmc_project.State.Watchdog.Action.PowerOff -> obmc-host-timeout@0.target
Sep 03 15:32:39 s2500 phosphor-watchdog[467]:   xyz.openbmc_project.State.Watchdog.Action.HardReset -> obmc-host-timeout@0.target
Sep 03 15:32:39 s2500 systemd[1]: phosphor-reset-host-reboot-attempts@0.service: Succeeded.
Sep 03 15:32:39 s2500 systemd[1]: Finished Reset host reboot counter.
Sep 03 15:32:39 s2500 systemd[1]: Started Wait for /xyz/openbmc_project/watchdog/host0.
Sep 03 15:32:39 s2500 systemd[1]: mapper-subtree-remove@-xyz-openbmc\x5fproject-logging\x3Axyz.openbmc_project.Logging.ErrorBlocksTransition.service: Succeeded.
Sep 03 15:32:40 s2500 systemd[1]: mapper-subtree-remove@-xyz-openbmc\x5fproject-software\x3Axyz.openbmc_project.Software.ActivationBlocksTransition.service: Succeeded.
Sep 03 15:32:40 s2500 systemd[1]: Reached target Power0 On (Pre).
Sep 03 15:32:40 s2500 systemd[1]: Starting Start Watchdog0...
Sep 03 15:32:40 s2500 systemd[1]: Starting Assert power_on LED...
Sep 03 15:32:40 s2500 systemd[1]: Starting Start Power0...
Sep 03 15:32:40 s2500 systemd[1]: Starting Wait for Power0 to turn on...
Sep 03 15:32:40 s2500 phosphor-watchdog[467]: watchdog: enabled and started
Sep 03 15:32:40 s2500 phosphor-ledmanager[250]: Failed to get endpoints property
Sep 03 15:32:40 s2500 phosphor-ledmanager[250]: Error setting property for physical LED
Sep 03 15:32:40 s2500 systemd[1]: Finished Start Watchdog0.
Sep 03 15:32:40 s2500 power_control.exe[174]: wangmin set power 1
Sep 03 15:32:40 s2500 power_control.exe[174]: PowerControl: setting power up BMC_POWER_ON , pin No: 19
Sep 03 15:32:40 s2500 power_control.exe[174]: state: 1, value : 1, write : 0
Sep 03 15:32:40 s2500 systemd[1]: Finished Assert power_on LED.
Sep 03 15:32:40 s2500 systemd[1]: Finished Start Power0.
Sep 03 15:32:42 s2500 sel-logger[316]: A System Event Log is created
Sep 03 15:32:42 s2500 phosphor-fru-fault-monitor[315]: /xyz/openbmc_project/logging/entry/411 created
Sep 03 15:32:42 s2500 sel-logger[316]: A System Event Log is created
Sep 03 15:32:42 s2500 phosphor-fru-fault-monitor[315]: /xyz/openbmc_project/logging/entry/412 created
Sep 03 15:32:42 s2500 sel-logger[316]: A System Event Log is created
Sep 03 15:32:42 s2500 phosphor-fru-fault-monitor[315]: /xyz/openbmc_project/logging/entry/413 created
Sep 03 15:32:42 s2500 sel-logger[316]: A System Event Log is created
Sep 03 15:32:42 s2500 systemd[1]: Finished Wait for Power0 to turn on.
Sep 03 15:32:42 s2500 systemd[1]: Reached target Power0 On.
Sep 03 15:32:42 s2500 phosphor-fru-fault-monitor[315]: /xyz/openbmc_project/logging/entry/414 created
Sep 03 15:32:42 s2500 systemd[1]: Reached target Power0 (On).
Sep 03 15:32:42 s2500 sel-logger[316]: A System Event Log is created
Sep 03 15:32:42 s2500 systemd[1]: Started Phosphor Fan Control Initialization.
Sep 03 15:32:42 s2500 systemd[1]: Started Phosphor Fan Monitor Initialization.
Sep 03 15:32:42 s2500 systemd[1]: Started Phosphor Fan Presence Tach Daemon.
Sep 03 15:32:42 s2500 systemd[1]: Reached target Chassis0 (Power On).
Sep 03 15:32:42 s2500 systemd[1]: Reached target Start Host0 Minimum.
Sep 03 15:32:42 s2500 systemd[1]: Reached target Start Host0.
Sep 03 15:32:42 s2500 phosphor-fan-monitor[480]: Configuration loaded
Sep 03 15:32:43 s2500 systemd[1]: phosphor-fan-monitor-init@0.service: Succeeded.
Sep 03 15:32:43 s2500 systemd[1]: unit_file_find_fragment: obmc-fan-control-ready@.target+0 → obmc-fan-control-ready@0.target
[... 102 lines cut ...]
Sep 03 15:38:46 s2500 systemd[1]: Finished Reset host sensors.
Sep 03 15:40:14 s2500 systemd-journald[102]: Forwarding to syslog missed 49 messages.
Sep 03 15:40:50 s2500 systemd[1]: Starting Cleanup of Temporary Directories...
Sep 03 15:40:50 s2500 systemd-journald[102]: Forwarding to syslog missed 1 messages.
Sep 03 15:40:50 s2500 systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Sep 03 15:40:50 s2500 systemd[1]: Finished Cleanup of Temporary Directories.
Sep 03 15:41:24 s2500 systemd-journald[102]: Forwarding to syslog missed 2 messages.

 after execute 'ipmitool power off', the state of the chassis  remained 'On' as below.

> root@s2500:~# busctl --no-pager get-property xyz.openbmc_project.State.Chassis /xyz/openbmc_project/state/chassis0 xyz.openbmc_project.State.Chassis CurrentPowerState
s "xyz.openbmc_project.State.Chassis.PowerState.On"

> root@s2500:~#` busctl --no-pager introspect org.openbmc.control.Power  /org/openbmc/control/power0 org.openbmc.control.Power

NAME                      TYPE      SIGNATURE RESULT/VALUE FLAGS
.getPowerState            method    -         i            -
.setPowerState            method    i         -            -
.pgood                    property  i         1            emits-change
.pgood_timeout            property  i         10           emits-change writable
.state                    property  i         1            emits-change
.PowerGood                signal    -         -            -
.PowerLost                signal    -         -            -

Any  help will  be appreciate, thanks.

mdmillerii, 2021-09-08:
> Sep 03 15:33:11 s2500 phosphor-host-state-manager[317]: Change to Host State: xyz.openbmc_project.State.Host.HostState.Quiesced
You are not reaching a normal powered on state.  The system tried but failed to start your host processor and gave up.   Later the soft power off failed because the host was not running to recieve the soft power off command and echo it back as a completed hard power off request.

But I also see you are using `op-` packages.  This likely means you are using trying to start an OpenPOWER processor.  

Also note the default OpenPOWER platforms have a sequener that follows the hgh and low of a GPIO from the BMC as the desired ON or OFF state of the power rails.  It is not an ATX supply.

mdmillerii, 2021-09-08:
Please look at the the documentation in our Openbmc/docs repository and consider asking how to questions on the mailing list or via discord.

wangmin239, 2021-09-08:
Thanks your respond.

> Sep 03 15:33:11 s2500 phosphor-host-state-manager[317]: Change to Host State: xyz.openbmc_project.State.Host.HostState.Quiesced
> You are not reaching a normal powered on state.  The system tried but [...]
It seems like the function of the chassiss hard power off isn't called from the log text. I digged into the codes of the component named  'phosphor-state-mananger'，solved the issue that 'ipmitool power off' was unable to make effect by replacing 'obmc-host-shutdown@.target' with 'obmc-chassis-hard-poweroff@.target' replaced.  However, there is still a puzzle. I wander how bmc knows host os is running or not, thanks a lot.
> But I also see you are using `op-` packages. This likely means you are using trying to start an OpenPOWER processor.
>
> Also note the default OpenPOWER platforms have a sequener that follows the hgh [...]

wangmin239, 2021-09-08:
> Please look at the the documentation in our Openbmc/docs repository and consider asking how to questions on the mailing list or via discord.

I am sorry， I will refer to the requirements of the Openbcm/docs to submit a issue next time.

mdmillerii, 2021-09-08:
The IPMI power off is being treated as a soft power off request, where an interrupt to the OS is raised asking them to issue the power off request.  The power off over the host channel is treated as a hard power off request and actually triggers the power off request.   This is a bit of a hack and may be partly related to past incomplete understandings of the standard function when adopting the pre-existing concepts the from the OpenPOWER platform into the IPMI stack.

If you provide suitable replacements for the action scripts from the open power processor control you may have more success.

wangmin239, 2021-09-08:
@mdmillerii thanks. there is no an interrupt  for notifying OS on my platform. I have read README.MD[https://github.com/openbmc/phosphor-state-manager](url) to know that BMC sends hearbeat to host to determine if it's running. but I still have some questions that the what physical channel is used to deliver heartbeat.

mdmillerii, 2021-09-08:
usually its either IPMI kcs, bt, etc or PLDM but you can customize.

wangmin239, 2021-09-08:
> usually its either IPMI kcs, bt, etc or PLDM but you can customize.

Is there a daemon always running on the host os to respond the hearbeat sent by BMC?

mdmillerii, 2021-09-08:
It's a standard watchdog timer message but requires a firmware or os kernel to respond or shut it down (could be shut down from userspace if that's your IPMI stack.

wangmin239, 2021-09-08:
> It's a standard watchdog timer message but requires a firmware or os kernel to respond or shut it down (could be shut down from userspace if that's your IPMI stack.

Thank your answer. I typed the below command to wander knowing the respond of heartbeat, but failed to get a result.  I wander whether heartbeat is packeted as an IPMI command to be sent to host. Do you have some suggestion to help me ensure the guess ? Thanks.
> busctl --no-pager call xyz.openbmc_project.Control.Host /xyz/openbmc_project/control/host0 xyz.openbmc.Control.Host Execute s "xyz.openbmc_project.Control.Host.Command.Heartbeat"

> Call failed: The name is activetable

`constexpr auto MAPPER_BUSNAME = "xyz.openbmc_project.ObjectMapper";
constexpr auto MAPPER_PATH = "/xyz/openbmc_project/object_mapper";
constexpr auto MAPPER_INTERFACE = "xyz.openbmc_project.ObjectMapper";
constexpr auto CONTROL_HOST_DEFAULT_SVC = "xyz.openbmc_project.Control.Host";
constexpr auto CONTROL_HOST_PATH = "/xyz/openbmc_project/control/host0";
constexpr auto CONTROL_HOST_INTERFACE = "xyz.openbmc_project.Control.Host";
// Send hearbeat to host to determine if it's running
void sendHeartbeat(sdbusplus::bus::bus& bus)
{
    auto mapper = bus.new_method_call(MAPPER_BUSNAME, MAPPER_PATH,
                                      MAPPER_INTERFACE, "GetObject");

    mapper.append(CONTROL_HOST_PATH,
                  std::vector<std::string>({CONTROL_HOST_INTERFACE}));

    std::map<std::string, std::vector<std::string>> mapperResponse;

    try
    {
        auto mapperResponseMsg = bus.call(mapper);
        mapperResponseMsg.read(mapperResponse);
    }
    catch (const SdBusError& e)
    {
        log<level::INFO>("Error in mapper call for control host, use default "
                         "service",
                         entry("ERROR=%s", e.what()));
    }

    std::string host;
    if (!mapperResponse.empty())
    {
        log<level::DEBUG>("Use mapper response");
        host = mapperResponse.begin()->first;
    }
    else
    {
        log<level::DEBUG>("Use hard coded host");
        host = CONTROL_HOST_DEFAULT_SVC;
    }

    auto method = bus.new_method_call(host.c_str(), CONTROL_HOST_PATH,
                                      CONTROL_HOST_INTERFACE, "Execute");
    method.append(convertForMessage(Host::Command::Heartbeat).c_str());

    try
    {
        auto reply = bus.call(method);
    }
    catch (const SdBusError& e)
    {
        log<level::ERR>("Error in call to control host Execute",
                        entry("ERROR=%s", e.what()));
        throw;
    }

    return;
}`

Pravee207, 2022-12-15:
This issue is not reproducible on the openbmc version 2.11.0. 'ipmitool power on', 'ipmitool power off', 'ipmitool power status' worked properly on  openbmc V2.11.0 . These commands were tested on qemu. Do we need to test anything else ? ,If not I would recommend this issue to be closed.
[image]
Above image is about power status.

[image]
This one is power on.

[image]
This one is power off.

shubhabrata1979, 2023-01-03:
Can this ticket be closed as the issue does not seem to be occurring as per the logs updated by @Pravee207 ??

Thanks,

closed 2023-01-03 by williamspatrick