openbmc/openbmc #3814: Failed to start virtual media when KVM is already opened

leiyu-bytedance, 2021-09-10:
**Unexpected behavior you saw**
The WebUI fails to start virtual media when KVM page is already opened.

**Expected behavior**
The virtual media should be started without issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'KVM page', the KVM is working OK;
2. Click on 'Open in new tab' to make sure another KVM session is opened;
3. Go to "Virtual media" page, click "Add file", choose an ISO file, and click "Start"
4. See the error

**OpenBMC Information:**
This issue occurs on latest master revision, e.g. be8c511a94783c784d12827d8ebf2627dc800eed

**Additional context**
The journal logs shows below error:
```
Sep 10 09:22:46 g220a kernel:  nbd0: p1
Sep 10 09:22:46 g220a kernel: Mass Storage Function, version: 2009/09/11
Sep 10 09:22:46 g220a kernel: LUN: removable file: (no medium)
Sep 10 09:22:47 g220a kernel: udc-core: couldn't find an available UDC or it's busy
Sep 10 09:22:47 g220a bmcweb[1043]: nbd-proxy: state hook failed (rc: 1); exiting
Sep 10 09:22:48 g220a kernel: block nbd0: shutting down sockets
Sep 10 09:23:17 g220a kernel: block nbd0: Connection timed out, retrying (0/1 alive)
```

leiyu-bytedance, 2021-09-10:
The issue os related to the `state_hook` in https://github.com/openbmc/openbmc/blob/master/meta-phosphor/aspeed-layer/recipes-connectivity/jsnbd/jsnbd/state_hook#L51, that it uses the hard-coded portId `1e6a0000.usb-vhub:p4`.

When the KVM is opened, it will connect the input devices, and it will find a "spare" portId, and on g220a it somehow uses the `1e6a0000.usb-vhub:p4`, then it causes the issue for virtual media.

The `state_hook` needs to be fixed to dynamically select a unused port.

LeeTroy, 2021-09-22:
phosphor-misc/usb-ctl seems smarter, but nbd_proxy doesn't support it.

leiyu-bytedance, 2021-09-24:
> phosphor-misc/usb-ctl seems smarter, but nbd_proxy doesn't support it (yet?).

Yup, we need someone why is good at bash programming to fix the `nbd-proxy`

jk-ozlabs, 2021-09-28:
I'm happy to check out the `nbd-proxy` bits, but this sounds like it's all the `state_hook` & `usb-ctl` parts, right?

leiyu-bytedance, 2021-09-28:
> I'm happy to check out the `nbd-proxy` bits, but this sounds like it's all the `state_hook` & `usb-ctl` parts, right?

Yup, `state_hook` needs to be updated to choose a portId, like how `usb-ctl` does.

williamspatrick, 2021-09-28:
This came in yesterday and might be related?  https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/47187

mine260309, 2021-09-28:
> This came in yesterday and might be related? https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/47187

Checked the change but it's still using the hard-coded port id. Added comment in the gerrit.

LeeTroy, 2021-10-20:
I met similar situation but the other way around, where HID device were not able to be insert after establish virtual media:
```
{
        "_CMDLINE" : "/usr/bin/obmc-ikvm -v /dev/video0 -k /dev/hidg0 -p /dev/hidg1",
        "ERROR" : "basic_ios::clear: iostream error",
        "MESSAGE" : "Failed to connect HID gadget"
}
```

Still looking into `obmc-ikvm::ikvm_input::connect()`.

leiyu-bytedance, 2022-04-28:
https://github.com/openbmc/openbmc/blob/master/meta-phosphor/dynamic-layers/aspeed-layer/recipes-connectivity/jsnbd/jsnbd/state_hook#L51

The issue still exists.