openbmc/openbmc #3815: ast2600 rx_error problem

tungnguyen-ampere, 2021-09-15:
Run the iperf3 test on ast2600 evb board with **_latest_** openbmc source code, i can see the rx-error appears frequently:

```
root@ast2600-evb:~# ifconfig eth0 | grep errors | grep RX
          RX packets:5408 errors:0 dropped:37 overruns:0 frame:0
root@ast2600-evb:~# iperf3 -c 10.38.64.33 -p 5201 -t 60 -i 5 -R
Connecting to host 10.38.64.33, port 5201
Reverse mode, remote host 10.38.64.33 is sending
[  5] local 10.38.15.209 port 54486 connected to 10.38.64.33 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-5.00   sec   560 MBytes   939 Mbits/sec
[  5]   5.00-10.00  sec   561 MBytes   941 Mbits/sec
[  5]  10.00-15.00  sec   561 MBytes   941 Mbits/sec
[  5]  15.00-20.00  sec   552 MBytes   927 Mbits/sec
[  5]  20.00-25.00  sec   554 MBytes   930 Mbits/sec
[  5]  25.00-30.00  sec   561 MBytes   941 Mbits/sec
[  5]  30.00-35.00  sec   558 MBytes   937 Mbits/sec
[  5]  35.00-40.00  sec   559 MBytes   938 Mbits/sec
[  5]  40.00-45.00  sec   561 MBytes   941 Mbits/sec
[  5]  45.00-50.00  sec   561 MBytes   941 Mbits/sec
[  5]  50.00-55.00  sec   552 MBytes   927 Mbits/sec
[  5]  55.00-60.00  sec   559 MBytes   938 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-60.00  sec  6.55 GBytes   937 Mbits/sec  1570             sender
[  5]   0.00-60.00  sec  6.54 GBytes   937 Mbits/sec                  receiver

iperf Done.
root@ast2600-evb:~# ifconfig eth0 | grep errors | grep RX
          RX packets:4863005 errors:5 dropped:84 overruns:12 frame:55
root@ast2600-evb:~#

root@ast2600-evb:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP    MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Half 1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Half 1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        master-slave cfg: preferred slave
        master-slave status: slave
        Port: Twisted Pair
        PHYAD: 0
        Transceiver: external
        MDI-X: Unknown
        Link detected: yes

```

**OpenBMC Information:**
 - /etc/os-release: 
 `Linux ast2600-evb 5.10.60-09869b9-dirty-0d860ac #1 SMP Wed Sep 15 04:27:52 UTC 2021 armv7l GNU/Linux`
 - uname -a:
```
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v0.01.210907-ampere"
VERSION_ID=v0.01.210907-ampere-308-g9c7228271-dirty
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v0.01.210907-ampere"
BUILD_ID="v0.01.210907-ampere"
OPENBMC_TARGET_MACHINE="ast2600-evb"
```

LeeTroy, 2021-09-15:
Which soc revision you are using?

tungnguyen-ampere, 2021-09-15:
It's AST2600-A1 and the mac is configured with rgmii mode.

```
&mac0 {
	status = "okay";

	phy-mode = "rgmii";
	phy-handle = <&ethphy0>;

	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_rgmii1_default>;
}
```;

LeeTroy, 2021-09-29:
You will need to use `phy-mode = "rgmii-rxid";`.

closed 2023-09-25 by williamspatrick