openbmc/openbmc #3816: asrock-layer not compatible with the e3c246d4i

Mr-Antoine, 2021-09-15:
**Unexpected behavior you saw**
Getting an error when executing the following command `bitbake obmc-phosphor-image`:

`ERROR: Layer asrock-layer is not compatible with the core layer which only supports these series: honister (layer is compatible with hardknott)`

**Expected behavior**
The image should build.

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone git@github.com:openbmc/openbmc.git`
2. `cd openbmc`
3. `. setup e3c246d4i`
4. `bitbake obmc-phosphor-image`

**Screenshots**
[image]

**Additional context**
Commands executed on the e3c246d4i motherboard with Ubuntu 20.04.

I searched through stackoverflow, discord, and the issues on github but I couldn't find a way to fix my problem.

williamspatrick, 2021-09-15:
This stems from changes upstream Yocto did to recipe syntax a month or so back.  `meta-asrock` has been slow at getting updated.

It appears there is this pending commit to resolve the issue:
    https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/46640

If you want to test out that commit and give it a +1 that would probably help get it merged.

Mr-Antoine, 2021-09-15:
Thank you for the quick answer, I'll give it a try tomorrow and keep you updated.

zevweiss, 2021-09-15:
If you're interested in running mainline OpenBMC on e3c246d4i, note also these two additional pending patches which should improve usability significantly (enabling power control and serial console support):
 - https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/46211
 - https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/46642

closed 2021-09-15 by geissonator