openbmc/openbmc #3818: ast2600 emmc image not booting with QEMU

namo999, 2021-09-17:
**Note on Bugs**
Hi All,

I am able to boot AST2600 flash image with QEMU using "format=raw,if=mtd" argument.

But now I want to modify eMMC partition (8 GB in my case) and wanted to verify the same with QEMU, hence I changed partitions in "meta-aspeed/wic/emmc-aspeed.wks.in" file and generated obmc-phosphor-image-evb-ast2600-20210916181556.rootfs.wic and obmc-phosphor-image-evb-ast2600-20210916181556.rootfs.ext4 file.

The size of obmc-phosphor-image-evb-ast2600-20210916181556.rootfs.wic image is 8GB.

Now, giving the same image in QEMU with arguments "format=raw,if=sd" but it's not booting and gets stuck at the following message.
[image]

I have tried with .ext4 image as well but the same result.

Can someone help me with how to boot the eMMC image in QEMU?

amboar, 2021-09-17:
The eMMC image needs some preparation before it will work - the wic image data consists of just the MMC user partition. Roughly, you need:

```sh
# Set this to the appropriate board
board=ast2600-evb

# Fix this path t point to your build directory
fw_dir=.../tmp/deploy/images/${board}

wicxz="${fw_dir}/obmc-phosphor-image-${board}.wic.xz"
mmc="mmc-${board}.img"                                                         

dd of=$mmc if=/dev/zero bs=1M count=128                                               
dd of=$mmc if=${fw_dir}/u-boot-spl.bin conv=notrunc
dd of=$mmc if=${fw_dir}/u-boot.bin conv=notrunc bs=1K seek=64
xzdec $wicxz | dd of=$mmc conv=notrunc bs=1M seek=2
truncate --size 16G $mmc                                                              
```

namo999, 2021-09-17:
ok Thanks a lotfor the response.

Does that mean we need to create mmc image manually by including all the components like uboot, kernel and file system?

amboar, 2021-09-17:
@namo999  What you need to do is fix the paths in the script above to point to the directory containing the build artifacts, and then run it. Out the other side you will get a 16GB eMMC image that you can pass to qemu

namo999, 2021-09-17:
Thanks, @amboar for the quick response, really appreciated it.

I have created an eMMC image but still no success to boot with QEMU.

**eMMC image generation :**

+ board=evb-ast2600
+ fw_dir=/home/openbmc/build/tmp/deploy/images/evb-ast2600
+ wicxz=/home/openbmc/build/tmp/deploy/images/evb-ast2600/obmc-phosphor-image-evb-ast2600.wic.xz
+ mmc=mmc-evb-ast2600.img
+ dd of=mmc-evb-ast2600.img if=/dev/zero bs=1M count=128
128+0 records in
128+0 records out
134217728 bytes (134 MB, 128 MiB) copied, 0.115312 s, 1.2 GB/s
+ dd of=mmc-evb-ast2600.img if=/home/openbmc/build/tmp/deploy/images/evb-ast2600/u-boot-spl.bin conv=notrunc
85+0 records in
85+0 records out
43520 bytes (44 kB, 42 KiB) copied, 0.000747723 s, 58.2 MB/s
+ dd of=mmc-evb-ast2600.img if=/home/openbmc/build/tmp/deploy/images/evb-ast2600/u-boot.bin conv=notrunc bs=1K seek=64
402+1 records in
402+1 records out
411786 bytes (412 kB, 402 KiB) copied, 0.0027959 s, 147 MB/s
+ xzdec /home/openbmc/build/tmp/deploy/images/evb-ast2600/obmc-phosphor-image-evb-ast2600.wic.xz
+ dd of=mmc-evb-ast2600.img conv=notrunc bs=1M seek=2
dd: error writing 'mmc-evb-ast2600.img': No space left on device
0+218160 records in
0+218159 records out
1612939264 bytes (1.6 GB, 1.5 GiB) copied, 15.0308 s, 107 MB/s
+ truncate --size 16G mmc-evb-ast2600.img

**QEMU emulator:**

root@pc:/home/openbmc# ./qemu-system-arm -m 1G -M ast2600-evb -nographic -drive file=./mmc-evb-ast2600.img,format=raw,if=sd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,
hostname=qemu
qemu-system-arm: warning: Aspeed iBT has no chardev backend
qemu-system-arm: warning: nic ftgmac100.0 has no peer
qemu-system-arm: warning: nic ftgmac100.2 has no peer
qemu-system-arm: warning: nic ftgmac100.3 has no peer

amboar, 2021-09-17:
@namo999 how did you obtain the qemu binary? Did you build it yourself? Did you download it from somewhere?

namo999, 2021-09-17:
@amboar 
QEMU binary I downloaded from here,
https://jenkins.openbmc.org/job/latest-qemu-x86/lastSuccessfulBuild/artifact/qemu/build/qemu-system-arm

amboar, 2021-09-17:
Right, yes, the CI qemu is a bit out of date. I need to fix that.

namo999, 2021-09-17:
@amboar , Thanks for the prompt response.
1. Is there any workaround or an alternate solution, that I can boot eMMC image and verify partitions in the emulator?

2. I wanted to verify my eMMC partition table, only **meta-aspeed/wic/emmc-aspeed.wks.in** file change we need for partition modification correct?

amboar, 2021-09-17:
Two things:

1. You can build the aspeed-6.2 branch from https://github.com/legoater/qemu which is more up to date than the CI build

2. Build yourself a kernel from the dev-5.10 branch of https://github.com/openbmc/linux and then use the `-kernel` / `-dtb` / `-initrd` options along with `-drive type=sd,...`

amboar, 2021-09-20:
@namo999 it's hard to help you find the issue if you don't provide us with the command lines you used.

That said, it's only possible to boot the AST2600 with the "eMMC" controller slot and not the "SD" controller. The eMMC controller is integrated into the SoC model after the SD controller and the SD controller has two slots, so you need to make sure you have `index=2` in your dive options on the qemu command line for the image to be attached to the eMMC controller's single slot. For example `-drive type=sd,format=raw,index=2,file=...`

namo999, 2021-09-21:
@amboar 
Yes, giving the same arguments, it gets stuck after uboot SPL print.

root@devpc:/home/qemu/build# ./qemu-system-arm -m 1G -M ast2600-evb -nographic -drive file=../../openbmc/mmc-evb-ast2600.img,format=raw,if=sd,index=2 -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfw
d=:127.0.0.1:2443-:443,hostname=qemu
qemu-system-arm: warning: nic ftgmac100.1 has no peer
qemu-system-arm: warning: nic ftgmac100.2 has no peer
qemu-system-arm: warning: nic ftgmac100.3 has no peer

U-Boot SPL 2019.04 (Sep 20 2021 - 15:57:44 +0000)
Trying to boot from RAM

amboar, 2021-09-21:
@namo999 alright, progress at least! @LeeTroy might be able to help out from here. I haven't tried building/booting an ast2600-evb image recently. You might need an eMMC-specific u-boot config to make sure the u-boot FIT is being loaded from eMMC and not from the SPI-NOR, but that's just speculation on my part.

namo999, 2021-09-21:
ok, Thanks @amboar 

Hi @LeeTroy , it would be really appreciated if you can help me here.

LeeTroy, 2021-09-21:
I'll check it tomorrow, we're having a long weekend this week.

It seems you already bring up the u-boot SPL, so it would not be too much trouble. It might be relate to u-boot defconfig, could you post your u-boot defconfig with gist or pastebin?

LeeTroy, 2021-09-22:
@namo999 It seems the ast2600-evb machine's **hardware strap 1** doesn't config for **boot from emmc**, you'll need to patch QEMU for it.
```
diff --git a/hw/arm/aspeed.c b/hw/arm/aspeed.c
index 2206c33daa..14b3c309f3 100644
--- a/hw/arm/aspeed.c
+++ b/hw/arm/aspeed.c
@@ -158,7 +158,7 @@ struct AspeedMachineState {
         SCU_AST2400_HW_STRAP_BOOT_MODE(AST2400_SPI_BOOT))

 /* AST2600 evb hardware value */
-#define AST2600_EVB_HW_STRAP1 0x000000C0
+#define AST2600_EVB_HW_STRAP1 (0x000000C0 | AST26500_HW_STRAP_BOOT_SRC_EMMC)
 #define AST2600_EVB_HW_STRAP2 0x00000003

 /* Tacoma hardware value */
```

With that patch, I could run into u-boot:
```
[troylee@troylee-vm]$ qemu-system-arm -m 1G -M ast2600-evb -nographic -drive file=mmc-ast2600-evb.img,format=raw,if=sd,index=2 -net nic 
qemu-system-arm: warning: hub 0 is not connected to host network
qemu-system-arm: warning: nic ftgmac100.1 has no peer
qemu-system-arm: warning: nic ftgmac100.2 has no peer
qemu-system-arm: warning: nic ftgmac100.3 has no peer

U-Boot SPL 2019.04-v00.04.06 (Aug 11 2021 - 19:13:16 +0800)
Trying to boot from MMC1
## Checking hash(es) for Image uboot ... OK
## Checking hash(es) for Image fdt ... OK

U-Boot 2019.04-v00.04.06 (Aug 11 2021 - 19:13:16 +0800)

SOC: AST2600-A3
eSPI Mode: SIO:Enable : SuperIO-2e
Eth: MAC0: RGMII, MAC1: RGMII, MAC2: RGMII, MAC3: RGMII
Model: AST2600 EVB
DRAM:  already initialized, 1008 MiB (capacity:1024 MiB, VGA:64 MiB, ECC:off)
QEMU: Terminated
```

@amboar Is there a better way to supply HWSTRAP register into qemu model?

amboar, 2021-09-22:
@LeeTroy maybe if we have `-drive type=sd,index=2` on the qemu commandline we update the strapping to set the `AST2600_HW_STRAP_BOOT_SRC_EMMC` bit dynamically? Can you send a (RFC?) patch to the openbmc list along those lines?

LeeTroy, 2021-09-22:
Sounds good. I'll sent out a RFC patch for this.

LeeTroy, 2021-09-22:
FWIW, I only modify the following machine config to build emmc image for evb-ast2600:
```
diff --git a/meta-aspeed/conf/machine/evb-ast2600.conf b/meta-aspeed/conf/machine/evb-ast2600.conf
index d3ee47b9c..58bdecf0d 100644
--- a/meta-aspeed/conf/machine/evb-ast2600.conf
+++ b/meta-aspeed/conf/machine/evb-ast2600.conf
@@ -1,5 +1,8 @@
 KERNEL_DEVICETREE = "aspeed-ast2600-evb.dtb"
-UBOOT_MACHINE = "ast2600_openbmc_spl_defconfig"
+UBOOT_MACHINE = "ast2600_openbmc_spl_emmc_defconfig"
+UBOOT_SIGN_ENABLE = "1"
+UBOOT_FITIMAGE_ENABLE = "1"
+#UBOOT_MACHINE = "ast2600_openbmc_spl_defconfig"
 UBOOT_DEVICETREE = "ast2600a1-evb"
 SPL_BINARY = "spl/u-boot-spl.bin"
```

Then taking the u-boot-spl.bin, u-boot.bin and obmc-phosphor-image-evb-ast2600.wic.xz to pack emmc image by https://github.com/openbmc/openbmc/issues/3818#issuecomment-921469623

namo999, 2021-09-22:
Hi @LeeTroy 
That's great, qemu patch worked and now able to boot from eMMC, thank you very much for your quick help.

@amboar Thank you so much for your time and effort.

closed 2021-09-22 by namo999

shitalkumar-gandhi, 2021-09-29:
@LeeTroy 
Are you able to boot into rootfs?
It is always landing in ramfs.

mdmillerii, 2021-09-29:
@amboar The prep work above seems to be prepending 128MB to the WIC image to

legoater, 2021-09-30:
It's 2 partitions of 1MB. See commit https://github.com/legoater/qemu/commit/4416722549e50b362ad08fb10812de691436aa53  or directly the model file : https://github.com/legoater/qemu/blob/aspeed-6.2/hw/sd/sd.c#L833

@amboar has included a tool to build the emmc image : https://github.com/openbmc/openbmc-tools/blob/master/prepare-emmc-qemu/prepare-emmc-qemu

shitalkumar-gandhi, 2021-09-30:
@legoater @LeeTroy 
eMMC image created with script https://github.com/openbmc/openbmc-tools/blob/master/prepare-emmc-qemu/prepare-emmc-qemu giving following error and landing to ramfs in QEMU.

[image]

How to get into rootfs?
Thanks.

mdmillerii, 2021-09-30:
@legoater that's a start.   I was hoping for user documentation and preferably something showing upstream qemu has accepted the image format.

legoater, 2021-09-30:
@mdmillerii This is not upstream. Only in my branch https://github.com/legoater/qemu/commits/aspeed-6.2. Upstreaming will require considerable changes in the `sd` model to introduce different backends  to handle commands depending on  the media.
See http://patchwork.ozlabs.org/project/qemu-devel/list/?series=250563