openbmc/openbmc #3819: SDK: package openssh-8.7p1-r0.arm1176jzs conflicts with dropbear provided by dropbear-2020.81-r0.arm1176jzs

geissonator, 2021-09-20:
SDK is failing to build up at https://jenkins.openbmc.org/job/latest-master-sdk/

```
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - package openssh-8.7p1-r0.arm1176jzs conflicts with dropbear provided by dropbear-2020.81-r0.arm1176jzs
 *   - package openssh-dev-8.7p1-r0.arm1176jzs requires openssh = 8.7p1-r0, but none of the providers can be installed
 *   - package packagegroup-core-ssh-dropbear-1.0-r1.all requires dropbear, but none of the providers can be installed
 *   - conflicting requests
 *   - problem with installed package packagegroup-core-ssh-dropbear-1.0-r1.all
 * 
 * Solution 1:
 *   - allow deinstallation of packagegroup-core-ssh-dropbear-1.0-r1.all

 *   - do not ask to install a package providing dropbear-dev

 *   - do not ask to install a package providing packagegroup-core-ssh-dropbear-dev

 * Solution 2:
 *   - do not ask to install a package providing openssh-dev
```

One of these three commits is causing it:
```
    dropbear: use OpenSSH's sftp-server (commit: 26dbcdb) (details / githubweb)
    Add repo tests to main repository (commit: d5cea14) (details / githubweb)
    Implement a test for patch compliance (commit: 7f8954e) (details / githubweb)
```
26dbcdb is the most likely the culprit

mkj, 2021-09-22:
It looks like OpenSSH package needs to conflict only with the `-ssh` and `-sshd` packages, not plain `openssh` package. 
https://github.com/openbmc/openbmc/blob/71e95da6057fbd20a6c3f6a3d6a0e471f8fb3a53/poky/meta/recipes-connectivity/openssh/openssh_8.7p1.bb#L175

I'm testing that now here. I guess @zevweiss wouldn't have seen a problem with a normal build because it's just `bitbake -c populate_sdk` that pulls in openssh-dev -> openssh and conflicts.

mkj, 2021-09-22:
Removing the `RCONFLICTS:${PN}` isn't sufficient, I had missed that openssh depends on openssh-sshd. Some packages use `openssh` to pull in OpenSSH client+server+scp+etc.

I've pushed a potential fix  https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/47089 , removing the openssh-dev package since it seems unused.

zevweiss, 2021-09-22:
Thanks @mkj!  And yeah, I'd actually never even tried building the SDK previously (and it looks like CI doesn't check it either).

closed 2021-09-22 by geissonator

leeeening, 2023-09-25:
Sorry, has this problem been resolved

williamspatrick, 2023-09-25:
> Sorry, has this problem been resolved

Do you have evidence it isn’t?

leeeening, 2023-09-25:
邮件收到，谢谢

leeeening, 2023-09-26:
> > Sorry, has this problem been resolved
> 
> Do you have evidence it isn’t?

You may have misunderstood my meaning. I would like to ask for advice on how to enable openssh replace dropbear?

williamspatrick, 2023-09-26:
Dropbear is currently required if you want to use obmc-console. If you don’t want to use obmc-console, you can look under meta-facebook for examples on switching to openssh as that is what we use. The project as a whole has not switched.

leeeening, 2023-12-27:
> Dropbear is currently required if you want to use obmc-console. If you don’t want to use obmc-console, you can look under meta-facebook for examples on switching to openssh as that is what we use. The project as a whole has not switched.

On Meta Facebook, I see that Dropbear is also used and OpenSSH is not used.

williamspatrick, 2023-12-27:
> On Meta Facebook, I see that Dropbear is also used and OpenSSH is not used.

I'm not sure where you are seeing that.  Can you point me to what you are looking at?

The repository facebook/openbmc does not use Dropbear.  In this repository meta-facebook/meta-bletchley uses OpenSSH.

leeeening, 2023-12-27:
邮件收到，谢谢

leeeening, 2023-12-28:
[image]

**I found a lot of examples of using dropbear on meta-facebook, but I couldn't find any examples of using OpenSSH**

leeeening, 2023-12-28:
> > On Meta Facebook, I see that Dropbear is also used and OpenSSH is not used.
> 
> I'm not sure where you are seeing that. Can you point me to what you are looking at?
> 
> The repository facebook/openbmc does not use Dropbear. In this repository meta-facebook/meta-bletchley uses OpenSSH.

**I found a lot of examples of using dropbear on meta-facebook, but I couldn't find any examples of using OpenSSH**

leeeening, 2023-12-28:
> > On Meta Facebook, I see that Dropbear is also used and OpenSSH is not used.
> 
> I'm not sure where you are seeing that. Can you point me to what you are looking at?
> 
> The repository facebook/openbmc does not use Dropbear. In this repository meta-facebook/meta-bletchley uses OpenSSH.

[image]

I found a lot of examples of using dropbear on meta-facebook, but I couldn't find any examples of using OpenSSH.

williamspatrick, 2023-12-28:
> I found a lot of examples of using dropbear on meta-facebook, but I couldn't find any examples of using OpenSSH.

https://github.com/openbmc/openbmc/blob/d97a0174cfe7fc881fc1b34f39ffbab7ee351b7f/meta-facebook/recipes-phosphor/images/fb-phosphor-image.inc#L10

leeeening, 2024-01-11:
> > I found a lot of examples of using dropbear on meta-facebook, but I couldn't find any examples of using OpenSSH.
> 
> https://github.com/openbmc/openbmc/blob/d97a0174cfe7fc881fc1b34f39ffbab7ee351b7f/meta-facebook/recipes-phosphor/images/fb-phosphor-image.inc#L10

I would like to ask how to add sshd to Systemd and start it after successfully installing OpenSSH. Thank you.