openbmc/openbmc #3820: Server can't fully start after flash

Mr-Antoine, 2021-09-21:
**Unexpected behavior you saw**
After flashing the e3c246d4i motherboard with the OpenBMC image built with the commit 605c37cb989a95c02633fcb93efb45102781b4bb the BMC successfully starts but the server can't fully start.

I am able to communicate with the server using the OpenBMC Tool from IBM and with Redfish but the server gets stuck at the PEI Intel Reference Code 9A ("System Initializing ...     9A").

I tried using an image built with the older commit e8963ad91976924003cb17c0bf966619b3d87307 but it didn't work.

**Expected behavior**
The server is supposed to start.

**To Reproduce**
Steps to reproduce the behavior:
1. '. setup e3c246d4i'
2. 'bitbake obmc-phosphor-image'
3. 'cd tmp/deploy/images/e3c246d4i/'
4. Transfer the socflash to the current folder
5. 'chmod 777 socflash_x64'
6. 'sudo socflash_x64 flash-e3c246d4i backup.bin'

**Additional context**
The server currently has the latest BIOS available (2.10). It also currently has the bare minimum, I've removed anything that is not necessary for the server to start.

I havn't been able to try the image with qemu since this motherboard isn't supported by it.

[Socflash](https://www.touslesdrivers.com/index.php?v_page=23&v_code=60176&v_langue=en) download link.
[OpenBMC Tool](https://www14.software.ibm.com/support/customercare/sas/f/lopdiags/scaleOutLCdebugtool.html#OpenBMC) download link.

zevweiss, 2021-09-21:
Host power-control on e3c246d4i isn't yet merged in OpenBMC master, but there's a pending review currently awaiting approval in gerrit that should make it work: https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/46642 (if you could test that out and verify that it works for you that'd be useful feedback).

Note though that even with that change in place, the physical power button probably won't work, but power control through the BMC (shell commands or redfish requests/web UI) should.

zevweiss, 2021-09-21:
Oh wait, just realized I misread what you wrote -- the host is powering on (though I'm curious how that's getting triggered with what's in commit 605c37c), but the boot sequence is hanging?

Also, can you elaborate a bit on what exactly you mean by this?
> It also currently has the bare minimum, I've removed anything that is not necessary for the server to start.

Mr-Antoine, 2021-09-21:
`the host is powering on (though I'm curious how that's getting triggered with what's in commit 605c37c)`

I'm powering on the server with a power button, I didn't have time to try the commit because I tried to fix this issue first.

The boot sequence is hanging indeed, the server gets stuck at the PEI Intel Reference Code 9A ("System Initializing ... 9A" is what my monitor displays).

I don't know if I didn't flash it properly, or if I didn't build the image properly but the motherboard worked before flashing. 

Tomorrow I'll send a photo showing what my monitor displays.

`Also, can you elaborate a bit on what exactly you mean by this?`

What I meant was that I removed all non-ethencial devices such as the keyboard, SSDs, the other stick of ram etc in case it creates a conflict for some reason (I met motherboards who refused to boot because the keyboard is plugged in). The only non-ethencial device that I kept is the monitor to see if the server boots properly.

zevweiss, 2021-09-22:
> I'm powering on the server with a power button

Interesting -- with the current state of the code I wouldn't have expected the power button to do anything...out of curiosity, has the board had a full power cut/restore since the transition from the factory firmware to OpenBMC?  (Wondering if there might be some internal SoC state that survived a BMC reboot without getting reset.)

> The boot sequence is hanging indeed, the server gets stuck at the PEI Intel Reference Code 9A

Okay, hmmm...that sounds vaguely familiar, like I might have encountered similar behavior at some point while developing/debugging the port, but unfortunately at the moment I can't remember the details or how I eventually resolved it.  In the meantime while I try to dredge something up on that front though, have you tried removing power from the board entirely (this is apparently called a "flea drain" in some circles), and/or doing a long press-and-hold of the host power signal?  For the latter, you can bypass the question of whether or not the power button is really "connected" by manually poking the GPIO from a shell on the BMC, e.g. something like:

```
root@e3c246d4i:~# gpioset -m time -s 10 /dev/gpiochip0 25=0 && gpioset -m time -s 1 /dev/gpiochip0 25=1
```

(That will do the equivalent of pressing the power button for 10 seconds and then releasing it.)

A potential caveat here is that my employer (for whom I'm working on this port) uses a custom BIOS that's slightly different than the publicly available one, so it's _possible_ that yours might behave slightly differently (though at this point I'd guess that's probably not likely a factor here).

> What I meant was that I removed all non-ethencial devices

Got it, thanks for the clarification -- I wouldn't expect any of those devices to cause any particular problems, but them being absent shouldn't hurt anything either of course.  Re: the monitor though, does the OpenBMC iKVM (Control -> KVM in the web UI) work for you?  It might take a moment to start working after you power on the host, but should come on at some point during the POST phase (before you reach the 9A hang I'd expect).

Another potentially relevant tip: you can also watch the POST codes issued by the host in real time by running the `snooper` command from the BMC shell.

Mr-Antoine, 2021-09-22:
> has the board had a full power cut/restore since the transition from the factory firmware to OpenBMC?

The board did get a full power cut/restore since the transition.

> have you tried removing power from the board entirely

I removed the power from the board entirely a couple of times.

> doing a long press-and-hold of the host power signal?

I tried it this morning, it didn't change anything.

`root@e3c246d4i:~# gpioset -m time -s 10 /dev/gpiochip0 25=0 && gpioset -m time -s 1 /dev/gpiochip0 25=1`

This command is quite interesting, the command works but once I use it the power button doesn't work anymore (besides the led).

I had to remove the power from the board entirely to be able to use the power button again. Tried it a second time just to be sure, had the same results.

> A potential caveat here is that my employer (for whom I'm working on this port) uses a custom BIOS that's slightly different than the publicly available one, so it's possible that yours might behave slightly differently (though at this point I'd guess that's probably not likely a factor here).

We use a custom BIOS as well, I thought my issue might be related to that which is why I flashed my BIOS to the public one. It did not solve the issue but there's one interesting fact about it.

With the custom BIOS I was able to go further with the image built with the commit `e8963ad91976924003cb17c0bf966619b3d87307`. I'd get stuck at the `Update FRU System device`. I'd even be able to boot if I powered on the server immidiatly once I restored the power to the board because for some reason if I did that it wouldn't try to update the FRU system device. Restarting the server (if it managed to boot properly) would get me stuck at the `Update FRU System device`.

> does the OpenBMC iKVM (Control -> KVM in the web UI) work for you?

I currently don't have a web UI which is why I'm using a monitor on the server. Does this mean I did something wrong ?

> you can also watch the POST codes issued by the host in real time by running the snooper command from the BMC shell

I ran the `snooper` command this morning, I waited until the boot sequence hangs, here's the ouput :

`recv: 0x1
recv: 0x3
recv: 0x4
recv: 0x2
recv: 0x3
recv: 0x4
recv: 0xc
recv: 0x2
recv: 0x0
recv: 0x0
recv: 0x7f
recv: 0x0
recv: 0xf
recv: 0x3
recv: 0x23
recv: 0x7f
recv: 0x15
recv: 0x19
recv: 0x0
recv: 0x2
recv: 0x7f
recv: 0x3
recv: 0x48
recv: 0x49
recv: 0x4a
recv: 0x4d
recv: 0x15
recv: 0x52
recv: 0x20
recv: 0x2f
recv: 0x3f
recv: 0x4f
recv: 0xa
recv: 0x6f
recv: 0x0
recv: 0x8
recv: 0x10
recv: 0x1f
recv: 0x30
recv: 0x40
recv: 0x42
recv: 0x50
recv: 0x51
recv: 0x52
recv: 0x7f
recv: 0x0
recv: 0x1b
recv: 0x1c
recv: 0x1e
recv: 0x21
recv: 0x22
recv: 0x23
recv: 0x24
recv: 0x27
recv: 0x5e
recv: 0x43
recv: 0x26
recv: 0x71
recv: 0x72
recv: 0x71
recv: 0x5c
recv: 0x61
recv: 0x60
recv: 0x5d
recv: 0x1
recv: 0x50
recv: 0x21
recv: 0x55
recv: 0x0
recv: 0x7f
recv: 0x0
recv: 0x9
recv: 0x7f
recv: 0x0
recv: 0x4
recv: 0x6
recv: 0x8
recv: 0xa
recv: 0xc
recv: 0xd
recv: 0x7
recv: 0xe
recv: 0x13
recv: 0x14
recv: 0x15
recv: 0x0
recv: 0x20
recv: 0x32
recv: 0x15
recv: 0x16
recv: 0x18
recv: 0x19
recv: 0x23
recv: 0x25
recv: 0x40
recv: 0x42
recv: 0x4f
recv: 0x41
recv: 0x42
recv: 0x47
recv: 0x3
recv: 0x64
recv: 0x6a
recv: 0x71
recv: 0x7f
recv: 0x4f
recv: 0x60
recv: 0x61
recv: 0x90
recv: 0x7f
recv: 0x9a`

zevweiss, 2021-09-22:
Hmm, I wonder if your board might be strapped slightly differently than the ones I've been dealing with -- which variant of the board are you using?  (e.g. the `-2T` with the on-board Intel X550 NIC, or the `-NL` with no on-board NIC)

Can you try the following:

 1. Assuming you haven't modified your u-boot environment from the defaults, run `fw_setenv bootargs "console=ttyS4,115200n8 root=/dev/ram rw mem.devmem=1"` (i.e. add `mem.devmem=1` to your OpenBMC kernel command-line)
 2. Reboot the BMC
 3. Run `devmem 0x1e6e2070`
 4. Run the `gpioset` command that makes the power button stop working
 5. Run `devmem 0x1e6e2070` again

and report the output from the two `devmem` commands?  (You'll want to undo the `mem.devmem=1` bit for any production use for security reasons, but it's a useful debug setting.)

Also, do you happen to be set up with a serial adapter you can attach to the `BMC_DEBUG` header on the board to access the BMC's serial console?  That can be useful for things like intercepting u-boot's default boot sequence and dropping into its command prompt instead; if you can do that, `md.l 0x1e6e2070 1` would be the equivalent of the `devmem` command above from the u-boot prompt.

> I currently don't have a web UI which is why I'm using a monitor on the server. Does this mean I did something wrong ?

Hmm, that's not expected -- do you get anything at all if you point a browser at https://$BMC_IP_ADDRESS?  (Note the https; OpenBMC's bmcweb server doesn't listen on port 80.)  If not, does `journalctl -t bmcweb` report anything interesting?

(Also, for github markdown you can use triple-backticks on their own line before and after the enclosed text to preserve its line structure instead of squashing it all together on one line.)

Mr-Antoine, 2021-09-22:
> which variant of the board are you using?

I have the -2T variant.

> and report the output from the two devmem commands?

```
root@e3c246d4i:~# fw_setenv bootargs "console=ttyS4,115200n8 root=/dev/ram rw mem.devmem=1"
root@e3c246d4i:~# devmem 0x1e6e2070
0x7125F256
# Reboot the BMC
root@e3c246d4i:~# gpioset -m time -s 10 /dev/gpiochip0 25=0 && gpioset -m time -s 1 /dev/gpiochip0 25=1
root@e3c246d4i:~# devmem 0x1e6e2070
0x7105F256
root@e3c246d4i:~# fw_setenv bootargs "console=ttyS4,115200n8 root=/dev/ram rw mem.devmem=0"
```

> Also, do you happen to be set up with a serial adapter you can attach to the BMC_DEBUG header on the board to access the BMC's serial console?

Unfortunately we don't have a serial adapter to attach to the BMC_DEBUG header on the board.

> do you get anything at all if you point a browser at https://$BMC_IP_ADDRESS?

Having `https://$BMC_IP_ADDRESS` instead of `http://$BMC_IP_ADDRESS` fixed it, I never encountered a problem like this, thanks!

> does the OpenBMC iKVM (Control -> KVM in the web UI) work for you?

So, to answer your question earlier yes the iKVM works.

> (Also, for github markdown you can use triple-backticks on their own line before and after the enclosed text to preserve its line structure instead of squashing it all together on one line.)

I edited my previous comment to fix it, thanks for the tip!

mdmillerii, 2021-09-22:
We have a feature in the works to redirect http to https but only for the default base URI `/`.   With our focus on security, even that will be opt in by the builder and can be disabled by the administrator as it's subject to mitm attacks.

[HTTP redirect design: Redirect HTTP requests to HTTPS](https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/24173) has the draft requirements, [Enable HTTP redirects](https://gerrit.openbmc-project.xyz/c/openbmc/bmcweb/+/35265) is the current community direction.  In addition there is [an awk script](https://github.com/openbmc/phosphor-misc/tree/master/http-redirect) that was written before the bmcweb community had found the approach above and sized it as too much of a maintenance burden.  The script has been tested but no bitbake recipe exists as we expect it to be obsolete and it doesn't have any resource consumption limits (some of which could be added via systemd unit).

zevweiss, 2021-09-22:
> I have the -2T variant.

OK, good to know -- I'm not aware of any particular difference that variation would make at the moment, but for what it's worth I've been developing on the `-NL`.

> ```
> root@e3c246d4i:~# fw_setenv bootargs "console=ttyS4,115200n8 root=/dev/ram rw mem.devmem=1"
> root@e3c246d4i:~# devmem 0x1e6e2070
> 0x7125F256
> # Reboot the BMC
> root@e3c246d4i:~# gpioset -m time -s [...]

OK, well I guess that explains the power button's behavior then -- bit 22 of that register controls the SoC's pass-through functionality between the pin that the power button's signal runs to and the pin that drives that actual power signal to the host.  It looks like your board (unlike mine) has that bit enabled in its strapping, so you have that functionality on when the BMC boots (but the kernel's pinctrl subsystem is resetting it once you touch the GPIO manually).  For what it's worth, if you want to manually re-enable that without going through a BMC reboot, `devmem 0x1e6e2070 0x200000` should turn the pass-through back on.

> So, to answer your question earlier yes the iKVM works.

Alright, that's good to hear -- so at least you can probably remove the monitor then (if you want).

Back to the host boot hang though, I'd be interested to see what's going on the KCS interface, so let's try:

 1. Build an strace binary -- in your e3c246d4i OpenBMC build directory, run `bitbake strace`
 2. `scp` the strace binary (`tmp/work/arm1176jzs-openbmc-linux-gnueabi/strace/5.13-r0/package/usr/bin/strace`) to the BMC
 3. Find the pid of the kcsbridged process on the BMC (e.g. `ps | grep kcsbridged`)
 4. Run `strace -xx -s256 -e read,write -p $KCSBRIDGED_PID -o /tmp/kcs.txt` and leave it running
 5. Boot the host
 6. Once the host hits the point where it hangs, stop the strace and post the `kcs.txt` file somewhere (e.g. pastebin or a github gist or something)

zevweiss, 2021-09-23:
And for another data point, I just tried downloading and flashing the public 2.10 BIOS (presumably the same one you're using, with a sha1sum of a7a5348d4e39ae6fdcaee625f445b416d6395fb2) and was able to boot the host without an issue.

Mr-Antoine, 2021-09-23:
> Back to the host boot hang though, I'd be interested to see what's going on the KCS interface, so let's try:
>
> 1. Build an strace binary -- in your e3c246d4i OpenBMC build directory, run bitbake [...]

I've been able to build the strace binary but I've been unable to use it.

> 2. scp the strace binary (tmp/work/arm1176jzs-openbmc-linux-gnueabi/strace/5.13-r0/package/usr/bin/strace) to the BMC

`tmp/work/arm1176jzs-openbmc-linux-gnueabi/strace/5.13-r0/package/usr/bin/strace` doesn't exist.

`tmp/work/arm1176jzs-openbmc-linux-gnueabi` exists but `tmp/work/arm1176jzs-openbmc-linux-gnueabi/strace` doesn't. The closest I've been able to find using the `find` command were :

```
- tmp/work/arm1176jzs-openbmc-linux-gnueabi/linux-libc-headers/5.13-r0/linux-5.13/tools/perf/trace/strace
- home/test/Download/openbmc/poky/build/tmp/work/core2-64-poky-linux/strace/5.13-r0/package/usr/bin/strace
```

I tried some other strace that I could find but it would do nothing (it would just get me back to the command line) or I'd get the error `-sh: /usr/bin/strace: cannot execute binary file: Exec format error`.

I've done some research and from what I've seen this error occurs when you try to execute a x86_64 program on an ARM or vice versa. I doubt that it'll help but apparently from `cat proc/cpuinfo` the BMC is an ARM.

> I just tried downloading and flashing the public 2.10 BIOS (presumably the same one you're using, with a sha1sum of a7a5348d4e39ae6fdcaee625f445b416d6395fb2)

My 2.10 BIOS also has a sha1sum of `a7a5348d4e39ae6fdcaee625f445b416d6395fb2`.

If you didn't do any modification to your BMC would you mind sharing it so that I can try it ?

zevweiss, 2021-09-23:
> `tmp/work/arm1176jzs-openbmc-linux-gnueabi/strace/5.13-r0/package/usr/bin/strace` doesn't exist.
>
> `tmp/work/arm1176jzs-openbmc-linux-gnueabi` exists but `tmp/work/arm1176jzs-openbmc-linux-gnueabi/strace` doesn't.

Hmm, that's odd -- but `bitbake strace` succeeded?  If that ran to completion I'm not sure where else the build directory could have ended up.

As an alternative, you could just add strace to your image via something like:

```diff
diff --git a/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb b/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb
index e9f676dc8035..9ce1918ecc61 100644
--- a/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb
+++ b/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb
@@ -30,4 +30,5 @@ RDEPENDS:${PN}-system = " \
 	phosphor-host-postd \
 	phosphor-ipmi-kcs \
 	phosphor-post-code-manager \
+	strace \
 	"
```

and then rebuild/reinstall obmc-phosphor-image, after which it should just be available in `/usr/bin`.

> If you didn't do any modification to your BMC would you mind sharing it so that I can try it ?

I wouldn't expect this to behave any differently than what you compiled yourself, but here's a flash image I built from the branch in my x86-power-control review (https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/46642/) if you want to try it (gzipped because github won't let me upload it raw):

[flash-e3c246d4i.gz](https://github.com/openbmc/openbmc/files/7220723/flash-e3c246d4i.gz)

Mr-Antoine, 2021-09-24:
> I wouldn't expect this to behave any differently than what you compiled yourself, but here's a flash image I built from the branch in my x86-power-control review [...]

To update the BMC I need the `obmc-phosphor-image-<platform>-<timestamp>.static.mtd.tar` file. I can then update it with `curl` or the OpenBMC Tool.

zevweiss, 2021-09-24:
Oh -- I had assumed you had a flash programmer or something available to just reprogram the whole chip.  Can I ask what mechanism you used to do the initial OpenBMC install?

In any case, here's the corresponding `.static.mtd.tar` (gzipped again): [obmc-phosphor-image-e3c246d4i.static.mtd.tar.gz](https://github.com/openbmc/openbmc/files/7224401/obmc-phosphor-image-e3c246d4i.static.mtd.tar.gz)

Mr-Antoine, 2021-09-24:
> I had assumed you had a flash programmer or something available to just reprogram the whole chip.

I used the Socflash for that, but since I can't boot I'm unable to use it.

zevweiss, 2021-09-24:
Oh right, forgot you had mentioned that previously.  Okay, well give that tarball a shot, but I'd guess pursuing getting a KCS trace (via strace) is more likely to be a fruitful avenue of exploration.  (At least, that's the best idea I've got at the moment anyway.)

amboar, 2021-09-24:
> > I had assumed you had a flash programmer or something available to just reprogram the whole chip.
> 
> I used the Socflash for that, but since I can't boot I'm unable to use it.

You can always netboot the BMC and then use flashcp. Failing that you can flash the spi-nor with socflash if you have access to the BMC serial console

Mr-Antoine, 2021-10-04:
First of all I'd like to apologize for not responding for 10 days but I was unable to work on it last week.

> Oh right, forgot you had mentioned that previously. Okay, well give that tarball a shot

As expected the tarball didn't work.

> As an alternative, you could just add strace to your image via something like:
> 
> ```
> diff --git a/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb b/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb
> index e9f676dc8035..9ce1918ecc61 100644
> --- a/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb
> +++ b/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb
> @@ -30,4 +30,5 @@ RDEPENDS:${PN}-system = " \
>  	phosphor-host-postd \
>  	phosphor-ipmi-kcs \
>  	phosphor-post-code-manager \
> +	strace \
>  	"
> ```
> and then rebuild/reinstall obmc-phosphor-image, after which it should just be available in /usr/bin.

Adding `strace` to my image doesn't work, I'm getting the following errors when I'm executing `bitbake obmc-phosphor-image` :

[image]

> You can always netboot the BMC and then use flashcp. Failing that you can flash the spi-nor with socflash if you have access to the BMC serial console

I don't have acces to the BMC serial console.

Mr-Antoine, 2021-10-04:
> Adding strace to my image doesn't work, I'm getting the following errors when I'm executing bitbake obmc-phosphor-image

This is happening with the latest commit of OpenBMC so I decided to use an older commit, the e8963ad91976924003cb17c0bf966619b3d87307.

Mr-Antoine, 2021-10-04:
First of all I'd like to apologize for not responding for 10 days but I was unable to work on it last week.

> Oh right, forgot you had mentioned that previously. Okay, well give that tarball a shot

As expected the tarball didn't work.

> As an alternative, you could just add strace to your image via something like:
>
> ```
> diff --git a/meta-asrock/meta-e3c246d4i/recipes-asrock/packagegroups/packagegroup-asrock-apps.bb [...]

Adding `strace` to my image did work but I'm unable to use it. I'm still getting the error `-sh: /usr/bin/strace: cannot execute binary file: Exec format error`.

> You can always netboot the BMC and then use flashcp. Failing that you can flash the spi-nor with socflash if you have access to the BMC serial console

Unfortunately I don't have acces to the BMC serial console.

zevweiss, 2021-10-04:
> `-sh: /usr/bin/strace: cannot execute binary file: Exec format error`

That sounds like it's probably left-over side effects of a previous attempt at installing strace -- to hazard a guess, did you perhaps manually install a new binary into `/usr/bin` prior to the update?  If so, note that due to the way OpenBMC's overlay mount setup is arranged, a reinstall won't replace that -- it will update the underlying read-only squashfs image, but the read-write jffs2 image that gets layered over that will remain (when I'm "installing" programs like that I generally just drop them in `/tmp` for that reason).  To manually undo that, you'll need to do something like:

```
$ rm /run/initramfs/rw/cow/usr/bin/strace
$ reboot
```

(If you're feeling a bit more adventurous you could probably also get away with a `sync && sysctl -w vm.drop_caches=3` before the `rm` instead of the reboot, but rebooting is probably a bit safer.)  If there are any other files you've manually modifed or replaced inside `/run/initramfs/rw/cow`, it might be good to reset those similarly.

Mr-Antoine, 2021-10-05:
```
$ rm /run/initramfs/rw/cow/usr/bin/strace
$ reboot
```

This worked thanks.

Since it works here's the KCS file generated.
[kcs.txt](https://github.com/openbmc/openbmc/files/7283529/kcs.txt)

zevweiss, 2021-10-05:
Great, thanks -- unfortunately, however, hasn't ended up providing any obvious clues I'm afraid, and I'm a bit stumped.  I'm wondering if it might be some incompatibility with the `-2T` variant of the board, e.g. with some option ROM it might be loading or something (we've only got `-NL` units, so that's all I've ever run it on).  As another somewhat blind shot in the dark, if you can get into the BIOS settings, maybe try experimenting with changing some of the following settings and seeing if it behaves any differently?

 - Set `Advanced -> Chipset Configuration -> Onboard LAN1 and LAN2` to disabled
 - Set `Boot -> Boot From Onboard LAN` to disabled
 - Change `Boot -> Boot option filter` to either `Legacy only` or `UEFI only` (not sure offhand which the X550 might be)

I've also reached out to a Coreboot developer I'm aware of having run OpenBMC successfully on this board to see which variant he was using, which might help eliminate some possibilities.

zevweiss, 2021-10-06:
I've now heard back from the Coreboot dev; he reports that he's also on a `-2T` board and had been able to boot the host successfully with the factory BIOS.

Mr-Antoine, 2021-10-06:
> I've now heard back from the Coreboot dev; he reports that he's also on a -2T board and had been able to boot the host successfully with the factory BIOS.

That's weird, do you know what hardware he's using ? Also would he mind sharing his BIOS settings ?

Mr-Antoine, 2021-10-06:
Also, how did you/he flash it ? I'm starting to have doubts about how I flashed it since it works for him but not for me.

zevweiss, 2021-10-06:
> That's weird, do you know what hardware he's using ? Also would he mind sharing his BIOS settings ?

Aside from that it's the -2T variant of the board, I don't know (i.e. the specific CPU SKU or DIMMs he had).  I doubt the precise BIOS settings he had are still available since it didn't stay on the AMI BIOS for long (that got replaced by Coreboot).

> Also, how did you/he flash it ? I'm starting to have doubts about how I flashed it since it works for him but not for me.

I don't know exactly what method he used, but given that your BMC itself boots healthily I doubt the programming mechanism is the source of the problem.  I've also used socflash to install OpenBMC successfully in the past.

Can you get in to your BIOS configuration to experiment with the settings I suggested above?

Mr-Antoine, 2021-10-06:
> Can you get in to your BIOS configuration to experiment with the settings I suggested above?

I couldn't get to the BIOS since the boot sequence would hang. "Thankfully" tho I managed to corrupt my BMC while trying to upgrade it to the old factory firmware since I wanted to be able to access the BIOS. Since I had no BMC working I was able to boot so I flashed it to the old factory firmware (the 1.00.00).

Now that I can access the BIOS, I'll do those settings all at once then upgrade the BMC to OpenBMC to see if the result is different.

Mr-Antoine, 2021-10-06:
> that hasn't ended up providing any obvious clues I'm afraid

Also, how did you retreive informations from the KCS file ? I know it's hex but other than that I can't really seem to get anything out of it.

zevweiss, 2021-10-06:
> Also, how did you retreive informations from the KCS file ? I know it's hex but other than that I can't really seem to get anything out of it.

I just did a rough manual decode of each message exchanged -- the protocol is just IPMI, so armed with the spec (https://www.intel.com/content/dam/www/public/us/en/documents/specification-updates/ipmi-intelligent-platform-mgt-interface-spec-2nd-gen-v2-0-spec-update.pdf, specifically the tables in Section 5 and Appendix G) you can decode what each command/response is (with the exception of the manufacturer-specific OEM commands, most of which I don't have any info on).  For example, the `\x18\x04` message in the first line is a "get self-test results" command (NetFn 0x06, a.k.a. "App", command number 0x04).

Mr-Antoine, 2021-10-07:
> maybe try experimenting with changing some of the following settings and seeing if it behaves any differently?

It didn't change a thing.

Also my BMC is now dead, I've no IP and I can't boot completly I get stuck this time at 90 (before 9A). I guess my only option is to use the BMC SMBus even tho I've never used it. I did some searching but I couldn't find what cable I need in order to use it.

Mr-Antoine, 2021-10-21:
> If you've still got a functional u-boot, a serial console (e.g. a USB adapter like this or similar) via the BMC_DEBUG header would allow you to access the u-boot prompt

I may be able to get my hand on this USB adapter but I don't know the pinout of the BMC Debug header, from my research it is what's in the red circle since the SMBus header won't be of any help :

[image]

> If you don't have a working u-boot, a flash programmer for the BMC flash header (the little 9-pin one at the BMC end of the PCIe slot).

I don't have the material to do that unfortunaly.

zevweiss, 2021-10-21:
Yep, the header circled in red has the relevant pins, here's the pinout: 
[image]

Given the apparent mismatch between the pinout of your adapter, it looks like you'll probably want to use some "extension" jumper wires (one male end, one female end, e.g. something like [this](https://www.adafruit.com/product/826)) to connect just the GND, RXD, and TXD pins of your adapter to the corresponding pins of the header on the board (you probably don't want to attach VCC, as that might fight with the 3VSB from the board and lead to bad things).  Once you've got that attached and the USB end plugged in on another system, you'll probably have a corresponding character device like `/dev/ttyUSB0`; you can then use screen to connect to it:

    $ screen /dev/ttyUSB0 115200

(the second argument being the baud rate.)  If you then reset the board (cut & restore power), if your u-boot is still alive at all you should hopefully see some output from it in `screen` on the other host (if at first it doesn't work, try swapping your RXD & TXD lines in case they're backwards).  If you hit a key while u-boot is printing its timeout countdown you should be dropped into its command prompt.  From there it should be possible to fetch a FIT image over the network via tftp, boot into that, and then use `flashcp` to restore a full firmware image to the flash.  (I can provide further info on the details of that as needed, but let's confirm if you've still got a working u-boot first.)

Mr-Antoine, 2021-11-02:
> let's confirm if you've still got a working u-boot first

I've got a working u-boot

> From there it should be possible to fetch a FIT image over the network via tftp

I'd love to but I can't even ping 8.8.8.8 even tho I'm connected to the network

> I can provide further info on the details of that as needed

It would really help me out, thanks

zevweiss, 2021-11-03:
> I've got a working u-boot

Great!

> I'd love to but I can't even ping 8.8.8.8 even tho I'm connected to the network

Perhaps you've already done this and there's some other problem, but note that you'll need to tell u-boot what MAC & IP addresses to use before it's able to do anything useful on the network -- did you do a `setenv ethaddr xx:xx:xx:xx:xx:xx` and `setenv ipaddr x.x.x.x` before attempting your ping?  (Or have those variables already set in your u-boot environment -- see the `printenv` command.)

_Hopefully_ with the u-boot environment set approprately you can get u-boot to be able to ping another host on your local network, so I'll assume for now that we'll be able to proceed with that.  To boot via TFTP you'll need another host on the network running a TFTP server -- I typically use the `tftpd-hpa` package on an Ubuntu or Debian system, personally (very easy to set up, just install the package, and make sure the service is running and that there isn't a firewall or something blocking access to UDP port 69).

You'll need a FIT image (containing the kernel, device tree, and initramfs) to boot from -- that can be found in your OpenBMC build directory at `tmp/deploy/images/e3c246d4i/fitImage-obmc-phosphor-initramfs-e3c246d4i-e3c246d4i`.  Copy that file to your TFTP server directory (`/var/lib/tftpboot` for `tftpd-hpa` on Ubuntu, for example) under a friendlier file name -- maybe just `e3c246d4i.fit`, say.

Then, back at your BMC's u-boot command prompt, run `setenv serverip x.x.x.x` to tell it the IP address of your TFTP server, and then `tftp e3c246d4i.fit` (or whatever you named the file above) to load the FIT image over the network from the server (note that the `tftp` command's output will print the memory address at which it loads the image).  Then, modify your `bootargs` environment variable to add the arguments `debug-init-sh` and `enable-initrd-debug-sh` to the kernel command-line, so depending on your existing `bootargs` perhaps something like:

```
ast# setenv bootargs console=ttyS4,115200n8 root=/dev/ram rw mem.devmem=1 debug-init-sh enable-initrd-debug-sh
```

(Those arguments will cause the init script in the initramfs to immediately drop into an interactive shell instead of proceeding through the usual boot sequence -- if your BMC wasn't booting properly I'm guessing that wouldn't get very far.)

With that done you can now boot into the network-loaded FIT image by running `bootm $ADDRESS`, where `$ADDRESS` is the hex number after `Load address:` printed by the `tftp` command.

You should then find yourself at an interactive shell prompt in the filesystem of the initramfs (no flash or anything else mounted).

Back on your tftp server, copy another file into the server directory from your OpenBMC build directory: `tmp/deploy/images/e3c246d4i/flash-e3c246d4i` (a symlink to the `*.static.mtd` file, i.e. the full raw OpenBMC flash image).  You can then TFTP _that_ to load it into a tmpfs on the BMC, e.g. via something like:

```
/ # ip link set eth0 up
/ # ip addr add x.x.x.x/q dev eth0 # /q being your network prefix length
/ # tftp -g -l /run/flash-e3c246d4i -r flash-e3c246d4i $SERVERIP # replace with the IP address of your tftp server
```

Note that: 
 - the busybox `tftp` command used here gives a lot less progress feedback than u-boot's, so don't be alarmed if it appears to hang silently for a while (transferring a 32MB image might take a minute or two) 
 - due to the limitations of the initramfs shell (no job control), you won't be able to use control-C to interrupt it (or anything else that might hang)

Once that's complete and you've got the image file on the BMC (it might be good to do a quick `sha1sum` check against the original on the build system to make sure it all got there correctly), you can then burn the image to the BMC flash by running:

```
/ # flashcp -v /run/flash-e3c246d4i /dev/mtd0
```

(That one fortunately gives a lot more feedback than the tftp command.)

Once the `flashcp` completes, a `reboot -f` should boot you into the newly-flashed BMC firmware image.

Mr-Antoine, 2021-11-03:
It worked, thank you so much!

I'm still stuck at 9A but you saved my motherboard, thanks a lot for that.

Just to be sure (because I don't want to cause more issues), once the factory BMC is transfered to `/tmp` I'd need to do something like this :

`flashcp -v /tmp/E3C246D4I-2T_P1.00.00.ima /dev/mtd0`

zevweiss, 2021-11-03:
Yes, that should allow you to restore the factory firmware.

And incidentally, I finally remembered (or was reminded by something else unrelated) the context in which I had previously encountered hangs at POST code 0x9a...a few months ago I experimented with using a newer version of u-boot (the e3c246d4i OpenBMC build currently uses 2016.07, which is kind of old, and 2019.04 is also available in OpenBMC), and while most things seemed to work OK, I got the same host boot hang you're seeing with the current version (with the older u-boot).  My guess at the time was that it was due to some BMC SoC state (pinctrl or gpio, perhaps) not getting reinitialized properly at boot time, but I didn't know of any great ways to debug it further off the top of my head, and that line of experimentation has been sitting on the metaphorical shelf since then, unfortunately.

If I do make any progress on that I'll try to remember to post an update here.

Glad your board's working again!

mdmillerii, 2021-11-03:
@zevweiss the init script has a commented out getty that can be used to gain job control.   Note you have to use the real tty name no t`/dev/console` that is the limiting factor

Mr-Antoine, 2021-11-03:
> the e3c246d4i OpenBMC build currently uses 2016.07, which is kind of old, and 2019.04 is also available in OpenBMC

I'll try to upgrade it to the 2019.04 version.

Also, thanks again for your help!

zevweiss, 2021-11-04:
> I'll try to upgrade it to the 2019.04 version.

Okay -- note that that's a slightly non-trivial undertaking if you do want to try it out.  This patch should enable it:

```diff
diff --git a/meta-asrock/meta-e3c246d4i/conf/machine/e3c246d4i.conf b/meta-asrock/meta-e3c246d4i/conf/machine/e3c246d4i.conf
index 38fb811fff36..11aba91cde3e 100644
--- a/meta-asrock/meta-e3c246d4i/conf/machine/e3c246d4i.conf
+++ b/meta-asrock/meta-e3c246d4i/conf/machine/e3c246d4i.conf
@@ -1,7 +1,8 @@
 KMACHINE = "aspeed"
 KERNEL_DEVICETREE = "${KMACHINE}-bmc-asrock-${MACHINE}.dtb"
 
-UBOOT_MACHINE = "ast_g5_phy_config"
+UBOOT_MACHINE = "evb-ast2500_defconfig"
+UBOOT_DEVICETREE = "ast2500-evb"
 
 FLASH_SIZE = "32768"
 
@@ -11,5 +12,9 @@ require conf/machine/include/obmc-bsp-common.inc
 PREFERRED_PROVIDER_virtual/obmc-flash-mgmt = "packagegroup-asrock-apps"
 PREFERRED_PROVIDER_virtual/obmc-system-mgmt = "packagegroup-asrock-apps"
 
+PREFERRED_PROVIDER_virtual/bootloader = "u-boot-aspeed-sdk"
+PREFERRED_PROVIDER_u-boot = "u-boot-aspeed-sdk"
+PREFERRED_PROVIDER_u-boot-fw-utils = "u-boot-fw-utils-aspeed-sdk"
+
 VIRTUAL-RUNTIME_obmc-host-state-manager = "x86-power-control"
 VIRTUAL-RUNTIME_obmc-chassis-state-manager = "x86-power-control"
diff --git a/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk/0001-HACK-disable-mac1-to-avoid-error-spew.patch b/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk/0001-HACK-disable-mac1-to-avoid-error-spew.patch
new file mode 100644
index 000000000000..c36e46b8ce04
--- /dev/null
+++ b/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk/0001-HACK-disable-mac1-to-avoid-error-spew.patch
@@ -0,0 +1,25 @@
+From bf66b2c8670aea8d12d03f2d542a4bcbb3213e16 Mon Sep 17 00:00:00 2001
+From: Zev Weiss <zev@bewilderbeest.net>
+Date: Thu, 4 Nov 2021 01:44:01 -0700
+Subject: [PATCH] HACK: disable mac1 to avoid error spew
+
+---
+ arch/arm/dts/ast2500-evb.dts | 2 +-
+ 1 file changed, 1 insertion(+), 1 deletion(-)
+
+diff --git a/arch/arm/dts/ast2500-evb.dts b/arch/arm/dts/ast2500-evb.dts
+index f668eec5c137..9009e3eb1f50 100644
+--- a/arch/arm/dts/ast2500-evb.dts
++++ b/arch/arm/dts/ast2500-evb.dts
+@@ -56,7 +56,7 @@
+ };
+ 
+ &mac1 {
+-	status = "okay";
++	status = "disabled";
+ 	phy-mode = "rgmii";
+ 	pinctrl-names = "default";
+ 	pinctrl-0 = <&pinctrl_mac2link_default &pinctrl_mdio2_default>;
+-- 
+2.33.0
+
diff --git a/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk/e3c246d4i.cfg b/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk/e3c246d4i.cfg
new file mode 100644
index 000000000000..e66af86ba022
--- /dev/null
+++ b/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk/e3c246d4i.cfg
@@ -0,0 +1,6 @@
+CONFIG_BOOTCOMMAND="bootm 20080000"
+CONFIG_ENV_OFFSET=0x60000
+
+# size reduction to fit in the 32M flash layout's 384K u-boot partition
+CONFIG_EFI_LOADER=n
+CONFIG_SYS_LONGHELP=n
diff --git a/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk_%.bbappend b/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk_%.bbappend
new file mode 100644
index 000000000000..a3e93c0b69da
--- /dev/null
+++ b/meta-asrock/meta-e3c246d4i/recipes-bsp/u-boot-aspeed-sdk/u-boot-aspeed-sdk_%.bbappend
@@ -0,0 +1,6 @@
+FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"
+
+SRC_URI:append = " \
+	file://e3c246d4i.cfg \
+	file://0001-HACK-disable-mac1-to-avoid-error-spew.patch \
+	"
```

Note that that includes a small hack to disable the second network interface, because I got u-boot spewing a bunch of garbage bytes to the console while trying to initialize it due to some bug I didn't quite fully track down (that behavior might be specific to my `-NL` variant of the board though, so you might be fine without it on your `-2T`).

Running that, I also see the host boot sequence hang at POST code 0x9a.  However, the very first time I booted it it got past that and started up OK, but I haven't been able to reproduce that (all my subsequent attempts have hung at 0x9a).  It doesn't seem like it should be related, but the one successful boot was before I added the patch to disable mac1 (trying again with it removed hasn't gotten any further though, I'm not sure what the critical difference is).

Another difference I noticed was that with the new u-boot I get the iKVM's display active and running significantly earlier in the host boot sequence than I do with the old u-boot.  With the old one, the display doesn't come on for a little while after powering on the host (tens of seconds, though the boot process proceeds OK), whereas with the new u-boot the display comes on pretty much immediately, but then it gets stuck at the 0x9a hang.  What have you seen w.r.t. to the timing of the iKVM display coming on when powering on the host?

zevweiss, 2021-11-05:
Ah, a discovery!  In sifting through the differences between the old & new versions of u-boot, I found a strap bit that (on my system) makes the difference between the host boot hanging at 0x9a and proceeding past it -- and looking back at your strap register value, it looks like you've got the bit in question missing too, so hopefully this does the trick.

Before messing around with switching u-boot versions, try running `devmem 0x1e6e2070 w 0x00100000` on the BMC and then power on the host -- with luck, your board will behave like mine does and your host will boot up (though you may find the iKVM display stays blank for a little while at first now like mine does, because those two behaviors seem to be tied to each other for some reason).

Mr-Antoine, 2021-11-05:
It works!

Here's what I did :

1. `fw_setenv bootargs "console=ttyS4,115200n8 root=/dev/ram rw mem.devmem=1"`
2. Restart
3. `devmem 0x1e6e2070 w 0x00100000`

How are we supposed to implement it permenantly ?

Thank you so much for the time you took to explain and help me!

zevweiss, 2021-11-05:
I've just posted a patch that should fix this: https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/48666

If you can test that out and report back (ideally on gerrit, but a comment here would also be useful) that'd be great.

Mr-Antoine, 2021-11-15:
> I've just posted a patch that should fix this: https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/48666

I can confirm it works!

Thank you so much for the time and effort you've put into this!

closed 2021-11-15 by Mr-Antoine

zevweiss, 2021-11-15:
Excellent, glad to hear that -- thanks for the testing and bug report!

commit 1d5acf18 (2021-11-15): e3c246d4i: enable phosphor-isolation

Without the patches enabled by this feature, the default e3c246d4i
strapping leaves the AST2500's SuperIO enabled, which causes the host
boot process to hang (POST code 0x9a).

Fixes: #3820

Change-Id: I45de4945ff7f3322ed87494ad2c7aa41ce345438