openbmc/openbmc #3821: Is openbmc able to dectect the host's cpus are present and running?

wangmin239, 2021-09-29:
Hi folks,
Is there a recipe to detect the host's cpus are present and running? Does openBmc support to monitor the host's cpus are running?
Thanks.

geissonator, 2021-09-29:
Hi @wangmin239 please refer to https://github.com/openbmc/openbmc#questions for asking questions.

closed 2021-09-29 by geissonator

mdmillerii, 2021-09-29:
Short answer is there's an interface to store it for the rest of the code, but detection depends on the host including it's processor architecture.

mdmillerii, 2021-09-29:
https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/yaml/xyz/openbmc_project/State/Host.interface.yaml

wangmin239, 2021-09-30:
@mdmillerii  Thanks your responder. I checked out the source codes from phosphor-dbus-interface and phoshor-state-manger. I wish to get a solution from the codes. Thanks again.