openbmc/openbmc #3823: AST2600 eMMC image landing to ramfs console instead of rootfs

shitalkumar-gandhi, 2021-09-30:
Hi All,

I am booting eMMC 8GB image using QEMU for AST2600 target machine, but during booting, it is landing to ramfs console, I wanted to boot rootfs.

If I try to boot rootfs manually from the ramfs console, then the emergency shell gets launched, refer the following logs.

I used sgdisk as well to repair the GPT, but no success.

Thanks.

[    3.951698] Segment Routing with IPv6
[    3.996463] mmc2: new high speed MMC card at address 0001
[    4.001723] mmcblk2: mmc2:0001 QEMU! 8.00 GiB
[    4.003108] mmcblk2boot0: mmc2:0001 QEMU! partition 1 1.00 MiB
[    4.004269] mmcblk2boot1: mmc2:0001 QEMU! partition 2 1.00 MiB
[    4.006032] mmcblk2rpmb: mmc2:0001 QEMU! partition 3 128 KiB, chardev (248:0)
[    4.008397] NET: Registered protocol family 17
[    4.009052] 8021q: 802.1Q VLAN Support v1.8
[    4.010210] Registering SWP/SWPB emulation handler
[    4.102404] printk: console [netcon0] enabled
[    4.102895] netconsole: network logging started
[    4.103249] GPT:Primary header thinks Alt. header is not at the end of the disk.
[    4.103789] GPT:15927369 != 16773119
[    4.104054] GPT:Alternate GPT header not at the end of the disk.
[    4.104402] GPT:15927369 != 16773119
[    4.104611] GPT: Use GNU Parted to correct GPT errors.
[    4.106341]  mmcblk2: p1 p2 p3 p4 p5 p6 p7 p8
[    4.177236] Freeing unused kernel memory: 1024K
[    4.318188] Checked W+X mappings: passed, no W+X pages found
[    4.319518] Run /init as init process
tail: can't open '/dev/mmcblk0': No such file or directory
tail: no files
Problem opening /dev/mmcblk0 for reading! Error is 2.
The specified file does not exist!
Warning: Not all of the space available to /dev/mmcblk2 appears to be used, you can fix the GPT to use all of the space (an extra 845750 blocks) or continue with the current setting?
Starting version 249.4+
[   19.485220] EXT4-fs (mmcblk2p4): mounted filesystem with ordered data mode. Opts: (null)
Cannot open /dev/mmcblk0p1: No such file or directory
sh: true: unknown operand
fsck.ext4: No such file or directory while trying to open /dev/disk/by-partlabel/rwfs
Possibly non-existent device?
mount: mounting /dev/disk/by-partlabel/rwfs on /mnt/rofs/var failed: No such file or directory
/bin/sh: can't access tty; job control turned off
/ #

mdmillerii, 2021-09-30:
Hmm the emmc is appearing as mmcblk2 but on hardware it's mmcblk0 and we have the device hardcoded.

@legoater suggestions?  Can we supress SD cards?

shitalkumar-gandhi, 2021-09-30:
@mdmillerii Yes, it's hardcoded as mmcblk0, and facing the same error, in QEMU I can see partitions as mmcblk2 hence changed to that in the ramfs init script, but no success.

mdmillerii, 2021-09-30:
Can you show the output when you change the assignment from `mmcblk0` to `mmcblk2`

shitalkumar-gandhi, 2021-09-30:
Thanks for the quick response.
Here is the output with mmcblk2 changes in ramfs init script.

[image]

mdmillerii, 2021-09-30:
Thanks the mmcblk0p1 needs to be something like `${mmc}p1` (GitHub is not showing the full src right now)

shitalkumar-gandhi, 2021-09-30:
Thanks, It is coming from line fw_printenv -n rwreset in initramfs, as it seems fw_printenv is not supported in initramfs.
/ # fw_printenv -n rwreset
Cannot open /dev/mmcblk0p1: No such file or directory
/ #

By manually mounting rootfs with switch_root command, it is landed in the emergency shell.

  OK  ] Reached target Emergency Mode.
[  426.003986] systemd[1]: Started Journal Service.
[  OK  ] Started Journal Service.
         Starting Flush Journal to Persistent Storage...
[  427.125264] systemd-journald[215]: Received client request to flush runtime journal.
[  427.310172] overlayfs: upper fs does not support xattr, falling back to index=off and metacopy=off.
[  427.327248] overlayfs: upper fs does not support xattr, falling back to index=off and metacopy=off.
[  OK  ] Finished Flush Journal to Persistent Storage.
[  427.424757] overlayfs: upper fs does not support xattr, falling back to index=off and metacopy=off.
[  427.475225] overlayfs: upper fs does not support xattr, falling back to index=off and metacopy=off.
[  OK  ] Finished Bind mount volatile /var/cache.
[  OK  ] Finished Bind mount volatile /var/lib.
[  OK  ] Finished Bind mount volatile /var/spool.
[  OK  ] Finished Bind mount volatile /srv.
         Starting Load/Save Random Seed...
         Starting Create Volatile Files and Directories...
[  OK  ] Finished Load/Save Random Seed.
[  OK  ] Started Rule-based Manager for Device Events and Files.
         Starting Network Configuration...
[  OK  ] Finished Create Volatile Files and Directories.
[  OK  ] Started Hardware RNG Entropy Gatherer Daemon.
         Starting Network Time Synchronization...
You are in emergency mode. After logging in, type "journalctl -xb" to view
system logs, "systemctl reboot" to reboot, "systemctl default" or "exit"
to boot into default mode.
Give root password for maintenance
(or press Control-D to continue): [  438.037046] ftgmac100 1e690000.ftgmac eth3: Link is Up - 100Mbps/Full - flow control rx/tx
[  438.038442] IPv6: ADDRCONF(NETDEV_CHANGE): eth3: link becomes ready
[  438.113349] ftgmac100 1e670000.ftgmac eth2: Link is Up - 100Mbps/Full - flow control rx/tx
[  438.175636] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[  438.178769] ftgmac100 1e680000.ftgmac eth1: Link is Up - 100Mbps/Full - flow control rx/tx
[  438.189763] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  438.215917] ftgmac100 1e660000.ftgmac eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[  438.216652] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  OK  ] Stopped Emergency Shell.
[FAILED] Failed to mount /home.
See 'systemctl status home.mount' for details.
[DEPEND] Dependency failed for Local File Systems.
[  OK  ] Finished Load Kernel Module fuse.
[  OK  ] Stopped Dispatch Password …ts to Console Directory Watch.
[  OK  ] Started Emergency Shell.
[  OK  ] Reached target Emergency Mode.
You are in emergency mode. After logging in, type "journalctl -xb" to view
system logs, "systemctl reboot" to reboot, "systemctl default" or "exit"
to boot into default mode.
Give root password for maintenance
(or press Control-D to continue):

**What I want is roots console like we are getting following in case of the static mtd image file.**

        Starting Phosphor Version Software Manager...
         Starting Phosphor Time Manager daemon...
[  OK  ] Finished Wait for /xyz/openbmc_project/state/chassis0.
[  OK  ] Finished Wait for /xyz/openbmc_project/led/groups.
[  OK  ] Started Phosphor Chassis State Manager.
[  OK  ] Started Avahi mDNS/DNS-SD Stack.
[  OK  ] Started System Logging Service.
[  OK  ] Started Phosphor BMC State Manager.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) nodistro.0 evb-ast2600 ttyS4

evb-ast2600 login: root
Password:

LeeTroy, 2021-09-30:
@shitalkumar-gandhi 
Please check comments in #3818 , and try to use [markdown code block](https://docs.github.com/en/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code) for code block (or log).

mdmillerii, 2021-09-30:
Yes I figured that out  , but also udev is not finding the partition labels and creating the by-partlabel symlink.
it looks like the gpt was fixed when we detected it was wrong (hence the second GPT: p1 p2 ... line)

can you confirm the rwfs partition is labeled and formatted ext4?  It seems that was not found.
the getroot seemed to find the root fs part label as we were able to mount rootfs

without rwfs on $rodir/var the and the etc overlay, the etc directory is readonly and that breaks systemd boot.

shitalkumar-gandhi, 2021-09-30:
Thanks, @mdmillerii  for your prompt response, will check that.
@LeeTroy I am following the same steps mentioned in the comments, @namo999 could you please comment, are you able to boot into rootfs?

namo999, 2021-09-30:
@shitalkumar-gandhi, I have not checked about roots, I need ramfs console only.

LeeTroy, 2021-10-01:
[boot.log](https://github.com/openbmc/openbmc/files/7263715/boot.log)
@shitalkumar-gandhi Yes, I could boot into rootfs. 

Could you post your u-boot-env? Mine like the following:
```
ast# pri
boota=setenv bootpart 2; setenv rootfs rofs-a; run bootmmc
bootargs=console=ttyS4,115200n8
bootb=setenv bootpart 3; setenv rootfs rofs-b; run bootmmc
bootcmd=setenv origbootargs ${bootargs}; run bootconfcmd; run bootsidecmd
bootconfcmd=if test ${fitconfig} -n; then setenv bootmconf "#${fitconfig}"; fi
bootdelay=2
bootmmc=run setmmcargs; ext4load mmc 0:${bootpart} ${loadaddr} fitImage && bootm ${loadaddr}${bootmconf}; echo Error loading kernel FIT image with fitconfig ${fitconfig}; bootm ${loadaddr}; echo Error loading kernel FIT image
bootside=a
bootsidecmd=if test ${bootside} = b; then run bootb; run boota; else run boota; run bootb; fi
fdtcontroladdr=bcf8f960
loadaddr=0x83000000
setmmcargs=setenv bootargs ${origbootargs} rootwait root=PARTLABEL=${rootfs}
stderr=serial@1e784000
stdin=serial@1e784000
stdout=serial@1e784000

Environment size: 836/65531 bytes
```

shitalkumar-gandhi, 2021-10-01:
Many thanks, yes it is booting with the default 7 partitions total 16GB size image.
When I changed it total 8GB with 8 partitions, at that time I was facing this issue.

mdmillerii, 2021-10-01:
I'd look at how big the WIC image with the gpt partitions.   It's possible you can shrink the ext4 but the size was chosen to allow future growth.

shitalkumar-gandhi, 2021-10-04:
ok, Thanks a lot @mdmillerii @LeeTroy 
Just one thing, I am preferring squashfs over ext4 for AST2600 rootfs RO partition.

I have done the required changes for squashfs in OpenBMC, no error in build but squashfs.xz is not getting generated, hence something is missing.

It would be great if you can point out, thanks.

[image]

mdmillerii, 2021-10-04:
Look for the image type variable.   It lists the image formats to package.

LeeTroy, 2021-10-05:
I think the problem is in mmc-init.sh. The _/dev/mmcblk2_ is the SD card, not the second partition of emmc. The emmc partition looks like _/dev/mmcblk0**p1**_.

I'm not quite familiar with squashfs with emmc, but why do you format  __rwfs__ with squashfs near line 52? Isn't it a readonly filesystem?

shitalkumar-gandhi, 2021-10-05:
Yeah that 52 line change I reverted, in QEMU I can see /dev/mmcblk2 partition instead of /dev/mmcblk0p1.
What do you suggest ext4 or squashfs filesystem for AST2600 machine?
Thanks

mdmillerii, 2021-10-05:
When discussing the upstream design we wanted to keep the read only base and limit overlay (just etc).  We chose ext4 as a well tested choice for sector addressable media.  We didn't expect to need to compress the rofs because our MMC is not constrained.  While you may save some IO bandwidth with compressed storage we didn't expect the CPU usage to decompress to be worth the tradeoff.

williamspatrick, 2021-10-05:
> While you may save some IO bandwidth with compressed storage we didn't expect the CPU usage to decompress to be worth the tradeoff.

I suspect you’d be surprised on the performance of btrfs+zstd compression for a (relatively) slow IO device like eMMC vs uncompressed ext4.

shitalkumar-gandhi, 2021-10-06:
ok Thanks, @mdmillerii @williamspatrick 

I will check the performance of squashfs vs ext4 on eMMC, what would be the exact place where I can add prepare-emmc-qemu script in OpenBMC repo for auto-building eMMC image during the build.

shitalkumar-gandhi, 2021-10-06:
ok got it thanks @amboar.

shitalkumar-gandhi, 2021-10-07:
Hi All,

By including "require conf/distro/include/phosphor-ubi.inc" file in local.conf, it is creating squashfs.xz, ubifs, .ubi.mtd files.

Now how to test squashfs partitions (uboot and rootfs) same as testing ext4 eMMC image packed with the wic image on QEMU.

I tried to create wic image for squashfs partition instead of the default ext4, following is the change, but no success.

[image]

Can anyone help me to create wic image for squashfs, which can be tested on QEMU?

Thanks.

shitalkumar-gandhi, 2021-10-12:
@mdmillerii thanks for the response.

The problem is: I am able to create squashfs.xz rootfs image, but for QEMU booting I wanted to build a wic image that has fit image along with roots, so that I can test that single image in QEMU.

ext4 eMMC wic file meta-aspeed/wic/emmc-aspeed.wks.in is like :
[image]

ext4 eMMC wic image is like :
[image]

The same thing I wanted to build for squashfs partition, hence wic file is modified as follows.
[image]