openbmc/openbmc #3824: devtool modify busybox

savatagecriss, 2021-10-04:
Hi, nice to meet you, 
I want to modify busybox source code, so I try following steps:
1. git clone openbmc.
2. devtool modify busybox
3. It show some error message in post-patch task as following:

`ERROR: Error executing a python function in exec_python_func() autogenerated:`

`The stack trace of python calls that resulted in this exception/failure was:`
``File: 'exec_python_func() autogenerated', lineno: 2, function: <module>``
`     0001:
 *** 0002:devtool_post_patch(d)`
`     0003:
File: '/opt/johnwei/openbmc/openbmc_test_witherspoon/meta/classes/devtool-source.bbclass', lineno: 226, function: devtool_post_patch`
`     0222:                bb.build.exec_func('do_patch', localdata)
     0223:                rm_patches()
     0224:                # Now we need to reconcile the new branch with the no-overrides one
     0225:                # (otherwise we'd likely be left with identical commits that have different hashes)
 *** 0226:                bb.process.run('git rebase devtool-no-overrides', cwd=srcsubdir)
     0227:        bb.process.run('git checkout %s' % devbranch, cwd=srcsubdir)
     0228:    bb.process.run('git tag -f devtool-patched', cwd=srcsubdir)
     0229:}
     0230:
File: '/opt/johnwei/openbmc/openbmc_test_witherspoon/poky/bitbake/lib/bb/process.py', lineno: 184, function: run
     0180:        if not stderr is None:
     0181:            stderr = stderr.decode("utf-8")
     0182:
     0183:    if pipe.returncode != 0:
 *** 0184:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0185:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of 'git rebase devtool-no-overrides' failed with exit code 128:
fatal: It looks like 'git am' is in progress. Cannot rebase. `

`ERROR: Logfile of failure stored in: /opt/johnwei/openbmc/openbmc_test_witherspoon/build/witherspoon-tacoma/tmp/work/armv7ahf-vfpv4d16-openbmc-linux-gnueabi/busybox/1.34.0-r0/devtooltmp-0c6xgmn6/temp/log.do_patch.28618` `

It looks like some issue are from recipe, I am not familiar with bitbake, so could anyone help me to solve this issue? thank you very much.

mdmillerii, 2021-10-04:
It looks like you did not leave the repository in a clean state before starting.   I am not experienced with devtool but git is complaining that a rebase or am apply is in is in progress when you tried to finish the update.   

Check git status for hints in what's wrong and how to resolve.

edtanous, 2022-05-23:
No response from submitter.  Assuming that mdmilleriis response solved their issue.  Please reopen if this is still a problem.

closed 2022-05-23 by edtanous

KamilPiekutowski, 2023-02-14:
Try ```devtool modify -O busybox```, or ```devtool modify --no-overrides busybox```.