openbmc/openbmc #3828: LUKS encrypted partition command hangs in QEMU

namo999, 2021-11-01:
**Note on Bugs**

I am trying to create and mount LUKS encrypted partition for AST2600 Machine in QEMU emulator.
But the command for the same is neither giving any output nor doing any action, it is just hanging without any console output, then the only option left is I have to restart the QEMU.
However, the same commands for the LUKS partition are working in the local x86 system.
 
**Unexpected behavior you saw**
LuksOpen command is waiting and not giving any output.

**Expected behavior**
It should open the LUKS partition.

**To Reproduce**
Steps to reproduce the behavior:
1. Create 5Mb test image : dd if=/dev/urandom of=test count=5 bs=1M
2. Create a secret key for LUKS : echo "mypassphrase" > /etc/key.txt 
3. Format LUKS partition : cryptsetup -y -v luksFormat test /etc/key.txt
4. Open LUKS partition : cryptsetup luksOpen test volume1 --key-file /etc/key.txt

**Screenshots**
[image]

mdmillerii, 2021-11-05:
So I see does a semaphore operation,DM_DEV_CREATE , DM_TABLE_LOAD, and DM_DEV_SUSPEND

namo999, 2021-11-09:
@mdmillerii Thanks.

I'll investigate. It appears that qemu is unable to generate a filesystem encryption /dev/mapper/volume node.

Moreover, I have another query, how to update the entire firmware from fit image to filesystem?

And, more importantly, how is this dual boot image state managed?

How does openbmc logic work if the device is booted from the A partition during firmware update?

For partition switching and activation, I am not able to see any bootside variable state management.

mdmillerii, 2021-11-09:
We don't just update a fit.  The details vary a bit by storage layout; we have MTD partitions (for directly attached NOR that only have one image); UBI (a layer on top of MTD, where we use UBI for wear leveling over most of the device and have support for two images), and eMMC (where the aspeed stored u-boot in a boot partition and the main flash has 2 images).   

https://github.com/openbmc/docs/tree/master/architecture/code-update

We do the side switch in unit files triggered from our code update manager

https://github.com/openbmc/phosphor-bmc-code-mgmt

namo999, 2021-11-11:
Ok, thanks a lot @mdmillerii

closed 2022-02-08 by namo999