openbmc/openbmc #3830: Support for openssl 3.0 within openbmc

geissonator, 2021-11-04:
Recently a rebase with upstream yocto brought in openssl 3.0. https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/48473 is the gerrit commit.

As discussed in the "yocto" channel in discord, this has resulted in 4 recipes failing due to certain things being deprecated in openssl 3.0.

Errors look like this:
```
13:58:58 | ../git/include/ssl_key_handler.hpp:112:41: error: 'rsa_st* EVP_PKEY_get1_RSA(EVP_PKEY*)' is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
13:58:58 |   112 |             RSA* rsa = EVP_PKEY_get1_RSA(pkey);
```

Impacted recipes are:
- bmcweb_git.bb (@edtanous @jmbills @gtmills)
- openpower-software-manager_git.bb (@anoo1)
- phosphor-software-manager_git.bb (@anoo1, @gtmills @leiyu-bytedance )
- phosphor-settings-manager_git.bb (@dkodihal)

Some options discussed in discord are to allow the deprecation, or do the needed fixes. It's not clear how much work it is to move to the new API's.

Opening this bug to get visibility to everyone.

LeeTroy, 2021-11-26:
Is there a planned schedule to have OpenSSL 3.0.0 support?

jk-ozlabs, 2021-11-26:
I'll take a look at doing the fixes here.

jk-ozlabs, 2021-11-26:
So the bmcweb bits need a bit of a rework: the key handler needs to be passed a SSL context pointer to use the new API; there's a mix of boost ssl wrapper context there, which we may be able to use instead. It might be nice to integrate the cert checks with the actual creation of the context, to avoid any TOCTOU issues.

jk-ozlabs, 2021-11-26:
(should we track as individual projects' issues?)

geissonator, 2021-11-29:
> (should we track as individual projects' issues?)

Thanks for taking a look @jk-ozlabs . If it's quick for some repos then just a gerrit link in this issue is good but if we need some more visibility for the maintainers to drive, separate issues in the respective repos makes sense to me.

williamspatrick, 2021-12-07:
@geissonator I've pushed up a fix for at least one of the phosphor-bmc-code-mgmt issues:

https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-bmc-code-mgmt/+/49439

@anoo: can you review this and apply the equivalent to openpower-... ?

@geissonator: I don't see how settingsd is affected by OpenSSL3.  There is nothing in that repository that uses OpenSSL.

geissonator, 2021-12-07:
Thanks @williamspatrick. I did a rebase, let me see if I can find what happened with settings.

geissonator, 2021-12-08:
Weird phosphor-settings-manager failure:
```
17:42:51 | checking for /data0/jenkins/workspace/ci-openbmc/distro/ubuntu/label/docker-builder/target/romulus/build/work/arm1176jzs-openbmc-linux-gnueabi/phosphor-settings-manager/1.0+gitAUTOINC+5ff6375ba9-r1/recipe-sysroot-native/usr/bin/python3-native/python3 extension module directory... ${libdir}/python3.10/site-packages
17:42:51 | ../git/configure: line 5691: syntax error near unexpected token `PHOSPHOR_DBUS_INTERFACES,'
17:42:51 | ../git/configure: line 5691: `PKG_CHECK_MODULES(PHOSPHOR_DBUS_INTERFACES, phosphor-dbus-interfaces)'
```
https://jenkins.openbmc.org/job/ci-openbmc/8045/distro=ubuntu,label=docker-builder,target=romulus/console

williamspatrick, 2021-12-08:
> Weird phosphor-settings-manager failure:
>
> ```
> 17:42:51 | checking for [...]

That looks like something related to autotools and not OpenSSL.  We should track that separately.  I'm guessing the PKG_CHECK_MODULES had some syntax change (which is hopefully backwards compatible because if it isn't we're going to just have to bite the bullet and convert everything to Meson).

williamspatrick, 2021-12-08:
I'm fixing the settings one as well: 

> remote:   https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/49448 phosphor-settings-manager: add `inherit pkgconfig` [NEW]

williamspatrick, 2021-12-08:
Potential fix for bmcweb is here: 

> remote:   https://gerrit.openbmc-project.xyz/c/openbmc/bmcweb/+/49450 ssl_key_handler: support OpenSSL 3.0 [NEW]

I still need to do some testing of it.

anoo1, 2021-12-08:
openpower fix: https://gerrit.openbmc-project.xyz/c/openbmc/openpower-pnor-code-mgmt/+/49475

williamspatrick, 2021-12-08:
@anoo1 @geissonator : More changes for phosphor-bmc-code-mgmt...  I think this covers all the OpenSSL APIs.

remote:   https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-bmc-code-mgmt/+/49480 openssl: stop using deprecated SHA512_* functions [NEW]
remote:   https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-bmc-code-mgmt/+/49481 openssl: test: stop using deprecated SHA512_* functions [NEW]

anoo1, 2021-12-09:
> @anoo1 @geissonator : More changes for phosphor-bmc-code-mgmt... I think this covers all the OpenSSL APIs.

openpower side: https://gerrit.openbmc-project.xyz/c/openbmc/openpower-pnor-code-mgmt/+/49555

edtanous, 2022-05-23:
OpenSSL 3 is now the default on OpenBMC.  Closing.

closed 2022-05-23 by edtanous