openbmc/openbmc #3831: do compile webui error

liuzepeng980510, 2021-11-09:
Build Configuration:
BB_VERSION           = "1.51.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "romulus"
DISTRO               = "openbmc-openpower"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta                 
meta-oe              
meta-networking      
meta-perl            
meta-python          
meta-security        
meta-phosphor        
meta-aspeed          
meta-openpower       
meta-romulus         = "master:b2afe3b9caeb2be3aac9f81022bb8fc656ab5a4e"

Sstate summary: Wanted 9 Local 0 Network 0 Missed 9 Current 1502 (0% match, 99% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
NOTE: Running task 2907 of 4116 (/home/openbmc/work/openbmc/meta-phosphor/recipes-phosphor/webui/phosphor-webui_git.bb:do_compile)
NOTE: recipe phosphor-webui-1.0+gitAUTOINC+f695bbb4c9-r1: task do_compile: Started
NOTE: Running task 4036 of 4116 (/home/openbmc/work/openbmc/meta-aspeed/recipes-bsp/u-boot/u-boot-aspeed_2016.07.bb:do_compile)
NOTE: recipe u-boot-aspeed-1_v2016.07+gitAUTOINC+1ded9fa3a2-r0: task do_compile: Started
ERROR: u-boot-aspeed-1_v2016.07+gitAUTOINC+1ded9fa3a2-r0 do_compile: oe_runmake failed
ERROR: u-boot-aspeed-1_v2016.07+gitAUTOINC+1ded9fa3a2-r0 do_compile: ExecutionError('/home/openbmc/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+1ded9fa3a2-r0/temp/run.do_compile.2960', 1, None, None)
ERROR: Logfile of failure stored in: /home/openbmc/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/u-boot-aspeed/1_v2016.07+gitAUTOINC+1ded9fa3a2-r0/temp/log.do_compile.2960
NOTE: recipe u-boot-aspeed-1_v2016.07+gitAUTOINC+1ded9fa3a2-r0: task do_compile: Failed
ERROR: Task (/home/openbmc/work/openbmc/meta-aspeed/recipes-bsp/u-boot/u-boot-aspeed_2016.07.bb:do_compile) failed with exit code '1'
ERROR: phosphor-webui-1.0+gitAUTOINC+f695bbb4c9-r1 do_compile: ExecutionError('/home/openbmc/work/openbmc/build/romulus/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+f695bbb4c9-r1/temp/run.do_compile.2956', 1, None, None)
ERROR: Logfile of failure stored in: /home/openbmc/work/openbmc/build/romulus/tmp/work/all-openbmc-linux/phosphor-webui/1.0+gitAUTOINC+f695bbb4c9-r1/temp/log.do_compile.2956
NOTE: recipe phosphor-webui-1.0+gitAUTOINC+f695bbb4c9-r1: task do_compile: Failed
ERROR: Task (/home/openbmc/work/openbmc/meta-phosphor/recipes-phosphor/webui/phosphor-webui_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4091 tasks of which 4089 didn't need to be rerun and 2 failed.

wangmin239, 2021-11-19:
The log is not helpful to find the cause of compiling errors. You may need make efforts to dig into why the errors generated.

edtanous, 2022-05-23:
Suspect this is a duplicate of a prior bug that was resolved in yocto, and hasn't had any input for several months.  Closing, please reopen if this is still and issue and not a duplicate.

closed 2022-05-23 by edtanous