openbmc/openbmc #3834: meta-evb-ast2500 with phosphor-webui, but how to include?

Knogle, 2021-11-13:
Hey friends.
Thanks to all of you, for developing such a great solution.
I currently have an ASRock X470D4U motherboard and i want to get started with OpenBMC.
I was able to get a build running, using the meta-evb-ast2500 target, and i am able to access it through SSH, but unfortunately not through web interface. As i have seen in a bunch of similar requests, phosphor-webui is not part of the build by default.
But i didn't find any great answer regarding my question, what's the appropriate way, to include the phosphor-webui or some kind of webui to my build?

Thanks in advance!

wangmin239, 2021-11-19:
@Knogle ,  I think that you may add the below code snippets  to  your file whose name is similar as 'meta-ibm/meta-romulus/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend'.  And you may get help from by skimming through the [manul](https://github.com/openbmc/docs/blob/master/development/add-new-system.md) which is provied by OpenBMC.

`RDEPENDS_${PN}-extras_append_romulus = " phosphor-webui"`

edtanous, 2022-05-23:
Question was answered.  Closing.

closed 2022-05-23 by edtanous