openbmc/openbmc #3836: I2C sensor LM75 does not show on D-Bus and WebUI

LukeKungCJ, 2021-12-02:
Hi all,

I'm trying to add an LM75 temp sensor based on Romulus and run on the AST2500 EVB board, but I met some issues as follows:

**1. The default sensor w83773g does not show on the D-Bus and WebUI.**
Based on my understanding (may be wrong), even the w83773g does not physically exist, it should still be created on D-Bus and WebUI. BTW, I have not modified any code and recipe, but only modified the kernel DTS from NCSI to dedicate LAN.

**2. I try to add an LM75 sensor which is physically existing on the EVB board but it does not show on the D-Bus and WebUI.**
This time I modify the I2C part of DTS and successfully create an LM75 sensor and could read the value by the following path:
> /sys/bus/i2c/drivers/lm75/7-004d/hwmon/hwmon1/temp1_input

And could also see the following path has been created:
> ./sys/firmware/devicetree/base/ahb/apb/bus@1e78a000/i2c-bus@1c0/lm75@4d

And the boot-up message shows as follows:
> [    2.246388] w83773g 12-004c: error writing config rate register
> [    2.256768] lm75 7-004d: hwmon1: sensor 'lm75'

I think the driver part is ok, but when I run _busctl | grep Hwmon_, there is still only ADC and TACH shows on the D-Bus and without LM75:
[image]

For file phosphor-hwmon_%.bbappend, I only add one lm75 line.
>        CHIPS = " \
>        bus@1e78a000/i2c-bus@440/w83773g@4c \
>        bus@1e78a000/i2c-bus@1c0/lm75@4d \
>        pwm-tacho-controller@1e786000 \
>        "

Any help would be really appreciated.

closed 2021-12-20 by LukeKungCJ

yty2023, 2023-04-03:
Hi,I'm a noob for openbmc, i tried qemu  + romulus and have the same problem, could you tell me how you solve this prombem?

yty2023, 2023-04-03:
@LukeKungCJ Hi, i'm a rookie for openbmc, i met the same problem on qemu, may i know how you solve this problem?

Vinay-Rama, 2024-09-17:
@LukeKungCJ facing same issue in qemu can i get how this issue is resolved

martmartalb, 2025-05-27:
@LukeKungCJ @Vinay-Rama I'm probably very late to the conversation, but in case someone else runs into this issue, here's how I solved it.
### Context

1. I am using OpenBMC version 2.14.0, working with the Romulus image.
2. I am going to explain the process with the w83773g sensor, but the same can be applied for the LM75.
3. I modified the driver for the w83773g sensor so that it can be read even when the hardware is not physically present.
     * I used `devtool modify linux-aspeed` to patch the driver in the kernel source.
     * I modified the  `w83773_read` function to return a sawtooth waveform — values increase until a threshold, then wrap around to 0.
     * I made the `w83773_write` always returns success.
4. With this in place, I could see this fake sensor on D-Bus.
     * By running `busctl introspect xyz.openbmc_project.Hwmon-<sensor_id>.Hwmon1 /xyz/openbmc_project/sensors/temperature/outlet`, I got the fake sensor values for the outlet channel.
     * The ID of the sensor can be check in the config file by running `cat /etc/default/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@440/w83773g\@4c.conf`

### Additional Steps to see the Sensor on WebUI

Even after the sensor showed up on D-Bus, it wasn't visible in the WebUI, which uses the Redfish REST API. According to [Sensor Support for OpenBMC using phosphor-hwmon](https://gerrit.openbmc.org/plugins/gitiles/openbmc/docs/+/b6954a295de7076427b35eb733e013594edd5509/architecture/sensor-architecture.md), Redfish requires associations for sensors to be surfaced correctly. I followed the Type 1 association approach described there:

1. I created an `associations.json` file:
      * Path: `meta-ibm/meta-romulus/recipes-phosphor/inventory/phosphor-inventory-manager/associations.json`
      * Note that `phosphor-inventory-manager` directory did not exists, so I create it.
      * The content of this file is provided bellow:
```
[
    {
        "path": "system/chassis",
        "endpoints":
        [
            {
                "types":
                {
                    "rType": "chassis",
                    "fType": "all_sensors"
                },
                "paths":
                [
                    "/xyz/openbmc_project/sensors/temperature/inlet_cpu",
                    "/xyz/openbmc_project/sensors/temperature/inlet_io",
                    "/xyz/openbmc_project/sensors/temperature/outlet"
                ]
            }
        ]
    },
    {
        "path": "/xyz/openbmc_project/sensors/temperature/inlet_cpu",
        "endpoints":
        [
            {
                "types":
                {
                    "rType": "all_sensors",
                    "fType": "chassis"
                },
                "paths":
                [
                    "/xyz/openbmc_project/inventory/system/chassis"
                ]
            }
        ]
    },
    {
        "path": "/xyz/openbmc_project/sensors/temperature/inlet_io",
        "endpoints":
        [
            {
                "types":
                {
                    "rType": "all_sensors",
                    "fType": "chassis"
                },
                "paths":
                [
                    "/xyz/openbmc_project/inventory/system/chassis"
                ]
            }
        ]
    },
    {
        "path": "/xyz/openbmc_project/sensors/temperature/outlet",
        "endpoints":
        [
            {
                "types":
                {
                    "rType": "all_sensors",
                    "fType": "chassis"
                },
                "paths":
                [
                    "/xyz/openbmc_project/inventory/system/chassis"
                ]
            }
        ]
    }
]

```
2. I modify the bbappend file to install the JSON file:
    * Path: `meta-ibm/meta-romulus/recipes-phosphor/inventory/phosphor-inventory-manager_%.bbappend`
    * The content of this file is:
```
DEPENDS:append = " inventory-cleanup"
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

PACKAGECONFIG:append:romulus = " associations"
SRC_URI:append:romulus = " file://associations.json"

do_install:append:romulus() {
    install -d ${D}${base_datadir}
    install -m 0755 ${WORKDIR}/associations.json ${D}${base_datadir}/associations.json
}
```

After these modifications, I rebuild the image I was able to see the sensor values on the WebUI. I hope this peace of information helps you also. Note that I made the association of all the channels of the sensor with chassis, which is actually not correct. You can also link a low-level hardware item to its sensors (check the link above). This is suppose to serve as example of how to create the associations.