openbmc/openbmc #3837: FRU Device service did not start

damon3000, 2021-12-09:
**Note on Bugs**
1. I created a layer based on meta-romulus
2. Modify the aspeed-bmc-opp-romulus.dts to configure eeprom
```
&i2c3 {
        status = "okay";
        eeprom@50 {
                compatible = "atmel,24c64";
                reg = <0x50>;
        };
};

```
3. Modify the yaml file to add my fru configuration
```
1:
    /xyz/openbmc_project/FruDevice/XX0123456789ABEVT001:
        entityID: 3
        entityInstance: 1
        interfaces:
            xyz.openbmc_project.Inventory.Decorator.Asset:
                BuildDate:
                    IPMIFruProperty: Mfg Date
                    IPMIFruSection: Board
                Manufacturer:
                    IPMIFruProperty: Manufacturer
                    IPMIFruSection: Board
                PartNumber:
                    IPMIFruProperty: Part Number
                    IPMIFruSection: Board
                SerialNumber:
                    IPMIFruProperty: Serial Number
                    IPMIFruSection: Board
            xyz.openbmc_project.Inventory.Decorator.Revision:
                Version:
                    IPMIFruProperty: Custom Field 2
                    IPMIFruSection: Board
                    IPMIFruValueDelimiter: 58
            xyz.openbmc_project.Inventory.Item:
                PrettyName:
                    IPMIFruProperty: Name
                    IPMIFruSection: Board

```
4. Boot up BMC 

**Unexpected behavior you saw**
FRU Device service did not start
```
root@sky:/etc/systemd/system/multi-user.target.wants# systemctl status nbmc_project.FruDevice.service
* xyz.openbmc_project.FruDevice.service - Fru Device
     Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.FruDevice.service; disabled; vendor preset: disabled)
     Active: inactive (dead)
root@sky:
```
Checked the preset file, but found that:
```
/lib/systemd/system-preset/98-fru-device.preset /lib/systemd/system/xyz.openbmc_project.FruDevice.service
```

**Expected behavior**
FRU Device service can start normal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**OpenBMC Information:**
```
root@sky:/etc/systemd/system/multi-user.target.wants# cat /etc/os-release 
ID=openbmc-romulus
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.9.0-0"
VERSION_ID=2.9.0-0-g835472cca-dirty
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.9.0-0"
BUILD_ID="2.9.0"
OPENBMC_TARGET_MACHINE="sky"
root@sky:/etc/systemd/system/multi-user.target.wants# uname -a
Linux sky 5.8.17 #1 Mon Dec 6 09:41:35 UTC 2021 armv6l GNU/Linux
root@sky:/etc/systemd/system/multi-user.target.wants#
```

damon3000, 2022-01-12:
I download the tag 2.9.0 and built it using cleaning code(do not create layer)again. The service still did not start. No any reply for one month. Seems that nobody maintances the released tag.

damon3000, 2022-02-07:
No solution, no reply, close it.

edtanous, 2022-05-23:
Romulus as a target does not support entity-manager or FruDevice.  I'd probably recommend starting with a platform that uses Entity-Manager, but its difficult to triage your issue without knowing what you modified.

Closing as requested.

closed 2022-05-23 by edtanous