openbmc/openbmc #3839: Update uboot 2019 with ast2500, but kernel panic

damon3000, 2022-02-08:
**Note on Bugs**
I used meta-romulus to update uboot to 2019, but occurred kernel panic.

**Unexpected behavior you saw**
I used meta-romulus to update uboot to 2019, but occurred kernel panic.

**Expected behavior**
BMC can boot up normal.

**To Reproduce**
Steps to reproduce the behavior:
1. Modify the conf/machine/romulus.conf **based on the latest openbmc code according to the commit** 

> Author: Thang Q. Nguyen <thang@os.amperecomputing.com>  2021-12-23 12:48:11
Committer: Thang Q. Nguyen <thang@os.amperecomputing.com>  2022-01-21 16:19:19
Parent: 8cb21e7ea509e802e5419f7cfa37d38737193ea2 (meta-quanta: s6q: Add x86-power-ctrl config files)
Child:  04777e3acd3b4342f74d3f74cf936a9c7507f476 (libpeci: srcrev bump 6a00e9aa72..bdefaa3c95)
Branches: master, remotes/origin/master
Follows: 2.11.0-dev
Precedes: 

    meta-ampere: mtjade: switch to use u-boot 2019.04

`
-PREFERRED_PROVIDER_virtual/bootloader = "u-boot-aspeed"
-PREFERRED_PROVIDER_u-boot = "u-boot-aspeed"
-PREFERRED_PROVIDER_u-boot-fw-utils = "u-boot-fw-utils-aspeed"
-UBOOT_MACHINE = "ast_g5_phy_config"
+UBOOT_MACHINE = "evb-ast2500_defconfig"
+UBOOT_DEVICETREE = "ast2500-evb"
`
2. Build the image
3. Run the image using qemu
4. See error
`U-Boot 2019.04 (Oct 08 2021 - 00:31:50 +0000)

SOC : AST2500-A1 
RST : Power On 
2nd Boot : Enable
LPC Mode : SIO:Enable : SuperIO-2e
Eth : MAC0: RMII/NCSI, , MAC1: RMII/NCSI, 
Model: AST2500 EVB
DRAM:  448 MiB
MMC:   sdhci_slot0@100: 0, sdhci_slot1@200: 1
Loading Environment from SPI Flash... unrecognized JEDEC id bytes: 20, ba, 19
*** Warning - spi_flash_probe_bus_cs() failed, using default environment

In:    serial@1e784000
Out:   serial@1e784000
Err:   serial@1e784000
Net:   
Warning: ethernet@1e660000 (eth0) using random MAC address - 7a:13:a1:65:3b:ab
eth0: ethernet@1e660000
Warning: ethernet@1e680000 (eth1) using random MAC address - fe:b2:4a:9c:96:a6
, eth1: ethernet@1e680000
Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20100000 ...
   Using 'conf-aspeed-bmc-opp-romulus.dtb' configuration
   Trying 'kernel-1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20100128
     Data Size:    3122624 Bytes = 3 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha256
     Hash value:   145fbfd443847f9739d2dee3581dcaa2543520316b9378662db75c15fb1557ea
[... 596 lines cut ...]
        mount /dev/mtdblock5 run/initramfs/rw -t jffs2 -o rw
or perform a factory reset with the clean-rwfs-filesystem option.
Fatal error, triggering kernel panic!
[    6.118411] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
[    6.119076] CPU: 0 PID: 1 Comm: init Not tainted 5.15.14-a86c6a5 #1
[    6.119503] Hardware name: Generic DT based system
[    6.119898] Backtrace: 
[    6.120226] [<807d2bc4>] (dump_backtrace) from [<807d2dbc>] (show_stack+0x20/0x24)
[    6.121060]  r7:808f25c8 r6:80b0a0e8 r5:00000000 r4:80901dd8
[    6.121394] [<807d2d9c>] (show_stack) from [<807dd058>] (dump_stack+0x28/0x30)
[    6.121783] [<807dd030>] (dump_stack) from [<807d3154>] (panic+0xe8/0x30c)
[    6.122137]  r5:00000000 r4:80ba62b0
[    6.122314] [<807d306c>] (panic) from [<801192c8>] (do_exit+0x8e0/0x938)
[    6.122648]  r3:00000001 r2:00000000 r1:00000100 r0:808f25c8
[    6.122906]  r7:ffffe000
[    6.123037] [<801189e8>] (do_exit) from [<801193b8>] (do_group_exit+0x4c/0xb4)
[    6.123426]  r7:000000f8
[    6.123560] [<8011936c>] (do_group_exit) from [<80119440>] (__wake_up_parent+0x0/0x2c)
[    6.124093]  r5:00000004 r4:01e31190
[    6.124303] [<80119420>] (sys_exit_group) from [<80100040>] (ret_fast_syscall+0x0/0x48)
[    6.124874] Exception stack(0x80caffa8 to 0x80cafff0)
[    6.125236] ffa0:                   01e31190 00000004 00000001 00000000 00000001 01e31190
[    6.125648] ffc0: 01e31190 00000004 00000004 000000f8 00000000 76e89e64 0056acd8 7eef0f10
[    6.126059] ffe0: 0056bb54 7eef0a04 004f2ee4 76dd45e4
`

koweda, 2022-02-09:
The romulus qemu only emulates 32MB of spi flash, so if you use a 64MB layout you will get kernel panic.
You may need to build modified qemu, or use a 32MB layout.

32MB layout example:
https://gist.github.com/koweda/9d12e259b5b5c7b2a4cdf94c905e3ba8

damon3000, 2022-02-11:
I offline updated the image to a 64M flash and run it on the real machine, the issue still exist. Cannot the romulus use 64M on the real machine?

koweda, 2022-02-11:
Did you modify openbmc-flash-layout.dtsi to openbmc-flash-layout-64.dtsi in the device tree?

damon3000, 2022-02-14:
No, I did not modify it. According to your suggestion, it works, thank you.

closed 2022-02-14 by damon3000

QGB, 2023-03-15:
```

2023-03-15__19.10.47__.524 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 2001
2023-03-15__19.12.25__.408 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 4001
2023-03-15__19.13.54__.197 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 6001
2023-03-15__19.15.22__.464 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 8001
2023-03-15__19.16.53__.420 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 10001
2023-03-15__19.18.23__.173 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 12001
2023-03-15__19.19.51__.918 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 14001
2023-03-15__19.21.19__.634 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 16001
2023-03-15__19.22.47__.234 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 18001
2023-03-15__19.24.15__.355 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 20001
2023-03-15__19.25.43__.909 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 22001
2023-03-15__19.27.11__.999 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 24001
2023-03-15__19.28.40__.095 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 26001
2023-03-15__19.30.07__.738 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 28001
2023-03-15__19.31.34__.836 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 30001
2023-03-15__19.33.02__.360 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 32001
2023-03-15__19.34.29__.631 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 34001
2023-03-15__19.35.56__.655 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 36001
2023-03-15__19.37.34__.066 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 38001
2023-03-15__19.39.20__.254 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 40001
2023-03-15__19.40.50__.665 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 42001
2023-03-15__19.43.11__.221 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 44001
2023-03-15__19.56.45__.176 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 46001
2023-03-15__20.10.20__.641 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 48001
2023-03-15__20.23.55__.879 <Thread(Thread-3, started 6376)> ('192.168.1.48', 64586) 50001
```

ganwengithub, 2024-04-02:
this issue still existing in obmc

williamspatrick, 2024-04-02:
> this issue still existing in obmc

Please open a new issue with your exact scenario.  This issue was previously closed as a user error between the build and their QEMU arguments.