openbmc/openbmc #3841: Compile failed with honister branch

KunLuffy, 2022-02-15:
Using honister branch, executed command:
. setup romulus
bitbake obmc-phosphor-image

Then report error log:
ERROR: No recipes in default available for:
  /home/baikun/zd-tech/openbmc-v211/openbmc-v211-next/meta-phosphor/recipes-core/systemd/systemd_249.3.bbappend

Maybe need to delete this file? Because systemd upgrade to 249.7 in poky project.

williamspatrick, 2022-02-15:
I'll fix this.  I probably pulled the upstream security fixes from Yocto that updated the underlying systemd recipe.  

I need to confirm if the patches we have are all fixed in 249.7 or if we need to rename the bbappend.

KunLuffy, 2022-02-16:
Hi Williams, 
According your message, I checked the patch which 249.3 bbappend include and confirmed 249.7 doesn't fixed. Then I rename bbappend file to 249.7, it can build passed.

williamspatrick, 2022-02-16:
A fix is pending in Gerrit. https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/51228

williamspatrick, 2022-02-16:
Fixed via 1a38de9b8a6a9532f1de9ff4648037b20edbf16a

KunLuffy, 2022-02-17:
Problem resolved. Thanks a lot.

closed 2022-02-17 by KunLuffy