openbmc/openbmc #3842: How to select target hardware

z13986553031, 2022-02-24:
My hardware is Xilinx Zynq-7000,Which installation should I choose from the following：
bletchley               gsj                     romulus
dl360poc                kudo                    s2600wf
e3c246d4i               mihawk                  swift
ethanolx                mtjade                  tiogapass
evb-ast2500             nicole                  transformers
evb-ast2600             olympus-nuvoton         witherspoon
evb-npcm750             on5263m5                witherspoon-tacoma
f0b                     p10bmc                  x11spi
fp5280g2                palmetto                yosemitev2
g220a                   qemuarm                 zaius
gbs                     quanta-q71l

williamspatrick, 2022-02-24:
That is not a supported system. At one point we did have a meta-xilinx but they didn’t maintain their subtree with support for the latest Yocto and so it needed to be dropped.

Vishnu070803, 2025-12-11:
> That is not a supported system. At one point we did have a meta-xilinx but they didn’t maintain their subtree with support for the latest Yocto and so it needed to be dropped.

Can we use the older versions now?