openbmc/openbmc #3845: webui-vue/phosphor-webui compilation fails

pratikfarkase94, 2022-03-08:
Hello OpenBMC community,

I am trying to build and openbmc image for raspberry pi 3 with WebUI. But the compilation for webui-vue/phosphor-webui won't complete and stuck in an infinite loop resulting in obmc-phosphor-image compilation fail. Below are the logs. Kindly assist me. 

`npm WARN old lockfile FetchError: request to https://registry.npmjs.org/supports-color failed, reason: read ECONNRESET
npm WARN old lockfile     at ClientRequest.<anonymous> (/media/pratik/ee/openbmc/build/tmp/work/all-openbmc-linux/webui-vue/1.0+gitAUTOINC+19b2cfba73-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/minipass-fetch/lib/index.js:110:14)
npm WARN old lockfile     at ClientRequest.emit (node:events:390:28)
npm WARN old lockfile     at TLSSocket.socketErrorListener (node:_http_client:447:9)
npm WARN old lockfile     at TLSSocket.emit (node:events:402:35)
npm WARN old lockfile     at emitErrorNT (node:internal/streams/destroy:157:8)
npm WARN old lockfile     at emitErrorCloseNT (node:internal/streams/destroy:122:3)
npm WARN old lockfile     at processTicksAndRejections (node:internal/process/task_queues:83:21)
npm WARN old lockfile  Could not fetch metadata for supports-color@5.5.0 FetchError: request to https://registry.npmjs.org/supports-color failed, reason: read ECONNRESET
npm WARN old lockfile     at ClientRequest.<anonymous> (/media/pratik/ee/openbmc/build/tmp/work/all-openbmc-linux/webui-vue/1.0+gitAUTOINC+19b2cfba73-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/minipass-fetch/lib/index.js:110:14)
npm WARN old lockfile     at ClientRequest.emit (node:events:390:28)
npm WARN old lockfile     at TLSSocket.socketErrorListener (node:_http_client:447:9)
npm WARN old lockfile     at TLSSocket.emit (node:events:402:35)
npm WARN old lockfile     at emitErrorNT (node:internal/streams/destroy:157:8)
npm WARN old lockfile     at emitErrorCloseNT (node:internal/streams/destroy:122:3)
npm WARN old lockfile     at processTicksAndRejections (node:internal/process/task_queues:83:21) {
npm WARN old lockfile   code: 'ECONNRESET',
npm WARN old lockfile   errno: 'ECONNRESET',
npm 
WARN old lockfile   syscall: 'read',
npm WARN old lockfile   type: 'system'
npm WARN old lockfile }

npm WARN old lockfile FetchError: request to https://registry.npmjs.org/@babel%2fhelper-compilation-targets failed, reason: read ECONNRESET
npm WARN old lockfile     at ClientRequest.<anonymous> (/media/pratik/ee/openbmc/build/tmp/work/all-openbmc-linux/webui-vue/1.0+gitAUTOINC+19b2cfba73-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/minipass-fetch/lib/index.js:110:14)
npm
 WARN old lockfile     at ClientRequest.emit (node:events:390:28)
npm WARN old lockfile     at TLSSocket.socketErrorListener (node:_http_client:447:9)
npm WARN old lockfile     at TLSSocket.emit (node:events:402:35)
npm WARN old lockfile     at emitErrorNT (node:internal/streams/destroy:157:8)
npm WARN old lockfile     at emitErrorCloseNT (node:internal/streams/destroy:122:3)
npm WARN old lockfile     at processTicksAndRejections (node:internal/process/task_queues:83:21)
npm WARN old lockfile  Could not fetch metadata for @babel/helper-compilation-targets@7.13.10 FetchError: request to https://registry.npmjs.org/@babel%2fhelper-compilation-targets failed, reason: read ECONNRESET
npm WARN old lockfile     at ClientRequest.<anonymous> (/media/pratik/ee/openbmc/build/tmp/work/all-openbmc-linux/webui-vue/1.0+gitAUTOINC+19b2cfba73-r1/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/minipass-fetch/lib/index.js:110:14)
npm WARN old lockfile     at ClientRequest.emit (node:events:390:28)
npm WARN old lockfile     at TLSSocket.socketErrorListener (node:_http_client:447:9)
npm WARN old lockfile     at TLSSocket.emit (node:events:402:35)
npm WARN old lockfile     at emitErrorNT (node:internal/streams/destroy:157:8)
npm WARN old lockfile     at emitErrorCloseNT (node:internal/streams/destroy:122:3)
npm WARN old lockfile     at processTicksAndRejections (node:internal/process/task_queues:83:21) {
npm WARN old lockfile   code: 'ECONNRESET',
npm WARN old lockfile   errno: 'ECONNRESET',
npm WARN old lockfile   syscall: 'read',
npm WARN old lockfile   type: 'system'
npm WARN 
old lockfile }

npm http fetch GET 200 https://registry.npmjs.org/chalk 71000ms (cache stale)

Currently  1 running tasks (1788 of 1788/3339 of 4356)  76% |###############################################################################                         |
0: webui-vue-1.0+gitAUTOINC+19b2cfba73-r1 do_compile - 1m45s (pid 2132)

`

williamspatrick, 2022-03-08:
This appears to be a network connection issue on your end connecting to the NPM repositories to download the source code?

pratikfarkase94, 2022-03-08:
I checked with 5 different connections. None working.

gtmills, 2022-03-08:
Saw some hits on a quick search: https://stackoverflow.com/questions/18419144/npm-not-working-read-econnreset
Maybe try `rm -rf ~/.npm`

Have a proxy set? Need a proxy?
https://github.com/openbmc/openbmc/commit/31ea32d3953358fa63c97b5e5e0d22cf5dd23cfb

pratikfarkase94, 2022-03-09:
> Saw some hits on a quick search: https://stackoverflow.com/questions/18419144/npm-not-working-read-econnreset Maybe try `rm -rf ~/.npm`
> 
> Have a proxy set? Need a proxy? [31ea32d](https://github.com/openbmc/openbmc/commit/31ea32d3953358fa63c97b5e5e0d22cf5dd23cfb)

Tried this but still not working.

pratikfarkase94, 2022-03-09:
I am completely stuck here. Cannot proceed further. I had built an image back in November on the same machine i am using now with the same connection but did not face any issues that time.

Bigmounten, 2022-03-20:
> I am completely stuck here. Cannot proceed further. I had built an image back in November on the same machine i am using now with the same connection but did not face any issues that time.

That's strange , I met this bug , too

Bigmounten, 2022-03-21:
Hey, I find the problem , because the npm version is too old ,so it has to fetch data to fix ,but the complile file didn't open the network ,try to change the git file(openbmc/meta-phosphor/recipes-phosphor/webui/phosphor-webui_git.bb) into this:
python __anonymous() {
  d.appendVarFlag('do_compile', 'network', '1') 
}

do_compile () {
  cd ${S}
  rm -rf node_modules
  npm --loglevel info --proxy=${http_proxy} --https-proxy=${https_proxy} install
  npm run-script build
}

mdmillerii, 2022-03-21:
https://github.com/openbmc/openbmc/blob/35323b9bcd18a1fbc1e134e7489f114c37bd85df/meta-phosphor/recipes-phosphor/webui/webui-vue_git.bb#L38 was added mid February

williamspatrick, 2023-09-25:
Resolved.

closed 2023-09-25 by williamspatrick