openbmc/openbmc #3851: evb-ast2500 bitbake warnings

shenki, 2022-04-27:
This issue is to track the cleanup of the following warnings:

```
. setup evb-ast2500
bitbake obmc-phosphor-image
```

```
WARNING: Layer evb should set LAYERSERIES_COMPAT_evb in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-aspeed should set LAYERSERIES_COMPAT_evb-aspeed in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-ast2500 should set LAYERSERIES_COMPAT_evb-ast2500 in its conf/layer.conf file to list the core layer names it is compatible with.
Loading cache: 100% |###########################################################################################| Time: 0:00:00
Loaded 4160 entries from dependency cache.
Parsing recipes: 100% |#########################################################################################| Time: 0:00:00
Parsing of 2683 .bb files complete (2682 cached, 1 parsed). 4161 targets, 376 skipped, 0 masked, 0 errors.
WARNING: No bb files in default matched BBFILE_PATTERN_evb ''
WARNING: No bb files in default matched BBFILE_PATTERN_evb-aspeed ''
WARNING: No bb files in default matched BBFILE_PATTERN_evb-ast2500 ''
```

shenki, 2022-04-27:
https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/53289

edtanous, 2022-05-23:
Aforementioned patchset seems to have been merged, and fixes this issue.  Closing.

shenki, 2022-06-08:
Hi @edtanous , did you test this was fixed before closing this issue?

edtanous, 2022-06-08:
I didn't test on that specific platform, but the warnings were gone for the qemu platform I was using, and the patchset merged so I assumed it was complete.

Did I mess up?  Does this need reopened?

blackcatevil, 2022-06-10:
I still see these warnings while building evb-ast2500/evb-ast2600

edtanous, 2022-06-10:
Understood, yep, closed this too soon.

blackcatevil, 2022-06-11:
Please review this commit to see if it can fix this issue
https://gerrit.openbmc.org/c/openbmc/openbmc/+/53184

williamspatrick, 2023-09-25:
This has been resolved by a1e084fdb5c , 0823bd2e6b8 , and dd6b1a1e16d.

closed 2023-09-25 by williamspatrick