openbmc/openbmc #3852: [AST2500 EVB] Qemu Boot on AST2500 EVB

qkpidan, 2022-04-27:
**Note on Bugs**
Hi Boss，i can't start qemu whith ast2500-evb when Compilation complete.
Help, please !

**Unexpected behavior you saw**
A clear and concise description of what unexpectedly happened.

**Expected behavior**
$ qemu-system-arm --version
QEMU emulator version 6.1.0 (v6.1.0-645-g2856503299-dirty)
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers
$ git log
commit 439a39392cfde76e727b0622e66475e9e8959e29 (HEAD -> master, origin/master, origin/HEAD)

**To Reproduce**
$ export TEMPLATECONF=meta-aspeed/conf
$ TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf
$ . openbmc-env 
$ bitbake obmc-phosphor-image
$ qemu-system-arm -m 1024 -M ast2500-evb -nographic -drive file=~/openbmc/build/tmp/deploy/images/evb-ast2500/obmc-phosphor-image-evb-ast2500-20851208213409.static.mtd,format=raw,if=mtd -net nic -net user,hostfwd=:127.0.0.1:2222-:22,hostfwd=:127.0.0.1:2443-:443,hostname=qemu

**Screenshots**
[image]

**OpenBMC Information:**
 - /etc/os-release:
 -$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

 - uname -a:
 -$ uname -a
Linux openbmc 4.15.0-156-generic #163-Ubuntu SMP Thu Aug 19 23:31:36 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

LeeTroy, 2022-04-27:
Please try to `bootm 20080000` in u-boot. If it works, please check this PR:
https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/53257

qkpidan, 2022-04-27:
It works!
Many thanks!

> Please try to `bootm 20080000` in u-boot. If it works, please check this PR: https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/53257

closed 2022-04-27 by qkpidan