openbmc/openbmc #3853: Service xyz.openbmc_project.State.Host@0.service failed to load properly

radix-platform, 2022-04-27:
**Note on Bugs**
Revision d10de96fc207d48b8357575833a6ae92a9f4ff0f master branch
static.mtd image built for s2600wf hardware

Boot process freezes with following messages:

[    3.997914] systemd[1]: Hostname set to <s2600wf>.
[    4.793293] systemd[110]: /lib/systemd/system-generators/systemd-gpt-auto-generator failed with exit status 1.
[    5.668371] systemd[1]: Failed to put bus name to hashmap: File exists
[    5.668894] systemd[1]: xyz.openbmc_project.State.Host@0.service: Two services allocated for the same bus name xyz.openbmc_project.State.Host0, refusing operation.
[    6.872636] systemd[1]: /lib/systemd/system/phosphor-ipmi-net@.socket:3: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
[    6.938917] systemd[1]: Failed to isolate default target: Unit xyz.openbmc_project.State.Host@0.service failed to load properly, please adjust/correct and reload service manager: File exists
[!!!!!!] Failed to isolate default target.
[    7.062976] systemd[1]: Freezing execution.

**Unexpected behavior you saw**
OS is freeze.

**Expected behavior**
Boot success till login

**To Reproduce**
Steps to reproduce the behavior:
1. source ./setup s2600wf
2. bitbake obmc-phosphor-image
3. run qemu:
sudo \
  qemu-system-arm \
    -m 512 \
    -M ast2500-evb,fmc-model=mx25l25635e,spi-model=mx25l25635e \
    -drive file=static.mtd,format=raw,if=mtd \
    -nic tap,ifname=tap0,mac=C0:FF:EE:00:00:02,model=ftgmac100,script=no,downscript=no \
    -nographic

4. See error

[    3.997914] systemd[1]: Hostname set to <s2600wf>.
[    4.793293] systemd[110]: /lib/systemd/system-generators/systemd-gpt-auto-generator failed with exit status 1.
[    5.668371] systemd[1]: Failed to put bus name to hashmap: File exists
[    5.668894] systemd[1]: xyz.openbmc_project.State.Host@0.service: Two services allocated for the same bus name xyz.openbmc_project.State.Host0, refusing operation.
[    6.872636] systemd[1]: /lib/systemd/system/phosphor-ipmi-net@.socket:3: Invalid interface name, ignoring: sys-subsystem-net-devices-%i.device
[    6.938917] systemd[1]: Failed to isolate default target: Unit xyz.openbmc_project.State.Host@0.service failed to load properly, please adjust/correct and reload service manager: File exists
[!!!!!!] Failed to isolate default target.
[    7.062976] systemd[1]: Freezing execution.

**OpenBMC Information:**
Revision d10de96fc207d48b8357575833a6ae92a9f4ff0f master branch

**Additional context**
QEMU 6.2.0 which works properly when image is built from master branch revision: f8a7774abd5e8650038240d45f590fb3cdc79fb3

Probably this issue related to phosphor-state-manager changes on Apr 22, 2022

radix-platform, 2022-04-28:
Do we have remove phosphor-state-manager from image and use only x86-power-control for Intel boards?

radix-platform, 2022-04-28:
Previously Intel meta-intel-openbmc layer includes both phpsphor-state-manager and x86-power-control. What we have to do now:
1) simply remove phosphor-state-manager ?
or
2) made the patch for removing Host and Chassis service form x86-power-control or from phosphor?
or
something else ?

jmbills, 2022-08-02:
Sorry for the delay.  Is this still an issue?  What is the proposed solution?  Do we need to add something to meta-intel-openbmc to use x86-power-control instead of PSM?

radix-platform, 2022-08-02:
I think is still an issue because phosphor-state-manager and x86-power-control delivered systemd-units with same names for 'Host@0'. Probably you have to change packagegroups or explicitly exclude PSM from meta-intel-openbmc layer. May be we have to install PSM partially when used x86-power-control or add into x86-power-control something more (for example in PSM we have useful obmcutil but it is not ported for Intel machines).

Any way s2600hf target is buildable but doesn't start.

jmbills, 2022-08-03:
> I think is still an issue because phosphor-state-manager and x86-power-control delivered systemd-units with same names for 'Host@0'.

This is expected as they both perform the same function and are mutually exclusive.

> Probably you have to change packagegroups or explicitly exclude PSM from meta-intel-openbmc layer.

In our downstream `conf/machine/include/intel.inc`, we set 
```VIRTUAL-RUNTIME_obmc-host-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-chassis-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-discover-system-state = "x86-power-control"
```
to replace PSM with x86-power-control.  Maybe this change needs to be added upstream to `meta-intel-openbmc/conf/machine/include/intel.inc`?  I'm happy to submit or review a patch for that.

> May be we have to install PSM partially when used x86-power-control or add into x86-power-control something more (for example in PSM we have useful obmcutil but it is not ported for Intel machines).

I haven't ever used obmcutil.  If it's possible to split it out from the state management portion of PSM and build it as a standalone utility, that should work.

> 
> Any way s2600hf target is buildable but doesn't start.

Yes, this is possible.  We haven't been able to maintain the s2600wf target.  It may be time to remove it if it's no longer useful.

radix-platform, 2022-08-03:
About obmcutil: this script is very useful and I port it to be able work with x86-power-control like on powerpc machines. After all, there were times when openbmc did not have a graphical interface and people were used to working on the command line.

Thank you for following lines:

```
VIRTUAL-RUNTIME_obmc-chassis-state-manager = "x86-power-control"
VIRTUAL-RUNTIME_obmc-discover-system-state = "x86-power-control"
```

I will try its for my target.

s2600wf I run on qemu as a starting example and I think this layer can be used by beginners as a tutorial and first working BMC:

```
  qemu-system-arm \
    -m 512 \
    -M ast2500-evb,fmc-model=mx25l25635e,spi-model=mx25l25635e \
    -drive file=${DEPLOY_DIR}/${MTD_IMAGE},format=raw,if=mtd \
    -nic tap,ifname=tap7,mac=00:16:3e:48:98:08,model=ftgmac100,script=no,downscript=no \
    -rtc clock=vm,base=`date +"%Y-%m-%dT%H:%M:%S"` \
    -nographic
```

zpdsg123, 2022-08-15:
> About obmcutil: this script is very useful and I port it to be able work with x86-power-control like on powerpc machines. After all, there were times when openbmc did not have a graphical interface [...]

how do you port obmcutil?

AswathSrinivasan, 2023-04-21:
Hi @williamspatrick, @jmbills ,
I have taken latest build for AST2600 and build the image and booted via qemu - it is booting as normal.

In order to check the Host power action, i am enabling x86-Power-control, after included the changes in local.conf, build got success, but when on booting it got hanged with the error message.

[image]

1. So, i tried the above step, https://github.com/openbmc/openbmc/issues/3853#issuecomment-1204041312, 
but even with the same step i am getting error.

2. I also tried disabling phosphor-state-manager by adding the line,
 IMAGE_INSTALL:remove =" phosphor-state-manager" in local.conf.

So I Have tried these above 2 steps to add x86-P-C but I cannot resolve this issue.
Kindly guide on the same.

jmbills, 2023-05-15:
That error message is indicating that there are two services allocated for `xyz.openbmc_project.State.Host0`.  If you have enabled x86-power-control, that should be one of them.  Can you search in your image to find the other service?  Maybe in the `.service` files?  Or `busctl | grep Host0` may show the currently loaded service for that bus name?

AswathSrinivasan, 2023-05-16:
[image]
I have attached the busctl | grep Host0 output for your reference.

radix-platform, 2023-05-19:
> how do you port obmcutil?

obmcutil - is a script and we can change some busctl calls to be able access x86 specific services. [(https://git.radix.pro/BMC/OpenBmc/openbmc.git/plain/meta-ibs/meta-common/recipes-ibs/ibmcutil/ibmcutil/ibmcutil?h=sila-32MiB)]

jmbills, 2023-05-19:
> [image] I have attached the busctl | grep Host0 output for your reference.

I think this means that the `phosphor-host-s*` application is running and has already loaded the State.Host@0 service.  If you can stop `phosphor-host-s*`, does that allow x86-power-control to load successfully?

williamspatrick, 2023-09-25:
These two services are not intended to work together.  They are mutually exclusive.

If you'd like to contribute code changes such that obmcutil is installed in a separate runtime package, you're welcome to do so.

closed 2023-09-25 by williamspatrick

IgorLinux1990, 2023-11-21:
When trying to run s2600wf on qemu, I get the following errors:
xyz.openbmc_project.State.Host@0.service: Two services allocated for the same bus name xyz.openbmc_project.State.Host0, refusing operation.
Failed to isolate default target: Unit xyz.openbmc_project.State.Host@0.service failed to load properly, please adjust/correct and reload service manager: File exists
This problem has already been solved, I understand it. Here is the link:
https://github.com/openbmc/openbmc/issues/3853
I have added the following lines to the file meta-intel-openbmc/conf/machine/include/intel.inc(attachments).
I reassembled it, but the error still remained. Please tell me what I did wrong?

williamspatrick, 2023-11-21:
> Please tell me what I did wrong?

You’re not likely to get any help on a closed issue if you haven’t built it from the latest master. 

I’d suggest opening a new issue with the information appropriate for your situation. We ask for information in the bug templates that make it nearly impossible for anyone to help in anyway without.