openbmc/openbmc #3856: [Add Sensor] How to add sensor in 2.12.0-dev

qkpidan, 2022-05-11:
Hi bro!
I added the sensor as #3683, error reported. How to add sensor in 2.12.0-dev?

**Screenshots**
hkzy@openbmc:~/test/openbmc/build$ bitbake obmc-phosphor-image -c clean
WARNING: Layer evb should set LAYERSERIES_COMPAT_evb in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-aspeed should set LAYERSERIES_COMPAT_evb-aspeed in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-ast2500 should set LAYERSERIES_COMPAT_evb-ast2500 in its conf/layer.conf file to list the core layer names it is compatible with.
Loading cache: 100% |##################################################################################################################################################| Time: 0:00:01
Loaded 4221 entries from dependency cache.
**ERROR: /home/hkzy/test/openbmc/meta-phosphor/recipes-phosphor/sensors/phosphor-hwmon_git.bb: Variable FILESEXTRAPATHS_prepend_aspeed file: /home/hkzy/test/openbmc/meta-phosphor/recipes-phosphor/sensors/phosphor-hwmon_%.bbappend line: 1 contains an operation using the old override syntax. Please convert this layer/metadata before attempting to use with a newer bitbake.
ERROR: Parsing halted due to errors, see error messages above**

Summary: There were 3 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.

**OpenBMC Information:**
 - /etc/os-release:
ID=openbmc-phosphor
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.12.0-dev"
VERSION_ID=2.12.0-dev-966-g439a39392-dirty
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.12.0-dev"
BUILD_ID="20220506102527"
OPENBMC_TARGET_MACHINE="evb-ast2500"
 - uname -a:
 Linux evb-ast2500 5.15.34-0626498 #1 Wed May 11 05:49:32 UTC 2022 armv6l GNU/Linux

wangzcheng, 2022-05-11:
FILESEXTRAPATHS_prepend_aspeed is old syntax, you can change it to FILESEXTRAPATHS:prepend:aspeed.

qkpidan, 2022-05-12:
> `FILESEXTRAPATHS_prepend_aspeed` is old syntax, you can change it to `FILESEXTRAPATHS:prepend:aspeed`.

Thanks dud!

Closed.

closed 2022-05-12 by qkpidan