openbmc/openbmc #3857: Помогите решить проблему со сборкой s2600wf

sdivcom-dotcom, 2022-05-15:
I built a s2600wf.
cd openbmc
. setup s2600wf
bitbake obmc-phosphor-image
But towards the end a mistake occurred. Help me solve it. Below is a detailed log of the build.
sem@DESKTOP-169STFE:~/openbmc/build$ bitbake obmc-phosphor-image
WARNING: Host distribution "ubuntu-20.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#########################################################################################################################################################################################################| Time: 0:00:00
Loaded 4177 entries from dependency cache.
Parsing recipes: 100% |#######################################################################################################################################################################################################| Time: 0:00:00
Parsing of 2695 .bb files complete (2694 cached, 1 parsed). 4178 targets, 376 skipped, 0 masked, 0 errors.
WARNING: No bb files in default matched BBFILE_PATTERN_s2600wf ''
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "s2600wf"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm thumb arm1176jzs"
TARGET_FPU           = "soft"
meta
meta-poky
meta-oe
meta-networking
meta-perl
meta-python
meta-phosphor
meta-aspeed
meta-webserver
meta-intel-openbmc
meta-s2600wf
workspace            = "master:21e9a7487d3b871732958155d0a28f49b445871b"

WARNING: /home/sem/openbmc/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_build is tainted from a forced run                                                                                                 | ETA:  0:00:00
Initialising tasks: 100% |####################################################################################################################################################################################################| Time: 0:00:04
Sstate summary: Wanted 24 Local 0 Mirrors 0 Missed 24 Current 1688 (0% match, 98% complete)
NOTE: Executing Tasks
ERROR: intel-ipmi-oem-0.1+gitAUTOINC+493d77616d-r0 do_configure: ExecutionError('/home/sem/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+493d77616d-r0/temp/run.do_configure.18903', 1, None, None)
ERROR: Logfile of failure stored in: /home/sem/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+493d77616d-r0/temp/log.do_configure.18903
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/sem/openbmc/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', '/home/sem/openbmc/meta/recipes-devtools/gcc/gcc-cross_11.3.bb:do_populate_sysroot', '/home/sem/openbmc/meta-intel-openbmc/meta-common/recipes-phosphor/dbus/intel-dbus-interfaces_git.bb:do_populate_sysroot', 'virtual:native:/home/sem/openbmc/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/sem/openbmc/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/home/sem/openbmc/meta-phosphor/recipes-phosphor/ipmi/phosphor-ipmi-host_git.bb:do_populate_sysroot', '/home/sem/openbmc/meta-phosphor/recipes-phosphor/logging/phosphor-logging_git.bb:do_populate_sysroot', '/home/sem/openbmc/meta-openembedded/meta-oe/recipes-support/libgpiod/libgpiod_1.6.3.bb:do_populate_sysroot', '/home/sem/openbmc/meta/recipes-core/systemd/systemd_250.4.bb:do_populate_sysroot', '/home/sem/openbmc/meta/recipes-devtools/gcc/gcc-runtime_11.3.bb:do_populate_sysroot', '/home/sem/openbmc/meta/recipes-devtools/cmake/cmake-native_3.23.1.bb:do_populate_sysroot', '/home/sem/openbmc/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/home/sem/openbmc/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/home/sem/openbmc/meta/recipes-support/boost/boost_1.79.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['ninja-native', 'gcc-cross-arm', 'intel-dbus-interfaces', 'pseudo-native', 'glibc', 'phosphor-ipmi-host', 'phosphor-logging', 'libgpiod', 'systemd', 'gcc-runtime', 'cmake-native', 'quilt-native', 'patch-native', 'boost', 'sdbusplus', 'linux-libc-headers', 'shadow-native', 'phosphor-mapper', 'sdeventplus', 'base-files', 'shadow-sysroot', 'nlohmann-json', 'phosphor-state-manager', 'base-passwd', 'shadow', 'phosphor-dbus-interfaces', 'curl-native', 'zlib-native', 'ncurses-native', 'xz-native', 'zstd-native', 'bzip2-native', 'libgcc', 'attr-native', 'libtool-native', 're2c-native', 'mpfr-native', 'texinfo-dummy-native', 'gmp-native', 'libmpc-native', 'flex-native', 'gnu-config-native', 'binutils-cross-arm', 'packagegroup-obmc-yaml-providers', 'libcereal', 'kmod', 'bash-completion', 'opkg-utils', 'libpam', 'util-linux', 'zstd', 'libcap', 'libseccomp', 'icu', 'bzip2', 'python3', 'zlib', 'libtinyxml2', 'function2', 'stdplus', 'cli11', 'libxcrypt', 'openssl-native', 'gettext-minimal-native', 'm4-native', 'phosphor-software-manager-yaml-provider', 'phosphor-led-manager-yaml-provider', 'xz', 'cracklib', 'flex', 'ncurses', 'util-linux-libuuid', 'libcap-ng', 'autoconf-archive', 'libtirpc', 'libnsl2', 'libffi', 'gdbm', 'sqlite3', 'readline', 'openssl', 'liburing', 'fmt', 'perl-native', 'gdbm-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_configure
| -- The CXX compiler identification is GNU 11.3.0
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Check for working CXX compiler: /home/sem/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+493d77616d-r0/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi/arm-openbmc-linux-gnueabi-g++ - skipped
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| CMake Error at /home/sem/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+493d77616d-r0/recipe-sysroot-native/usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
|   Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
| Call Stack (most recent call first):
|   /home/sem/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+493d77616d-r0/recipe-sysroot-native/usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
|   /home/sem/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+493d77616d-r0/recipe-sysroot-native/usr/share/cmake-3.23/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
|   CMakeLists.txt:61 (find_package)
|
|
| -- Configuring incomplete, errors occurred!
| See also "/home/sem/openbmc/build/tmp/work/arm1176jzs-openbmc-linux-gnueabi/intel-ipmi-oem/0.1+gitAUTOINC+493d77616d-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/sem/openbmc/meta-intel-openbmc/meta-common/recipes-intel/ipmi/intel-ipmi-oem_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3224 tasks of which 3222 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/sem/openbmc/meta-intel-openbmc/meta-common/recipes-intel/ipmi/intel-ipmi-oem_git.bb:do_configure
Summary: There were 3 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.

williamspatrick, 2022-05-15:
```
| Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
```

The `intel-ipmi-oem` repo is missing an `inherit pkgconfig`.

sdivcom-dotcom, 2022-05-16:
> ```
> | Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
> ```
> 
> The `intel-ipmi-oem` repo is missing an `inherit pkgconfig`.

Thank you. How do you fix this problem?

mdmillerii, 2022-05-17:
OpenBMC is an open source  project so I will answer instead

> how can I fix this?  

https://github.com/openbmc/docs#openbmc-development

Patrick was nice enough to identify the cause of your issue.   At this point you should update the package https://github.com/openbmc/openbmc/blob/1972808cb6b2a0c6ea791b50ec3056fbb7343ee8/meta-intel-openbmc/meta-common/recipes-intel/ipmi/intel-ipmi-oem_git.bb#L15  by adding to the inherit detective https://www.yoctoproject.org/docs/2.5/bitbake-user-manual/bitbake-user-manual.html#inherit-directive  the pkgconfig class https://www.yoctoproject.org/docs/2.3/ref-manual/ref-manual.html#ref-classes-pkgconfig 

After you have made the change locally contribute it back to the project https://github.com/openbmc/docs/blob/master/CONTRIBUTING.md so others can benefit and you don't have to carry the fix and others can benefit.

sdivcom-dotcom, 2022-05-17:
> OpenBMC is an open source project so I will answer instead
>
>
> https://github.com/openbmc/docs#openbmc-development
>
> Patrick was nice enough to identify the cause of your issue. At this point you [...]

Thank you very much. I got the build

edtanous, 2022-05-23:
> Thank you very much. I got the build

Closing;  Issue seems to be resolved.

closed 2022-05-23 by edtanous